esp8266/Arduino #3560: Default values for initialization of the ESP8266?

Humancell, 2017-09-01:
When the ESP8266 is first initialized via the SDK, are the default settings:  TX802.11n, MCS7, P OUT=+13 dbm ??

I'm looking at the source and have not found where the initial default values are specified?  Can someone point to a file/line number?

Thanks in advance!

devyte, 2017-09-03:
This is not the right place to ask these questions. This is an issue tracker, to track issues in the core lib source code. Please ask at a forum like esp8266.com, or stackoverflow.
Also, read the docs, including the espressif NONOS sdk api reference and programming guidelines.
Please close this.

closed 2017-09-05 by devyte

Humancell, 2017-09-05:
Uh, well it *IS* that place to ask this, since I'm asking about the the default values that the Core Libs use when initializing the Wifi.

I looked at the SDK docs, and then through the Core Libs source code in an attempt to find what the default values are.  I found these values enumerated in the various headers ... but could not find the line of code where the initialization takes place, and what values are used by default.

I know that when I'm using the Core Libs I do not pass any such parameters by default ... so there must be some defaults that are used by the Core Libs ... but I can't find where.

You're quite hasty to clean up outstanding issues, but you also blew off what I would suggest is a valid question here, since it directly pertains to the Core Libs.

devyte, 2017-09-06:
Like I said, this is an issue tracker for the core libs, not a place to request information, especially questions that have documented answers. 

About the cleanup, we passed 1000+ issues, and if we're to ever have hope of regaining control of the repo, a firmer policy for issues is required. I'm sorry if you feel that is hasty, but 1000+ opened issues shows it is not.

The reason you don't find code that sets those values here in the core is because they're not set in the core. It looks like you are forgettng that the core is built on top of the espressif SDK to make it look Arduino-like.
Most ESP-specific defaults are set in the SDK in one place or another. For example, if you were to read the espressif SDK api reference, like I referenced above, you would find that the output power can be set by an SDK api that takes a value in the range [0, 82], and that the unit is 0.25dbm. The document also says that it is stored as byte 34 of esp_init_data_default.bin.