esp8266/Arduino #3561: WL_NO_SSID_AVAIL reported after many hours of successful operation

zoidicus, 2017-09-01:
### Hardware
Hardware:		Adafruit Huzzah Esp8266 feather
Core Version:      	2.3.0

### Description
After many hours of successful operation, my project (station only mode) loses WiFi connection and starts reporting WL_NO_SSID_AVAIL.  The elapsed time from reboot to eventual failure is quite random; anywhere from about 1 to 20 hours.  I have a scheduled task that blinks the value of WiFi.status() to the on-board LED every 10 seconds, 24x7.  Normally it reports WL_CONNECTED, but once the random disconnect occurs, it reports WL_NO_SSID_AVAIL and remains that way no matter how long I let it sit.  Other functions of the application appear to keep working, although it's tough to be absolutely sure.  Mobile devices and laptops on the same WiFi continue to work fine.  The WiFi router's list of attached devices no longer includes the Esp.   If I manually reset the Esp8266, it reconnects quickly, so clearly the WL_NO_SSID_AVAIL status reported constantly for the past several hours is misleading.   

It is my understanding that the core will handle re-connection, but it does not always appear to be so.  If I manually power off the router, the Eep reports WL_NO_SSID_AVAIL as I would expect, but as soon as I power on the router, the Esp reconnects just fine.  That is not occurring after the random disconnect.  I have not tried any manual disconnect/reconnect procedures, partially because I see so many conflicting discussions about the "correct" way to do it, and whether it is needed at all.

The board is dual powered:  1A USB, and a LIPO battery (stock Adafruit Feather configuration).

### Settings in IDE

Module:  			Adafruit Huzzah Esp8266 
Flash Size: 		4/1
CPU Frequency:	80Mhz
Flash Mode:		?
Flash Frequency:	?
Upload Using:		OTA
Reset Method:		manual reset after board loses WiFi.  otherwise soft resets only.

### Sketch

I don't know that a sketch is really going to be helpful, but here are the relevant bits, trimmed for brevity.  

```cpp

//Partial sketch.  Trimmed for brevity.

#include "functional"
#include "ArduinoOTA.h"
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <EEPROM.h>
#include "door.h"

BlynkTimer myBlynkTimer;
Door door(ENCODER_PIN_A, ENCODER_PIN_B, LIMIT_SWITCH_1_PIN, Door::NoPin, TRIGGER_PIN);

int timerId_reportWifiStatus;
void blinkLedWifiStatus()
{
    //blink the blue LED n times, where n is value returned by WiFi.status();
    //LED is active LOW
    wl_status_t wlStatus = WiFi.status();
    myBlynkTimer.setTimer(500, []()
    {
        digitalWrite(BLUE_LED_PIN, 0);
        myBlynkTimer.setTimeout(100, []() {digitalWrite(BLUE_LED_PIN, 1); });
    }, wlStatus);

    //for reference:
    //typedef enum {
    //    WL_NO_SHIELD = 255,   // for compatibility with WiFi Shield library
    //    WL_IDLE_STATUS = 0,
    //    WL_NO_SSID_AVAIL = 1,
    //    WL_SCAN_COMPLETED = 2,
    //    WL_CONNECTED = 3,
    //    WL_CONNECT_FAILED = 4,
    //    WL_CONNECTION_LOST = 5,
    //    WL_DISCONNECTED = 6
    //} wl_status_t;
}

void setup()
{
    //trimmed for brevity

    WiFi.mode(WIFI_STA);

    Blynk.begin(connectionConfig.auth, connectionConfig.ssid, connectionConfig.pass, IPAddress(192, 168, 5, 160));

    ArduinoOTA.setPassword("398A0FBE5CD04029902FA45856313D93");
    ArduinoOTA.begin();

    timerId_reportWifiStatus = myBlynkTimer.setInterval(10005, blinkLedWifiStatus);

    pinMode(BLUE_LED_PIN, OUTPUT);
    digitalWrite(BLUE_LED_PIN, HIGH); //HIGH turns led OFF (they are wired active low).
}

void loop()
{
    Blynk.run();                   //Blynk library task
    ArduinoOTA.handle();  //OTA task
    myBlynkTimer.run();     //scheduler task
    door.run();                    //application task (no delays or wait-loops)
}

darrendavidhumphrey, 2018-08-29:
Did you ever find a solution to this problem?  I'm experiencing the exact same symptoms :(

d-a-v, 2018-08-29:
If not tried yet, you can have a go with [these tweaks](https://github.com/esp8266/Arduino/issues/5083#issuecomment-416194799) and report here.
You can also try to reset (only) wifi if the above don't help.

darrendavidhumphrey, 2018-08-29:
Thanks a lot for the link!  I will try the tweaks.

Have you had good success getting stable operation with your ESP8266
projects?

Darren