esp8266/Arduino #3563: ESP8266WiFiSTAClass::disconnect erases saved WiFi connection

hoekma, 2017-09-02:
### Basic Infos

Issue with the WPS connection

#### Hardware
Hardware:			ESP-12
Core Version:      	        2.3

### Description

Steps to reproduce:

1) Put a WiFi router into WPS mode to accept WPS connections
2) On ESP8266 device that does not have a saved WiFi connection, run WiFi.beginWPSConfig().  (Device successfully connects via WPS to a WiFi endpoint)
2) Each subsequent connection it works fine if WiFi in step 1 is available to connect
3) Power the WiFi endpoint off and repeat step 2  (this erases saved WiFi SSID and credentials)
4) Power WiFi router back on
5) Device is unable to connect because it no longer knows about the WiFi router until 

Use case is :

1) Device turns on and tries to connect to WiFi
2a) If device finds its saved connection, don't initiate WPS
2) If can't connect to its saved connection, initiate WPS automatically.

This way the wps configuration is easier for users who are trying to initialize a new device.

In the event of a power outage or temporary outage of wifi, current behavior will disable the esp8266 networking capabilities until WPS handshake is re-initiated

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch
...

//----------------------------
//--  void initWPS(){

bool initWPS() {
  // Attempt a WPS configuration
  // Response codes 3 flashes for success(green) fail(red)

  bool returnValue = false;
  String StoredSSID = WiFi.SSID().c_str();
  String StoredPSK = WiFi.psk().c_str();

  // Light both LEDs
  digitalWrite(RedLED, HIGH);
  digitalWrite(GreenLED, HIGH);

  if (!startWPSPBC()) {
    // WPS Negotiation failed
    digitalWrite(GreenLED, LOW);
    digitalWrite(RedLED, LOW);
    delay(500);
    flashRed(3);
    if (devMode) {
      Serial.print("WiFi.status() after WPS Failure: ");
      Serial.println(WiFi.status());
    }
  }
  else {
    //WPS returned success
    digitalWrite(GreenLED, LOW);
    digitalWrite(RedLED, LOW);

    // Validate that we can connect to WiFi
    delay(2000);
    if (devMode) {
      Serial.print("WiFi.status() after WPS Success: ");
      Serial.println(WiFi.status());
    }

    // WL_DISCONNECTED means it is ready for a WiFi.begin() method call.
    if (WiFi.status() == WL_DISCONNECTED) {
      returnValue = true;
      delay(200);
      flashGreen(3);
    }
    else {
      // Even though WPS says it succeeded,
      // we were not able to connect
      flashRed(3);
      if (devMode) {
        Serial.println("Failed to connect with WPS");
      }
    }
  }
  return returnValue;
}

//----------------------------
//----------------------------
//----------------------------
//--  startWPSPBC

bool startWPSPBC() {
  if (devMode) {
    Serial.println("WPS config start");
  }
  bool wpsSuccess = WiFi.beginWPSConfig();
  if (wpsSuccess && WiFi.status() == WL_CONNECTED) {
    if (devMode) {
      Serial.printf("WPS finished. Connected successfull to SSID '%s'\n", WiFi.SSID().c_str());
    } else {
      wpsSuccess = false;
      if (devMode) {
        Serial.printf("WPS Connect Failed");
      }
    }
  }
  return wpsSuccess;
}

...

### RECOMMENDED CHANGE TO ESP8266WiFiSTAClass::disconnect:

/**
 * Disconnect from the network
 * @param wifioff
 * @return  one value of wl_status_t enum
 */
bool ESP8266WiFiSTAClass::disconnect(bool wifioff) {
    bool ret;

     // from hoekma - remove to prevent losing WPS configuration
    // struct station_config conf;
    // *conf.ssid = 0;
    // *conf.password = 0;

    ETS_UART_INTR_DISABLE();

     // from hoekma - remove to prevent losing WPS configuration
    //if(WiFi._persistent) {
    //    wifi_station_set_config(&conf);
    //} else {
    //    wifi_station_set_config_current(&conf);
    //}

    ret = wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();

    if(wifioff) {
        WiFi.enableSTA(false);
    }

    return ret;
}

igrr, 2017-09-12:
My view is that `disconnect` should indeed remove saved connection parameters.

To achieve your desired behaviour, you may do the following in the application:

```c++
// Save current WiFi credentials
String old_ssid = WiFi.SSID();
String old_psk = WiFi.psk();
// Perform WPS
bool wpsSuccess = WiFi.beginWPSConfig();
if (wpsSuccess && WiFi.status() == WL_CONNECTED) {
  // all good
} else {
  // failed to connect
  if (old_ssid.length() > 0) {
    // restore old WiFi credentials
    WiFi.begin(old_ssid, old_psk);
  }
}
```

closed 2017-09-12 by igrr

hoekma, 2017-09-12:
I was able to make your solution work with a slight modification of one line:
`WiFi.begin(old_ssid.c_str(), old_psk.c_str());`

After considering your position on the disconnect() method, I would still maintain that calling WiFi.beginWPSConfig() should not erase the previously saved SSID / PSK.   What is the purpose of deleting this data from the device at the beginning of a WPS connection attempt?  I'm happy to code around it, but don't understand why it was developed that way.  It seems to me the existing connection information should stay in place until either:

1) A different SSID connection is saved 
                         or 
2) A developer specifically wants it to be deleted.

Anyway, thanks in advance for your explanation and thank you for your help with the work around.