esp8266/Arduino #3564: Strange behaviour/Exceptions with printf_P()

innodron, 2017-09-02:
I had been searching for days to explain the behaviour below; finally concluding there might a problem regarding printf_P as shown in the test code output below (I have been receiving exceptions and cyclic reboots with a larger sketch when I use PSTR() or printf_P).

The development environment is
- Sparkfun ESP8266 Thing Dev 
- MacBook Pro
- Arduino IDE 1.8.4
- Board Library: esp8266 by ESP8266 Community version 2.3.0:
- CPU Freq: 80Mhz
- Programmer: AVRISP mkII

```
#if (defined(__AVR__))
#include <avr/pgmspace.h>
#else
#include <pgmspace.h>
#endif
PROGMEM const char progmemtext1[] = "progmem text1";
PROGMEM const char progmemtext2[] = "progmem text2\n";
PROGMEM const char progmemtext3[] = "progmem format text for %s\n";

void setup() {
  Serial.begin (115200);
  delay(1000);
}

void loop() {
  Serial.println("\n+++ test begin");
  
  Serial.println("inline text");
  Serial.println(F("inline F() text"));
  Serial.println(FPSTR(progmemtext1));
  printf_P(progmemtext2);
  printf_P(PSTR("inline PSTR() text with printf_P\n"));
  char str[80];
  sprintf_P(str,PSTR("inline PSTR() format text with sprintf_P for %s"),"test123");
  Serial.println(str);
  printf_P(PSTR("inline PSTR() format text for %s\n"),"test123");
  printf_P(progmemtext3,"test123");

  Serial.println("\n+++ test end");
  delay(5000);
}
```

And the serial monitor showing:
```
+++ test begin
inline text
inline F() text
progmem text1
progmem text2
inline PSTR() text with printf_P
inline PSTR() format text with sprintf_P for test123
inline PSTR() format text for   <--- test123 missing
progmem format text for ⸮⸮⸮? <--- test123 corrupted

+++ test end
```

devyte, 2017-09-02:
Please try latest git, I seem to remember some changes to progmem.
If you are thinking "how do I use the latest git", readthedocs.

innodron, 2017-09-02:
With the latest git (2.4.0-rc.1) the output for the code above is exactly the same.
```
+++ test begin
inline text
inline F() text
progmem text1
progmem text2
inline PSTR() text with printf_P
inline PSTR() format text with sprintf_P for test123
inline PSTR() format text for 
progmem format text for @⸮?

+++ test end
```

However, diff for print.h reveals there Serial.printf_P is now added, which was missing in 2.3.0. Hence I tried changing all printf_P to Serial.printf_P:

```
#if (defined(__AVR__))
#include <avr/pgmspace.h>
#else
#include <pgmspace.h>
#endif
PROGMEM const char progmemtext1[] = "progmem text1";
PROGMEM const char progmemtext2[] = "progmem text2\n";
PROGMEM const char progmemtext3[] = "progmem format text for %s\n";

void setup() {
  Serial.begin (230400);
  delay(1000);
}

void loop() {
  Serial.println("\n+++ test begin");
  
  Serial.println("inline text");
  Serial.println(F("inline F() text"));
  Serial.println(FPSTR(progmemtext1));
  Serial.printf_P(progmemtext2);
  Serial.printf_P(PSTR("inline PSTR() text with printf_P\n"));
  char str[80];
  sprintf_P(str,PSTR("inline PSTR() format text with sprintf_P for %s"),"test123");
  Serial.println(str);
  Serial.printf_P(PSTR("inline PSTR() format text for %s\n"),"test123");
  Serial.printf_P(progmemtext3,"test123");

  Serial.println("\n+++ test end");
  delay(5000);
}
```
and the output is:
```
+++ test begin
inline text
inline F() text
progmem text1
progmem text2
inline PSTR() text with printf_P
inline PSTR() format text with sprintf_P for test123
inline PSTR() format text for test123
progmem format text for test123

+++ test end
```

So, the issue seems fixed. Thanks. Just one last question though _(apologies if it's a newbie question, but I just want to learn rather then copy/paste code)_
- Why do I get different results for Serial.printf_P and printf_P? They do both output to Serial monitor but I get different results as indicated in the outputs above

devyte, 2017-09-02:
That is something that should be investigated. Most likely they have
different implementations in the core. I would think they should be the
same, or better yet, one should call the other.
Care to take it up? It suspect it shouldn't be hard to find the code for
both functions and compare them.

igrr, 2017-12-27:
This has been fixed in 7a93478a99a23004e44758603bc8a4953f230205. Verified with latest git version.

closed 2018-01-02 by igrr