esp8266/Arduino #3566: Missing nodemcu board after switching from Board Manager Version to Git Version in mac

Bimblis, 2017-09-03:
I've been using Board Manager version 2.3, and I can revert to it in any moment. But after following the readme steps to prepare the Git version, the esp8266 boards dont appear.

[image]

[image]

I have emptied the "additional board manager url" and deleted the 2.3 version from packages folder. Am I missing something?

igrr, 2017-09-12:
Installation instructions do not make this clear, but you need to clone the git repository into either

- hardware/esp8266com/esp8266 directory in Arduino IDE (so the full path is Arduino.app/Contents/Java/hardware/esp8266com/esp8266),
- or into your sketchbook (by default, $HOME/Documents/Arduino, so the full path is $HOME/Documents/Arduino/hardware/esp8266com/esp8266)

Cloning into ~/Library/Arduino15/packages, as you did, will not work because Arduino IDE happens to use slightly different path rules for cores installed into the IDE/sketchbook and the ones installed into the downloaded packages directory.

closed 2017-09-12 by igrr

YUVARAJ1186, 2018-01-17:
igrr            IAM FACING PROBLEM WHAT U TOLD IN COMMENTS PLS EXPLAIN IT BRIEFLY BY STEP BY STEP IAM WAITING FOR U R REPLY

kishan2006, 2018-04-16:
arduino 1.8.1 in board manager not desply esp 8266. so u tell me what i do ?