esp8266/Arduino #3568: nodemcu crashes when connecting to wifi

JordanKlaers, 2017-09-03:
### Basic Infos

#### Hardware
Hardware:			nodemcu 1.0 (ESP-12E module)
Core Version:      	2.1.0
Arduino IDE:  1.8.2

### Description

it seems when the chip trying to call wifimulti.run() it causes the chip to crash. If I removed the function that connected to the wifi for the get request, the rest of the code does run correctly. This exact sketch was working before. I had shorted out the chip I was using and bought two more of the exact same chip. I apologize in advance for not being extremely educated I have tried my best to understand the issue through countless hours of researching but its still a bit over my head. 

### Settings in IDE

Module:  			nodemcu 1.0 (ESP-12E module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio (unknown might be dio?)
Flash Frequency:	n/a  (dont see this field)
Upload Using:		SERIAL
Reset Method:		n/a  (dont see this field)

### Sketch

```
   
    
    int red = 15;
    int green = 12;
    int blue = 13;
    int value = 0;

    int redbrightlvl = 0;
    int greenbrightlvl = 0;
    int bluebrightlvl = 0;
    int redlvl = 0;
    int greenlvl = 0;
    int bluelvl = 0;
    int buttonState = 0;         //0 not pressed/no data .     1 pressed, need to get data

    #include <Arduino.h>
    #include <ESP8266WiFi.h>
    #include <ESP8266WiFiMulti.h>
    #include <ESP8266HTTPClient.h>
    #define USE_SERIAL Serial
    ESP8266WiFiMulti WiFiMulti;

    String payload = "[['000 255 000 '],'15029324934080383']";

    int doit = 0;

    String urlTimeStamp = "";
    String urlFirstCall = String("0");
    String urlFirstHalf = String("http://ps.pndsn.com/subscribe/sub-c-cf99383a-7714-11e7-98e2-02ee2ddab7fe/theled/0/");
    String urlSecondHalf = String("/2a80a1c6-8743-46d8-aea6-eae26aabc454");
    
    boolean isFirstCall = true;

void setup() {
  delay(1000);
  Serial.begin(115200);
//  Serial.setDebugOutput(true);
  pinMode (blue, OUTPUT);
  pinMode (green, OUTPUT);
  pinMode (red, OUTPUT);
  pinMode(4,INPUT); 
  analogWrite(red, 1023);
  analogWrite(blue, 1023);
  analogWrite(green, 1023);

  for(uint8_t t = 4; t > 0; t--) {
[... 287 lines cut ...]
0x40201738: prep_pwm_steps at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_wiring_pwm.c line 89
0x40227b72: wifi_get_opmode at ?? line ?
0x40228997: wifi_station_get_connect_status at ?? line ?
0x40203848: ESP8266WiFiSTAClass::status() at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 635
0x402071c8: Print::write(unsigned char const*, unsigned int) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.cpp line 42
0x402035e3: ESP8266WiFiMulti::run() at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp line 156
0x4020742b: Print::print(long, int) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.cpp line 119
0x402071c8: Print::write(unsigned char const*, unsigned int) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.cpp line 42
0x402071c8: Print::write(unsigned char const*, unsigned int) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.cpp line 42
0x40205cb1: Print::write(char const*) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.h line 60
0x40206058: esp_yield at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_main.cpp line 43
0x402010ae: delay at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_wiring.c line 53
0x40202ae0: getData() at /Users/jordanklaers/Documents/SSD/ChromaCrown/ChromaCrown.ino line 99
0x40205cb1: Print::write(char const*) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.h line 60
0x402010a3: delay at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_wiring.c line 50
0x402073cd: Print::printNumber(unsigned long, unsigned char) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.cpp line 226
0x4010020c: _umm_free at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1281
0x402071c8: Print::write(unsigned char const*, unsigned int) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.cpp line 42
0x40205cb1: Print::write(char const*) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.h line 60
0x40205cb1: Print::write(char const*) at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.h line 60
0x402072a0: Print::println() at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/Print.cpp line 144
0x4020318e: loop at /Users/jordanklaers/Documents/SSD/ChromaCrown/ChromaCrown.ino line 85
0x40206079: esp_schedule at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_main.cpp line 43
0x402060a4: loop_wrapper at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_main.cpp line 43
0x40100958: cont_norm at /Users/jordanklaers/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/cont.S line 109

igrr, 2017-09-12:
Could you please see if you can reproduce the issue with a smaller sketch? Also, please clarify the version of the core which you are using. You have reported 2.1.0, which is very old (2.3.0 is the latest stable one, 2.4.0-rc1 is the latest pre-release one).

JordanKlaers, 2017-09-14:
I switched to version 2.3.0 and I am running 
`/**
 * BasicHTTPClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    USE_SERIAL.begin(115200);
   // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("pootdooter", "minenotyours");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        //http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"); //HTTPS
        http.begin("http://192.168.1.12/test.html"); //HTTP

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);
}
`

I am unable to load to the board. I have tried tying gpio 0 to gnd and holding down flash when uploading. I recieve the error espcomm_upload_mem failed  with
`esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.SLAB_USBtoUART at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 243184 bytes from /var/folders/3l/4d_mcscd7ss1v6xbbnjpmd0h0000gp/T/arduino_build_528486/BasicHttpClient.ino.bin to flash at 0x00000000
	erasing flash
	size: 03b5f0 address: 000000
	first_sector_index: 0
	total_sector_count: 60
	head_sector_count: 16
	adjusted_sector_count: 44
	erase_size: 02c000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 100
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed`

I tried going back to version 2.2 and that failed as well.
I then went abck to  2.1 with all the same setting I had documented in this post and I am getting the error below

`esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.SLAB_USBtoUART at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 286304 bytes from to flash at 0x00000000
	erasing flash
	size: 045e60 address: 000000
	first_sector_index: 0
	total_sector_count: 70
	head_sector_count: 16
	adjusted_sector_count: 54
	erase_size: 036000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting character timeout 150
	done
	serialport_receive_C0: 72 instead of C0
	setting timeout 1
	setting character timeout 1
	done
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
`

Im quite overwhelmed at how even going back to settings which I had documented, produces different results

devyte, 2017-10-10:
@JordanKlaers I suggest deleting .arduino15 and the arduino install (backup your projects first), then doing a clean install. Then, erase the entire flash (e.g.: with esptool), and flash a clean sketch.
Also, please try with latest git, since 2.3.0 is very old.

devyte, 2018-07-12:
Closing due to lack of feedback.

closed 2018-07-12 by devyte