esp8266/Arduino #3570: Wire.h cannot be initialized from a class constructor in a library

willz1200, 2017-09-04:
### Basic Infos

Wire.h cannot be initialized from a class constructor in a library.

#### Hardware
Hardware:		ESP-12E
Core Version:      	2.3.0

### Description

When trying to use Wire.begin(4, 5) from a class constructor within a library the esp crashes Exception (28). Because the class is initialized within the library, Wire.begin will be ran before the setup loop. I'm using software I2c which uses the twi.h esp8266 core.

By the way I've also tried doing this with SPI.begin() and it worked perfectly, so its just an issue with a feature of Wire.h

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M (3M SPIFFS)
CPU Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		NodeMCU

### LibraryUsingWire.h

```cpp
#include <Arduino.h>
#include <Wire.h>

class classUsingWire{
	public:
		classUsingWire(); //The Constructor
};

extern classUsingWire Something;
```

### LibraryUsingWire.cpp

```cpp
#include <LibraryUsingWire.h>

classUsingWire::classUsingWire(){
	Serial.begin(115200);
	Serial.println("Serial Online");
	Wire.begin(4, 5);
}

classUsingWire Something; //Initialize class
```

### Sketch

```cpp
#include <Arduino.h>
#include <Wire.h>
#include <LibraryUsingWire.h>

void setup() {
  
}

void loop() {

}
```

### Debug Messages

```
Serial Online

Exception (28):
epc1=0x40201d9e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000014 depc=0x00000000

ctx: sys 
sp: 3ffff900 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffaa0:  4022fd50 00000001 3ffee2e8 40201d9c  
3ffffab0:  3ffe8d9c 60000600 3ffee3ec 40201e05  
3ffffac0:  4020439a 00000000 3ffe8918 40201e20  
3ffffad0:  402050ca 00000000 60000600 402022b1  
3ffffae0:  00000000 40206ef9 00000000 40206f05  
3ffffaf0:  40101b70 000003fd 00000010 60000200  
3ffffb00:  3fffff10 d7be081a 9f9205fd c357c8b0  
3ffffb10:  ffffffff ffffffff ffffff00 ffffffff  
3ffffb20:  ffffffff ffffffff ffffffff ffffffff  
3ffffb30:  ffffffff ffffffff ffffffff ffffffff  
3ffffb40:  ffffffff ffffffff ffffffff ffffffff  
3ffffb50:  ffffffff ffffffff ffffffff ffffffff  
3ffffb60:  ffffffff ffffffff ffffffff ffffffff  
3ffffb70:  ffffffff ffffffff ffffffff ffffffff  
3ffffb80:  ffffffff ffffffff ffffffff ffffffff  
3ffffb90:  ffffffff ffffffff ffffffff ffffffff  
3ffffba0:  ffffffff ffffffff ffffffff ffffffff  
3ffffbb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffbc0:  00000005 63746157 3fff0068 00000269  
3ffffbd0:  00000269 4010020c 0000001c 4023f940  
3ffffbe0:  000003ff 63696863 356e656b 3fff7800  
3ffffbf0:  000003ff 40107688 402273a2 000003ff  
3ffffc00:  4023f940 402273bf 3fff9594 40227465  
3ffffc10:  3fff8b5c 000003ff 3fff6954 00000122  
3ffffc20:  00000122 82280200 44ffbfbc f80572d3  
3ffffc30:  b4e49a7c c8d21cc7 b9e7e111 67ae46ea  
3ffffc40:  40227a3e 00030183 ffffff04 ffffffff  
3ffffc50:  ffffffff ffffffff ffffffff ffffffff  
3ffffc60:  ffffffff ffffffff ffffffff ffffffff  
3ffffc70:  ffffffff ffffffff ffffffff ffffffff  
3ffffc80:  ffffffff ffffffff ffffffff ffffffff  
3ffffc90:  ffffffff ffffffff ffffffff ffffffff  
3ffffca0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcc0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffce0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffd00:  ffffffff ffffffff ffffffff ffffffff  
3ffffd10:  ffffffff ffffffff ffffffff ffffffff  
3ffffd20:  ffffffff ffffffff ffffffff ffffffff  
3ffffd30:  ffffffff ffffffff ffffffff ffffffff  
3ffffd40:  ffffffff ffffffff ffffffff ffffffff  
3ffffd50:  ffffffff ffffffff ffffffff ffffffff  
3ffffd60:  ffffffff ffffffff ffffffff ffffffff  
3ffffd70:  ffffffff ffffffff ffffffff ffffffff  
3ffffd80:  ffffffff ffffffff ffffffff ffffffff  
3ffffd90:  ffffffff ffffffff ffffffff ffffffff  
3ffffda0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdc0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdd0:  ffffffff ffffffff ffffffff ffffffff  
3ffffde0:  ffffffff ffffffff ffffffff ffffffff  
3ffffdf0:  ffffffff ffffffff ffffffff ffffffff  
3ffffe00:  ffffffff ffffffff ffffffff ffffffff  
3ffffe10:  ffffffff ffffffff ffffffff ffffffff  
3ffffe20:  ffffffff ffffffff ffffffff ffffffff  
3ffffe30:  ffffffff ffffffff ffffffff ffffffff  
3ffffe40:  ffffffff ffffffff ffffffff ffffffff  
3ffffe50:  ffffffff ffffffff ffffffff ffffffff  
3ffffe60:  ffffffff ffffffff ffffffff ffffffff  
3ffffe70:  ffffffff ffffffff ffffffff ffff0064  
3ffffe80:  ffffffff ffffffff ffffffff ffffffff  
3ffffe90:  ffffffff ffffffff ffffffff ffffffff  
3ffffea0:  ffffffff ffffffff ffffffff ffffffff  
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffec0:  ffffffff ffffffff ffffffff ffffffff  
3ffffed0:  ffffffff ffffffff ffffffff ffffffff  
3ffffee0:  ffffffff ffffffff ffffffff ffffffff  
3ffffef0:  ffffffff ffffffff ffffffff ffffffff  
3fffff00:  ffffffff ffffffff ffffffff ffffffff  
3fffff10:  ffffffff ffffffff ffffffff ffffffff  
3fffff20:  ffffffff ffffffff ffffffff ffffffff  
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff ffffffff ffffffff ffffffff  
3fffff60:  ffffffff ffffffff ffffffff ffffffff  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff ffffffff ffffffff ffffffff  
3fffff90:  ffffffff 0c784cb7 b43f520b 70625e9b  
3fffffa0:  400201e9 432cf9d6 ffffff01 55aa55aa  
<<<stack<<<
```

willz1200, 2017-09-08:
I had a look into the issue, it seems to be caused by the flush() called from the software begin function. 

### Wire.cpp (Original)
```cpp
void TwoWire::begin(int sda, int scl){
  default_sda_pin = sda;
  default_scl_pin = scl;
  twi_init(sda, scl);
  flush();
}
```

### Wire.cpp (Temporary Fix)
```cpp
void TwoWire::begin(int sda, int scl){
  default_sda_pin = sda;
  default_scl_pin = scl;
  twi_init(sda, scl);
  Wire.flush();
}
```

### Wire.cpp (Possible Long Term Fix)
```cpp
void TwoWire::begin(int sda, int scl){
  default_sda_pin = sda;
  default_scl_pin = scl;
  twi_init(sda, scl);
}
```

It appears that the flush function is conflicting with something external to Wire.h, possibly through the "Stream" class of which TwoWire is a member? The temporary fix will only work if the user doesn't require multiple objects of TwoWire.

I don't really see why the flush() is required at this point, because the buffers are initialised to 0.

Thanks,
Will

willz1200, 2017-09-08:
#3586

devyte, 2017-09-08:
@willz1200 I think the flush() is needed to allow calling begin() multiple times at runtime, i.e.: when you want to change pins to use a secondary bus. In that scenario, the buffers may be filled with something valid, and flushing could make sense.
Personally, I wonder why in the previous scenario it is not common use to instantiate a second object.

About your particular case, I have to wonder about the order in which the global objects are constructed. Global objects are constructed in the order in which they are declared _within one translation unit_. The order of construction _between translation units_ is not defined, and may change as you add more TUs to your project, or due to your build system. Here's a quick [reference](https://stackoverflow.com/questions/7222743/in-what-order-are-global-constructors-called)
In addition, there are 2 phases to global scope init: the static phase and the dynamic one. The static phase allocates the object (i.e.: creates the space for it), and in case of const expressions, such as int a = 10, it assigns the static values.
The dynamic phase consists of functions that need to be called for init, such as constructors, which comes later, and here the order of declaration can cause issues.
In other words, if your LibraryUsingWire constructs before Wire, you can't expect Wire.begin() to work correctly. If Wire is not constructed, flush() will try do its thing with uninited data members filled with garbage.

The Arduino style makes use of the singleton pattern. There is a reason why the begin() methods exists. I suggest making a begin() method for your LibraryUsingWire, and calling Wire.begin() from within it. Then, call your begin method from e.g.: setup. This makes it compliant with the Arduino style.

Please let me know if the above resolves your issue.

willz1200, 2017-09-08:
Ah I see, I didn't realise that was a possibility. I would of thought that Wire was initialised first because I made sure to put `#include <Wire.h>` before `#include <LibraryUsingWire.h>`, thank you for the information I with have a read into it, yes for now I just call a initialise function from the setup.

devyte, 2017-09-12:
@willz1200 you are talking about declaration, which is determined by the order of #includes. The order of linkage, i.e.: the order that TUs get put together into the binary, is what determines the init phases, and there isn't really control over that.
I've stumbled into this myself a few times, I forget about it every few years.
This is a dicussion related to how compilers in general work, and about the pitfalls of global object init sequences. I think it is a dicsussion best left for a forum, not this issue tracker.
Closing as off-topic.

closed 2017-09-12 by devyte