esp8266/Arduino #3574: Download a file to a FS

mcmega, 2017-09-05:
Hi All!
I need help.
How to download a file to a file system from an Internet server?
I want to download the file from the server (https).

### Settings in IDE
Module:  			Generic ESP8266 Module (D1 mini)
Flash Size: 		1MB(256K SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Reset Method:		nodemcu

```
void HTTPClient httpClient;
    httpClient.useHTTP10(true);
    httpClient.setTimeout(5000);
    char url[128];
    sprintf(
        url, "https://%s%s/%s.hex", 
        OTA_HOST, OTA_BASE_URL, NAME
    );
    httpClient.begin(url, OTA_SSL_FINGERPRINT);
    int code = httpClient.GET();
    if (code != HTTP_CODE_OK)
        httpClient.end();
}
```

suculent, 2017-09-05:
This is not an Arduino issue. Please ask elsewhere (ESP8266.com or StackOverflow). 

Duplicates https://github.com/esp8266/Arduino/issues/2834

Odesláno z iPhonu

devyte, 2017-09-05:
There is a new policy being discussed to formalize it, but @suculent is correct.

closed 2017-09-05 by devyte