esp8266/Arduino #3582: boards.txt generation script

davisonja, 2017-09-06:
There is a plan to genereate boards.txt, and related files (such as matching linker scripts) via a python script (as python is already in use).
The primary aim is to reduce (to 1) the number of places that changes need to be made for things like flash addresses (splits between program/SPIFFS/EEPROM/etc). The desire was inspired by a solution to #905 which stores update information in a flash page, along with some other updates that involved changes to flash allocation.

d-a-v, 2017-10-11:
I started such a script (python) and ask @igrr about its pertinence. He redirected me to #1969 and told me that there is not yet anything ongoing about it. I just found this #3582.

I was originally wanting to add D1-mini-lite (#3707 #3300 #3169) and I also have to deal with lwip2 PR which heavily modifies boards.txt.

What I propose to do is a separate repo to share and work on / try it, so to make an PR when it is acceptable ?

d-a-v, 2017-10-11:
Here's the repo with my preliminary contribution: https://github.com/d-a-v/esp82xx-arduino-boardgen
I would grant write access to anybody wanting to make this quickly working :)

davisonja, 2017-10-11:
Nice. Though I see I never came back to describe what I'd done :/
Would you like to have a discussion there or here? :)

d-a-v, 2017-10-11:
Here. There is only a lonely public repo.

davisonja, 2017-10-12:
Sounds like a plan.
I started from the 'other end' - take an existing boards.txt and turn it into a structure that reduced the huge amount of redundancy that is in boards.txt (most boards have identical options, save a few tweaks) while remaining relatively understandable and fairly simple to tweak for new entries.

The plan is effectively to have a 'generic' baseline, and a series of derivative boards which specified what changes needed to be made - in a very plaintext verbose kind of way.
My script currently produces the stuff below, which is a rough python structure, has the generic settings then differences for each board and currently has no support for things that are in the generic definition but shouldn't be in the derived board. The script processes the boards.txt file into `dict()`'s and prints them.

The result looks like this:
````python
{'build.board': 'ESP8266_ESP01',
 'build.core': 'esp8266',
 'build.debug_level': '',
 'build.debug_port': '',
 'build.f_cpu': '80000000L',
 'build.flash_mode': 'qio',
 'build.mcu': 'esp8266',
 'build.spiffs_pagesize': '256',
 'build.variant': 'generic',
 'menu.CpuFrequency.160': '160 MHz',
 'menu.CpuFrequency.160.build.f_cpu': '160000000L',
 'menu.CpuFrequency.80': '80 MHz',
 'menu.CpuFrequency.80.build.f_cpu': '80000000L',
 'menu.CrystalFreq.26': '26 MHz',
 'menu.CrystalFreq.40': '40 MHz',
 'menu.CrystalFreq.40.build.extra_flags': '-DF_CRYSTAL=40000000',
 'menu.Debug.Disabled': 'Disabled',
 'menu.Debug.Disabled.build.debug_port': '',
 'menu.Debug.Serial': 'Serial',
 'menu.Debug.Serial.build.debug_port': '-DDEBUG_ESP_PORT=Serial',
 'menu.Debug.Serial1': 'Serial1',
 'menu.Debug.Serial1.build.debug_port': '-DDEBUG_ESP_PORT=Serial1',
 'menu.DebugLevel.Core____': 'Core',
 'menu.DebugLevel.Core____.build.debug_level': '-DDEBUG_ESP_CORE',
 'menu.DebugLevel.HTTPClient': 'HTTPClient',
 'menu.DebugLevel.HTTPClient.build.debug_level': '-DDEBUG_ESP_HTTP_CLIENT',
 'menu.DebugLevel.HTTPClient2': 'HTTPClient + SSL',
 'menu.DebugLevel.HTTPClient2.build.debug_level': '-DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_SSL',
 'menu.DebugLevel.HTTPServer': 'HTTPServer',
 'menu.DebugLevel.HTTPServer.build.debug_level': '-DDEBUG_ESP_HTTP_SERVER',
 'menu.DebugLevel.HTTPUpdate': 'HTTPUpdate',
 'menu.DebugLevel.HTTPUpdate.build.debug_level': '-DDEBUG_ESP_HTTP_UPDATE',
 'menu.DebugLevel.HTTPUpdate2': 'HTTPClient + HTTPUpdate',
 'menu.DebugLevel.HTTPUpdate2.build.debug_level': '-DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_UPDATE',
 'menu.DebugLevel.HTTPUpdate3': 'HTTPClient + HTTPUpdate + Updater',
 'menu.DebugLevel.HTTPUpdate3.build.debug_level': '-DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_UPDATER',
 'menu.DebugLevel.None____': 'None',
 'menu.DebugLevel.None____.build.debug_level': '',
 'menu.DebugLevel.OTA2____': 'OTA + Updater',
 'menu.DebugLevel.OTA2____.build.debug_level': '-DDEBUG_ESP_OTA -DDEBUG_ESP_UPDATER',
 'menu.DebugLevel.OTA_____': 'OTA',
 'menu.DebugLevel.OTA_____.build.debug_level': '-DDEBUG_ESP_OTA',
 'menu.DebugLevel.SSL_MEM_': 'Core + SSL + TLS Mem',
 'menu.DebugLevel.SSL_MEM_.build.debug_level': '-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_TLS_MEM',
 'menu.DebugLevel.SSL_____': 'Core + SSL',
 'menu.DebugLevel.SSL_____.build.debug_level': '-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL',
 'menu.DebugLevel.UPDATER': 'Updater',
 'menu.DebugLevel.UPDATER.build.debug_level': '-DDEBUG_ESP_UPDATER',
 'menu.DebugLevel.WiFi____': 'WiFi',
 'menu.DebugLevel.WiFi____.build.debug_level': '-DDEBUG_ESP_WIFI',
 'menu.DebugLevel.WiFic___': 'Core + WiFi',
 'menu.DebugLevel.WiFic___.build.debug_level': '-DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI',
 'menu.DebugLevel.all_____': 'All',
 'menu.DebugLevel.all_____.build.debug_level': '-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM',
 'menu.FlashFreq.40': '40MHz',
 'menu.FlashFreq.40.build.flash_freq': '40',
 'menu.FlashFreq.80': '80MHz',
 'menu.FlashFreq.80.build.flash_freq': '80',
 'menu.FlashMode.dio': 'DIO',
 'menu.FlashMode.dio.build.flash_mode': 'dio',
 'menu.FlashMode.dout': 'DOUT',
 'menu.FlashMode.dout.build.flash_mode': 'dout',
[... 337 lines cut ...]
               'menu.ESPModule.ESP07192.upload.maximum_size': '827376',
               'menu.ESPModule.ESP12': 'ESP12 (4M/1M SPIFFS)',
               'menu.ESPModule.ESP12.build.board': 'ESP8266_ESP12',
               'menu.ESPModule.ESP12.build.flash_ld': 'eagle.flash.4m1m.ld',
               'menu.ESPModule.ESP12.build.flash_size': '4M',
               'menu.ESPModule.ESP12.build.spiffs_blocksize': '8192',
               'menu.ESPModule.ESP12.build.spiffs_end': '0x3FB000',
               'menu.ESPModule.ESP12.build.spiffs_pagesize': '256',
               'menu.ESPModule.ESP12.build.spiffs_start': '0x300000',
               'menu.ESPModule.ESP12.upload.maximum_size': '1044464',
               'name': 'WifInfo',
               'upload.resetmethod': 'nodemcu'},
 'menu': {'BoardModel': 'Model',
          'CpuFrequency': 'CPU Frequency',
          'CrystalFreq': 'Crystal Frequency',
          'Debug': 'Debug port',
          'DebugLevel': 'Debug Level',
          'ESPModule': 'Module',
          'FlashFreq': 'Flash Frequency',
          'FlashMode': 'Flash Mode',
          'FlashSize': 'Flash Size',
          'LwIPVariant': 'lwIP Variant',
          'ResetMethod': 'Reset Method',
          'UploadSpeed': 'Upload Speed'}}
````

d-a-v, 2017-10-12:
Thanks :) Good
We (at least I) may then think of the mix of the two.
Indeed I chose the (too) hard way, and it's going to be long. My idea was to have a small script, but I just realized that there is no such thing like #define in python :] 
There are blocks that are easily made from scratch (like the serial stuff) and common to every boards,
and other that are huge (like FlashSize) but not for all of them.
About FlashSize, or debug options, we could divide these blocks into smaller one and merge the appropriate ones into the appropriate boards.
I'm new to python and I don't know whether it is syntactically easy to merge dicts.
(time to get some sleep)

davisonja, 2017-10-12:
Merging dicts is easy enough, though our ultimate destination is a text file, so we don't need to merge things in python, just reference them from our definition structure.

I started with thoughts of a hierarchical or componentised system which would allow us to build a board from common attributes, or use inheritance and the like. My end conclusion is that path is overly and unnecessarily complex as we're really just dealing with a bunch of strings.

Though I can see merit in adopting that sort of thing for the flash size - various boards may be available in various sizes, so splitting that out of generic (which effectively has every flash size) would be beneficial. It would also neatly pull out the flash settings which I need to add the linker-file-generation aspect of this for #905.

Debug options need to be included for all boards as they're compiler-directives - they are a feature of the software build, not the target hardware.

davisonja, 2017-10-12:
I'm also curious what you wanted #define for..?

d-a-v, 2017-10-12:
We could have some predefined dicts (like FlashSize.4M3M.*), and include only those needed in relevant board's dicts. With #define (or sed/m4) it is easy. I'm confident we can find an elegant way of doing this in pure python.

davisonja, 2017-10-12:
Tweaking the layout slightly, this is the sort of solution I've used in the past (staying with plain text and `dict`s rather than full classes):
Define a basic structure with a dict that has labels (which are referred to elsewhere) as keys, and each value is the dict of data. In this case, each data-dict has keys

 - 'include' - a list of other entries to merge into this one
- 'lines' - raw config lines
- 'remove_lines' - a list of keys to remove from the set of lines before output
 - 'flash_menu' - a list of keys in the `FlashMenuConfig` to include

The entries in `Boards` can include each other (so a pre-process step for each entry would build a set of others to include and prevent endless recursion) allowing layering, and it separates out the flash info so it's easily reused in linker-file generation. Does it allow what you need?

````python

FlashMenuConfig = {
  '1M0' : {
      'lines': {
          'menu.FlashSize.1M0': '1M (no SPIFFS)',
          'menu.FlashSize.1M0.build.flash_ld': 'eagle.flash.1m0.ld',
          'menu.FlashSize.1M0.build.flash_size': '1M',
          'menu.FlashSize.1M0.upload.maximum_size': '1023984',
    }
  },
  '1M28' : {
      'lines': {
          'menu.FlashSize.1M128': '1M (128K SPIFFS)',
          'menu.FlashSize.1M128.build.flash_ld': 'eagle.flash.1m128.ld',
          'menu.FlashSize.1M128.build.flash_size': '1M',
          'menu.FlashSize.1M128.build.spiffs_blocksize': '4096',
          'menu.FlashSize.1M128.build.spiffs_end': '0xFB000',
          'menu.FlashSize.1M128.build.spiffs_start': '0xDB000',
          'menu.FlashSize.1M128.upload.maximum_size': '892912',
      }
  },
  '4M1M': {
      'lines': {
          'menu.FlashSize.4M1M': '4M (1M SPIFFS)',
          'menu.FlashSize.4M1M.build.flash_size': '4M',
          'menu.FlashSize.4M1M.build.flash_ld': 'eagle.flash.4m1m.ld',
          'menu.FlashSize.4M1M.build.spiffs_start': '0x300000',
          'menu.FlashSize.4M1M.build.spiffs_end': '0x3FB000',
          'menu.FlashSize.4M1M.build.spiffs_blocksize': '8192',
          'menu.FlashSize.4M1M.build.spiffs_pagesize': '256',
      }
  },
  '4M3M': {
      'lines': {
          'menu.FlashSize.4M3M': '4M (3M SPIFFS)',
          'menu.FlashSize.4M3M.build.flash_size': '4M',
          'menu.FlashSize.4M3M.build.flash_ld': 'eagle.flash.4m.ld',
          'menu.FlashSize.4M3M.build.spiffs_start': '0x100000',
          'menu.FlashSize.4M3M.build.spiffs_end': '0x3FB000',
          'menu.FlashSize.4M3M.build.spiffs_blocksize': '8192',
          'menu.FlashSize.4M3M.build.spiffs_pagesize': '256',
      },
  },
}

Boards = {
    'generic' = {
        'include': [],
        'lines' : {
            'name': 'Generic ESP8266 Module',
            'build.board': 'ESP8266_ESP01',
            'build.core': 'esp8266',
            'build.debug_level': '',
            'build.debug_port': '',
            'build.f_cpu': '80000000L',
            'build.flash_mode': 'qio',
            'menu.DebugLevel.all_____': 'All',
            'menu.DebugLevel.all_____.build.debug_level': '-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE       -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM',
            'menu.FlashFreq.40': '40MHz',
            'menu.FlashFreq.40.build.flash_freq': '40',
            'menu.FlashFreq.80': '80MHz',
            'menu.FlashFreq.80.build.flash_freq': '80',
            'menu.FlashMode.dio': 'DIO',
            'menu.FlashMode.dio.build.flash_mode': 'dio',
            'menu.FlashMode.dout': 'DOUT',
            'menu.FlashMode.dout.build.flash_mode': 'dout',
            'menu.FlashMode.qio': 'QIO',
            'menu.FlashMode.qio.build.flash_mode': 'qio',
            'menu.FlashMode.qout': 'QOUT',
            'menu.FlashMode.qout.build.flash_mode': 'qout',
        },
        'remove_lines': [],
        'flash_menu': ['1M0', '1M28'],
    },
    'arduino-esp8266': {
        'include': ['generic'],
        'lines': {
            'name': 'Arduino',
            'build.board': 'ESP8266_ARDUINO',
            'build.flash_freq': '40',
            'build.flash_size': '4M',
            'menu.BoardModel.primo': 'Primo',
            'menu.BoardModel.primo.build.board': 'ESP8266_ARDUINO_PRIMO',
            'menu.BoardModel.primo.build.extra_flags': '-DF_CRYSTAL=40000000',
            'menu.BoardModel.primo.build.variant': 'arduino_spi',
            'menu.BoardModel.starottodeved': 'Star OTTO',
            'menu.BoardModel.starottodeved.build.board': 'ESP8266_ARDUINO_STAR_OTTO',
            'menu.BoardModel.starottodeved.build.extra_flags': '-DF_CRYSTAL=40000000',
            'menu.BoardModel.starottodeved.build.variant': 'arduino_uart',
            'menu.BoardModel.unowifideved': 'Uno WiFi',
            'menu.BoardModel.unowifideved.build.board': 'ESP8266_ARDUINO_UNOWIFI',
            'menu.BoardModel.unowifideved.build.extra_flags': '-DF_CRYSTAL=40000000',
            'menu.BoardModel.unowifideved.build.variant': 'arduino_uart',
            'menu.FlashSize.4M3M.build.spiffs_pagesize': '256',
            'menu.UploadSpeed.19200': '19200',
            'menu.UploadSpeed.19200.upload.speed': '19200',
            'menu.UploadSpeed.230400': '230400',
            'menu.UploadSpeed.460800': '460800',
            'name': 'Arduino',
            'upload.maximum_size': '1044464',
            'upload.speed': '9600'
        },
        'remove_lines': [
            'menu.FlashFreq.40',
            'menu.FlashFreq.40.build.flash_freq',
            'menu.FlashFreq.80',
            'menu.FlashFreq.80.build.flash_freq',
            'menu.FlashMode.dio',
            'menu.FlashMode.dio.build.flash_mode',
            'menu.FlashMode.dout',
            'menu.FlashMode.dout.build.flash_mode',
            'menu.FlashMode.qio',
            'menu.FlashMode.qio.build.flash_mode',
            'menu.FlashMode.qout',
            'menu.FlashMode.qout.build.flash_mode',
        ],
        'flash_menu': ['4M1M', '4M3M'],
    },
}
````

igrr, 2017-10-12:
Perhaps you can define 'generic' as some hidden base configuration, plus
all the flash settings. Other boards can also reference the base
configuration.

In that case you won't even need to have a 'remove' action.

d-a-v, 2017-10-12:
I am currently progressing in the way you both describe. I am not far from facing forgotten details.

davisonja, 2017-10-12:
I started with a separate generic and decided I (personally :) liked the consistency of them all being the same. The alternative approach for a similar effect would be a 'hidden' attribute that meant (some) entries wouldn't output to boards.txt - if the boards.txt file becomes cluttered this may be useful regardless as it would allow uncommon or unused boards to be omitted. (I'm thinking in terms of the IDE menu becoming unwiedly)

davisonja, 2017-10-13:
> You're right. The current file is full of inconsistencies. The large
> number of entries is because of option menus. which are not available for
> every board only because (I think) it's painful to update. For example I
> guess it is desirable to consider that every board should have the debug
> menu available (instead of going to the generic board and trying to find
> out what the good options are, which is the last thing one wants to do when
> most of the work has to be done in the code).
>
> Its good you are willing to make your own version. I will continue mine
> and I don't care which one will be used, but we hopefully have one pushed
> in master very soon :)

It'll be a few days before I'll have a chance to do much actual work, I'm
quite happy to extend yours 😊

d-a-v, 2017-10-13:
Is there a major difference between generic and coredev ?

igrr, 2017-10-13:
Only the ability to choose LwIP version, as far as I know.

d-a-v, 2017-10-14:
1) Would debug options be visible for all boards ?
Same for the 3 lwip version (lwip, lwip_gcc, lwip_src) ?

2) We could also have two sets of boards: the regular one without these options, and the second one ("name"-devel) with these options.

3) or we could stick with the current configuration: debug and lwip options only visible for coredev, but the user has to know every hidden details of the boards (size, reset method) if debug is needed.

4) or the user could be asked to run the boards.txt generator with some options to enable debug options.

Anyway, you can try/review the script. It is currently in configuration 2): [link](https://raw.githubusercontent.com/d-a-v/esp82xx-arduino-boardgen/master/build-boards.txt.py)

igrr, 2017-10-14:
I think debug options should be available for all boards. LwIP choice can be left to "Generic" board only, then coredev can be removed.

d-a-v, 2017-10-14:
I forgot to mention cristal-freq and flash-freq menus.
So you would left them to generic only, with lwip.

igrr, 2017-10-14:
1. `cristalfreq` — i think it is spelled crystal, unless you mean the frequency of consumption of certain champagne :)

2. `flashmode_io/flashmode_out` — it looks a bit odd to me to bundle DIO with QIO and DOUT with QOUT. I would expect it to be QIO/QOUT and DIO/DOUT.

3. ESP8285 needs to have nodemcu reset method as well

4. I would suggest writing some code helpers to generate at least the FlashSize menu entries. For example, instead of

```
'512K': [
		[ '.menu.FlashSize.512K0', '512K (no SPIFFS)' ],
		[ '.menu.FlashSize.512K0.build.flash_size', '512K' ],
		[ '.menu.FlashSize.512K0.build.flash_ld', 'eagle.flash.512k0.ld' ],
		[ '.menu.FlashSize.512K0.upload.maximum_size', '499696' ],
		[ '.menu.FlashSize.512K64', '512K (64K SPIFFS)' ],
		[ '.menu.FlashSize.512K64.build.flash_size', '512K' ],
		[ '.menu.FlashSize.512K64.build.flash_ld', 'eagle.flash.512k64.ld' ],
		[ '.menu.FlashSize.512K64.build.spiffs_start', '0x6B000' ],
		[ '.menu.FlashSize.512K64.build.spiffs_end', '0x7B000' ],
		[ '.menu.FlashSize.512K64.build.spiffs_blocksize', '4096' ],
		[ '.menu.FlashSize.512K64.upload.maximum_size', '434160' ],
		[ '.menu.FlashSize.512K128', '512K (128K SPIFFS)' ],
		[ '.menu.FlashSize.512K128.build.flash_size', '512K' ],
		[ '.menu.FlashSize.512K128.build.flash_ld', 'eagle.flash.512k128.ld' ],
		[ '.menu.FlashSize.512K128.build.spiffs_start', '0x5B000' ],
		[ '.menu.FlashSize.512K128.build.spiffs_end', '0x7B000' ],
		[ '.menu.FlashSize.512K128.build.spiffs_blocksize', '4096' ],
		[ '.menu.FlashSize.512K128.upload.maximum_size', '368624' ],
		],
```

you would have

```
add_flash_size_macro('512K', '512K0', 'eagle.flash.512k0.ld', 'no SPIFFS', 499696, 0, 0, 0)
add_flash_size_macro('512K', '512K64', 'eagle.flash.512k64.ld', '64k SPIFFS', 434160, 0x6B000, 0x10000, 4096)
add_flash_size_macro('512K', '512K128', 'eagle.flash.512k128.ld', 128k SPIFFS', 368624, 0x5B000, 0x20000, 4096)
```

plus some code for `add_flash_size_macro` function.

Same may also make sense for DebugLevel menu, e.g.

```
add_debug_level_macro('SSL_MEM_', 'Core + SSL + TLS Mem', '-DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_TLS_MEM')
```

d-a-v, 2017-10-14:
1. I'm not even living in that county...
2. reason why I did so: 
```$ grep -i flashmode example-boards.txt 
menu.FlashMode=Flash Mode
espino.menu.FlashMode.dio=DIO
espino.menu.FlashMode.dio.build.flash_mode=dio
espino.menu.FlashMode.qio=QIO
espino.menu.FlashMode.qio.build.flash_mode=qio
wifinfo.menu.FlashMode.qio=QIO
wifinfo.menu.FlashMode.qio.build.flash_mode=qio
wifinfo.menu.FlashMode.dio=DIO
wifinfo.menu.FlashMode.dio.build.flash_mode=dio
```
(generic and coredev removed, they have the 4 options)

3. OK

4. I need a beer first

igrr, 2017-10-14:
re (2): The fact that they have both DIO and QIO listed means that QIO is likely supported, but the authors left QIO in for some unknown reason. DIO is only useful in two cases:

A. flash chip doesn't support QIO, or (more likely) uses a command to switch to QIO mode which ESP8266 ROM code does not know about.
B. GPIO9 and 10 are not connected to the flash chip (and can be used for other purposes).

Since QIO is supported, neither (A) nor (B) are the case here. Suggest leaving these boards with QIO only.

davisonja, 2017-10-14:
Definitely debug in all boards.

The initial plan was to build ld files from this script, so not pulling info from them. I'd opt for keeping this script as the source for both boards.txt and .ld files.

With suitable parameters on the command line we could offer options for more verbose boards.txt which also applied lwip to all boards.

My original post had flash snippets separate, which would be expanded into full entries (referenced by key name). So had an implicit macro by use of the dict.

d-a-v, 2017-10-15:
There are 511 combinations of 
```
[ 'CORE', 'SSL', 'TLS_MEM', 'WIFI', 'HTTP_CLIENT', 'HTTP_SERVER', 'HTTP_UPDATE', 'UPDATER', 'OTA' ]
```
Assuming there is no checkbox-like option in arduino's menus, there will a a single debug options enabled at a time plus an option for all all them at once.
Agreed ?

I guess this is sufficient for a first PR, before adding options for ldscript file generation _or_ reading ldscript files for getting lists of memory structure.

d-a-v, 2017-10-15:
I ended up with this list which is a mixture of full-combination(ssl, http) and the others (image attached)
[image]

davisonja, 2017-10-15:
Seems like a good starting point to me.

d-a-v, 2017-10-15:
#3722

d-a-v, 2018-01-08:
@davisonja We can close here, and you are welcome testing/reviewing/improving the script.

closed 2018-01-08 by d-a-v

nicocesar, 2018-04-04:
there is a common name for something called nodemcuv3 ... take a look into: https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/  ..is that on this list?

d-a-v, 2018-04-04:
The nodemcu we have are there: [board.txt.py](https://github.com/esp8266/Arduino/blob/b93cba104610a7db7021765a79a415e2546aa359/tools/boards.txt.py#L384).
I guess the v3 pinout is unchanged. The two we have only differ by the flashing mode.
If you try and find that the second one works with the nodemcuv3, then we can update our description.

nicocesar, 2018-04-05:
Yes they do.   The only issue that I'm having is that the LED_BUILTIN  is located on D4 and LOW will turn it on,, is that something that can be handled in  boards.txt or elsewhere?  I just want to keep my codebase identical for both boards read more here: https://arduino.stackexchange.com/questions/51493/which-board-should-i-use-with-esp8266-lolin-v3-in-arduino-ide

igrr, 2018-04-06:
> Then a new board and a new variant is needed for this LED_BUILTIN change.
>
> For the led, we can add a new define in all nodemcu's variant file to
> indicate whether it is lit on HIGH or LOW.
>
> @igrr <https://github.com/igrr> I read in #4613
> <https://github.com/esp8266/Arduino/pull/4613> that flash chips working
> in qio mode can work in dio mode too. Is it always true ?
> (in that case v0.9 and 1.0 nodemcu can have a shared single entry in the
> boards list)

Yes, they can, but with reduced performance.

I don't think this is important in Arduino use case, but in some more
demanding applications the difference between XIP in QIO and DIO can be
measurable.

d-a-v, 2018-04-06:
@igrr 
Thanks, good to know!

@nicocesar 
Please open a new issue for your enhancement request (nodemcuv3: led on D4, led inversed) so we can easily keep track of it.
(here is a closed issue with an unrelated subject)