esp8266/Arduino #3583: Exception thrown type 0

Jsolo1, 2017-09-07:
### Basic Infos

#### Hardware
Hardware:			ESP-12s
Core Version:      	2.3.0

### Description

The sketch runs for some minutes to an hour and then it throws an exception followed by a WDT reset and it doesn't reset properly, it gets stuck until a manual reset is performed.
I'm using a timer with the external c library "user_interface.c": os_timer_setfn(), os_timer_arm(), os_timer_disarm()
I'm also using some gpio interrupts.
The free heap memory remains constant and doesn't reduce with time.
The sketch works fine apart from this error.
I've tried the new core version 2.4.0rc1 but the same happens.
What's the problem?
Thanks

### Settings in IDE

Module:  			Nodemcu ESP8266 12e
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

[esp8266_AQS2_github.zip](https://github.com/esp8266/Arduino/files/1284669/esp8266_AQS2_github.zip)

Please see the files attached

### Debug Messages

Decoding 24 results
0x4010799c: interrupt_handler at C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 119
0x40107964: interrupt_handler at C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x40107e48: timer1_isr_handler at C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 40
0x40210a88: pm_get_sleep_type at ?? line ?
0x40105f36: spi_flash_read at ?? line ?
0x40107f58: pvPortZalloc at C:\Users\pdoyl\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x40210559: pm_set_sleep_time at ?? line ?
0x402109ee: pm_get_sleep_type at ?? line ?
0x40106988: igmp_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 217
0x40210a9b: pm_get_sleep_type at ?? line ?
0x40222171: ets_timer_handler_isr at ?? line ?
0x402221b6: ets_timer_handler_isr at ?? line ?

Exception (0):
epc1=0x4020223c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  4010799c 3ffe98c4 3ffeec68 00000020  
3ffffde0:  ffffffff 00000020 00000022 00000030  
3ffffdf0:  40000f83 00000030 00000020 00000022  
3ffffe00:  3fffc200 40107964 3fffc258 4000050c  
3ffffe10:  4000438f 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000007 000196fa 80000000  
3ffffe30:  20000000 3fff1980 80000000 203fc260  
3ffffe40:  00000000 3fffc6fc 3ffeec68 3fff1984  
3ffffe50:  00000094 003fc260 60000600 00000030  
3ffffe60:  00000009 00000000 0000002f 00001642  
3ffffe70:  00000042 3fffc6fc 3ffeec68 3ffedd30  
3ffffe80:  3ffeec90 9a026d62 60000600 00000030  
3ffffe90:  00000000 00000000 00000020 40107e48  
3ffffea0:  4000050c 3ffeec68 3fffc230 4000050c  
3ffffeb0:  40000f68 00000030 0000001c ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000013 40210a88 3ffeec68 00000001  
3ffffee0:  ffffffff 3ffe98c4 3ffeec68 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffeecb8 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff170c 000002f4 003fc000  
3fffff20:  40105f36 3ffeec90 3ffedd30 40107f58  
3fffff30:  40210559 3ffedd30 3ffeec90 9a026d62  
3fffff40:  3fff170c 00001000 402109ee 00000008  
3fffff50:  40106988 00000000 40210a9b 3ffedde4  
3fffff60:  3ffeec90 2fa64381 3ffeec90 60000600  
3fffff70:  40222171 3ffedde4 3ffeec90 9a020db6  
3fffff80:  402221b6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffeeca0 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 0001d8de 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

igrr, 2017-09-07:
It is likely that your GPIO interrupt handler needs to have `ICACHE_RAM_ATTR` attribute.

devyte, 2017-09-07:
@Jsolo1 please try the above suggestion and report back.

Jsolo1, 2017-09-08:
After 24h of testing, I can confirm that did the trick, no exceptions thrown. Thanks
To avoid the "function not declared on this scope" error just declare the interrupt function before the setup or place a pre-definition of the interrupt function before the setup.

closed 2017-09-08 by Jsolo1