esp8266/Arduino #3584: ESP8266 + ArduinoIDE + time.h

MrD2, 2017-09-07:
I'm trying to convert Timedate from separated values Year, Month, Day, Hour, Minute, Seconds, DST
into epoch_ms.

For that, i read many suggestions in using **struct tm** and then from that use **mktime()** to convert into epoch. This is what i do

`#define CONV_MACRO_EPOCH_MS(data) ({struct tm t, zero, *pt = &t;\
                                      time_t t_of_day;\
                                      memset(&zero, 0, sizeof(struct tm));\
                                      pt->tm_year = data->clock.year - 1970;\
                                      pt->tm_mon  = data->clock.month -1;\
                                      pt->tm_mday = data->clock.day_of_month;\
                                      pt->tm_hour = data->clock.hour;\
                                      pt->tm_min  = data->clock.minute;\
                                      pt->tm_sec  = data->clock.second;\
                                      pt->tm_isdst= ( data->clock.clock_status != 0 ? 1 : 0 );\
                                      mktime(pt) - mktime(&zero);\
                                    })`

**---------------------------------------------------------------------------------------**
**Please note that i try to use a "point of origin (zero)" struct tm because if i dont, when i get mktime the time comes disallined by 80 years or something, originally (or by recomendation) my routine should be:**

`#define CONV_MACRO_EPOCH_MS(data) ({struct tm t, *pt = &t;\
                                      time_t t_of_day;\
                                      pt->tm_year = data->clock.year - 1970;\
                                      pt->tm_mon  = data->clock.month -1;\
                                      pt->tm_mday = data->clock.day_of_month;\
                                      pt->tm_hour = data->clock.hour;\
                                      pt->tm_min  = data->clock.minute;\
                                      pt->tm_sec  = data->clock.second;\
                                      pt->tm_isdst= ( data->clock.clock_status != 0 ? 1 : 0 );\
                                      mktime(pt);\
                                    })`
**---------------------------------------------------------------------------------------**

And then i call this in my application:

`time_t epoch_ms = CONV_MACRO_EPOCH_MS(_p_struct_containing_data);`

Although this kinda works i have an offset of 1 day and x hours, and my time received by clock is ok.

Printing This returns:
`#define CONV_MACRO_TIMESTAMP(data)   String(String(data->clock.year)+"-"+String(data->clock.month)+"-"+String(data->clock.day_of_month)+"T"+String(data->clock.hour)+":"+String(data->clock.minute)+":"+String(data->clock.second))`

`>>> "datetime": "2017-9-7T17:44:21",`

but the epoch with the same data gives me:
`"timestamp": 1504744053`

devyte, 2017-09-07:
@MrD2 the ESP doesn't have a RTC, at least not a real one. It just a sort of pseudo software RTC based on an internal counter. Therefore, timekeeping on it doesn't make much sense*.
The only way to get time is to either have an external RTC, such as the wemos RTC module, in which case you still have to set the time at some point, or use NTP to query time over the network.
In the latter case, I believe that there are libs that will try to keep time between between queries (i.e.: sync) so that you don't have to query every single time.

Having said all that, this is not the right place to ask for support for your particular project. This is an issue tracker, meant for tracking issues. Please ask at esp8266.com or stackoverflow.
Closing.

*There are ideas out there to get around this, e.g.: the NodeMCU firmware has an RTC software module that does calculations (rather complex ones) to try and keep track of time even across sleep/wakeup cycles. If memory serves, it has a potential drift of 0.5s per day, or 15s per month. Personally, I'd love to have that functionality as a C/C++ lib, but no ports have been done so far.

closed 2017-09-07 by devyte

Juppit, 2017-09-08:
On 07.09.2017 at 18:58 wrote Diogo Guerra:
 > I'm trying to convert Timedate from separated values Year, Month, 
Day, Hour, Minute, Seconds, DST
 > into epoch_ms.

Did you found 
https://github.com/esp8266/Arduino/blob/master/tests/device/test_time/test_time.ino 
?

I hope you will find there a way to solve your problem.