esp8266/Arduino #3585: sometimes, my wemos d1 mini just refuses to connect

emilevs, 2017-09-07:
Please fill the info fields, it helps to get you faster support ;)

### Basic Infos

#### Hardware
Hardware:			wemos d1 mini

### Description

Problem description
I am working on a big sketch for a datalogger, this dataloggers connects to wifi every so often and then disconnects to save power. The sketch usually works but sometimes it just randomly refuses to connect. the network is working because I can connect to wifi. Usually this just causes me to get frustrated and stop programming and then the next day when I start programming again everything just works again. I have no idea what causes this, it is really frustrating. (restarting my router causes the esp to connect once and then I have the same issue again)
### Settings in IDE

### Sketch

while(WiFi.status() != WL_CONNECTED){
          Serial.println("WiFi not connected!");
          delay(1000);
          Serial.println(WiFi.status());
  }
    if(WiFi.status() == WL_CONNECTED){
      Serial.println("connected");
}

this is the piece of code that I use to connect to the wifi network (the entire sketch is gigantic)
```

### output:

WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected!
6
WiFi not connected
messages here
```

devyte, 2017-09-07:
@emilevs 
Please edit and post a complete minimal working sketch.
Please fill out the other fields in the issue template, they are there for a reason (start a new issue to see the template again, then cancel it).
Please make use of the wifi event handlers to see what the wifi connection is doing. Implement a message in all of the callbacks. Post the resulting messages here.

RudyFiero, 2017-09-08:
Have you tried a longer delay than 1?

devyte, 2017-09-08:
@emilevs I understand about a gigantic sketch, but please reduce it to [MCVE](https://stackoverflow.com/help/mcve). It is always possible that something previous could cause odd behavior.
This is necessary to understand both the setup as well as the loop, as they relate to your issue.

suculent, 2017-09-08:
Or provide the sketch, as because it’s gigantic, there’s probably issue somewhere else.

devyte, 2018-02-07:
Closing due to lack of feedback.

closed 2018-02-07 by devyte