esp8266/Arduino #3587: Problems compiling blank sketch while including ESP8266WiFi.h

PilotC150, 2017-09-08:
### Basic Infos

#### Hardware
Hardware:			ESP-12E

### Description

I can't get a blank sketch to build if I include ESP8266WiFi.h.  I'm using the latest version of Arduino IDE and latest version of the ESP8266 in the Board Manager (2.3.0).

This is the error message I get:

C:\Users\Chris\Documents\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp: In member function void ESP8266WiFiClass::beginSmartConfig():

C:\Users\Chris\Documents\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp:393:76: error: cannot convert sc_type to sc_callback_t {aka void (*)(sc_status, void*)} for argument 1 to bool smartconfig_start(sc_callback_t, ...)

     smartconfig_start(SC_TYPE_ESPTOUCH, &ESP8266WiFiClass::_smartConfigDone);

                                                                            ^

C:\Users\Chris\Documents\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp: In member function bool ESP8266WiFiClass::smartConfigDone():

C:\Users\Chris\Documents\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp:409:35: error: smartconfig_get_status was not declared in this scope

     return smartconfig_get_status() == SC_STATUS_LINK_OVER;

                                   ^

Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\Chris\Documents\Arduino\libraries\ESP8266WiFi
 Not used: C:\Users\Chris\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266WiFi.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

devyte, 2017-09-08:
I used 2.3.0 myself with ESP826Wifi, and it worked for me. Please try uninstalling the Arduino IDE, delete your entire Arduino and Aruino15 directories (back them up first, in case you have projects in them!!!!!), and reinstall from scratch.
While you are at it, you may want to switch to latest git (see readthedocs), the 2.3.0 is rather old.

devyte, 2017-09-08:
I will leave this open only because I haven't tested under Windows since Windows 7. Please report back if it works.

PilotC150, 2017-09-08:
Thanks for the quick feedback.  I haven't had a chance to totally clean the IDE off my personal laptop and reinstall from scratch, but I just did install the Arduino IDE (and added the ESP8266 boards) on my work laptop, which has never had the IDE installed on it before, and a blank sketch that includes ESP8266WiFi.h compiled just fine.  Clearly there's something wrong with the installation on my personal laptop.

Thanks again for the help.

closed 2017-09-08 by PilotC150