esp8266/Arduino #3588: Wemos d1 r2 losing sketch after power cycle

callmeharry3, 2017-09-09:
### Basic Infos
Wemos d1 r2 losing sketch after power cycle

#### Hardware
Hardware:			Wemos d1 r2
Core Version:      	2.1.0

### Description

I am trying to get a basic button sketch to work on my wemos d1. However, I seem to lose my sketch after every power cycle. It doen't matter if I connect my wiring to the board, or try to blink the onboard LED with nothing attached.

My button wiring works after I flash the sketch to the wemos, however if I disconnect and connect the usb power I lose the sketch. Also, after flashing a sketch to the board the board disconnects itself from the COM port on my pc

### Settings in IDE

Module:  			wemos d1 r2
Flash Size: 		4MB/
CPU Frequency:	80Mhz/160 mhz
Flash Mode:		no idea
Flash Frequency:	no idea
Upload Using:		SERIAL
Reset Method:		no idea

### Sketch

// constants won't change. They're used here to
// set pin numbers:
const int D5buttonPin = 14;     // the number of the pushbutton pin
const int D0ledPin =  16;      // the number of the LED pin

// variables will change:
int buttonState = 0;         // variable for reading the pushbutton status

void setup() {
  // initialize the LED pin as an output:
  pinMode(D0ledPin, OUTPUT);
  // initialize the pushbutton pin as an input:
  pinMode(D5buttonPin, INPUT);
}

void loop() {
  // read the state of the pushbutton value:
  buttonState = digitalRead(D5buttonPin);

  // check if the pushbutton is pressed.
  // if it is, the buttonState is HIGH:
  if (buttonState == HIGH) {
    // turn LED on:
    digitalWrite(D0ledPin, HIGH);
  } else {
    // turn LED off:
    digitalWrite(D0ledPin, LOW);
  }
}

[image]

(I have a regular D1 instead of a mini, but somehow fritzing doesn't have a regular D1 to select. You get the idea though)

I am using the arduino environment (1.8.1) to program and flash the board.

I think I have to put my board in a different mode after flashing the sketch to it. It has something to do with this:

GPIO15 | GPIO0 | GPIO2 | Mode
0V         | 0V       | 3.3V    | Uart Bootloader
0V         | 3.3V    | 3.3V    | Boot sketch
3.3V      | x          | x         | SDIO mode (not used for Arduino)

But which mode do I have to put the board in after flashing a sketch to it? And more importantly, how do I do this? I tried connecting GPIO 0 to the 3.3v out pin, however this just switches off the board. One would say this can't be a shortcircuit, because GPIO0 has a 10k pull up resistor.

Or do I have to ook elsewhere? Any help would be greatly appreciated :)

tttapa, 2017-09-09:
Sounds like a (hardware?) problem with your board.
This issue section is meant for problems with the ESP8266 Arduino Core. Please use an online forum instead. For example, [forum.arduino.cc](http://forum.arduino.cc/) or [esp8266.com](http://www.esp8266.com/).

Pieter

callmeharry3, 2017-09-09:
Allright. I found some other (older) topics here outlining the same sort of problems with esp8266, but none of them were answered in a way that helped me.

I'll try a different forum.

closed 2017-09-09 by callmeharry3