esp8266/Arduino #3589: print() throws exception on 2.4.0-rc1 with Adafruit GFX libraries

xsrf, 2017-09-09:
### Basic Infos

Hey,
I've also posted this issue [here](https://github.com/adafruit/Adafruit-GFX-Library/issues/133), but I actually think it's a problem with the [Print](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Print.h#L68) implementation in core 2.4.0-rc1, so here it is again.

Since I upgraded from ESP8266 core 2.3.0 to 2.4.0-rc1 I have problems with the Adafruit GFX libraries, which might have to do with the print() implementation of either Adafruit or the esp8266 core - I'm not sure.
The [adafruit print()](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.h#L12) function of any library that is based on Adafruit_GFX throws an exception.

### Hardware
Hardware: WeMos D1 Mini / ESP12
ESP8266 Core Version: 2.4.0-rc1
Adafuit GFX Version: 1.2.2

### Description

The print() function of any library that is based on Adafruit_GFX throws an exception when Serial.begin() was used before.

### Settings in IDE

Module:  			WeMos D1 R2 & Mini
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload Using:		Serial/USB

### Sketch

```cpp

#include "Adafruit_GFX.h"
#include "Adafruit_NeoMatrix.h"

Adafruit_NeoMatrix disp = Adafruit_NeoMatrix(128, 64);

void setup()
{
	Serial.begin(115200);
	Serial.println("Boot");
	disp.drawPixel(1, 1, 1);
	Serial.println("drawPixel() OK");
	disp.print("Test"); // throws exception on esp8266 2.4.0-rc1 / works on esp8266 2.3.0
	Serial.println("print() OK");
}

void loop()
{

}

/*
Boot
drawPixel() OK

Exception (3):
epc1=0x40202043 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40233d24 depc=0x00000000

ctx: cont
sp: 3ffef830 end: 3ffefad0 offset: 01a0

>>>stack>>>
3ffef9d0:  feefeffe feefeffe feefeffe feefeffe
3ffef9e0:  00000000 00000000 00000000 00000000
3ffef9f0:  0000ffff feefeffe feefeffe feefeffe
3ffefa00:  00000005 feefeffe feefeffe 000001a4
3ffefa10:  00000008 feefeffe feefeffe 00000001
3ffefa20:  feefeffe feefeffe feefeffe 3ffe883c
3ffefa30:  3ffee8b8 00000000 3ffee8b8 40202320
3ffefa40:  00000001 00000001 3ffe8b89 40203330
3ffefa50:  40201b16 00000000 3ffe8838 40203330
3ffefa60:  3fffdad0 0000000e 3ffefb10 3ffeeaa8
3ffefa70:  3fffdad0 3ffee8b8 3ffee8b8 40202c41
3ffefa80:  3ffe8838 00000001 3ffefb10 40202c6c
3ffefa90:  0001c200 0000001c 3ffefb10 40202c54
3ffefaa0:  3fffdad0 3ffee8b8 3ffefb10 40201ef4
3ffefab0:  feefeffe 00000000 3ffeeaa0 40202960
3ffefac0:  feefeffe feefeffe 3ffeeab0 4010070c
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(3,6)
*/
```

The NeoMatrix.h is just an example, since Adafruit_GFX.h cannot be used directly.
Adding begin(), which was suggested on the other issue, results in the program locking up and getting a wdt reset. Also begin() is not even available on all Adafruit_GFX based libraries.

xsrf, 2017-09-09:
I was suspecting Serial having to do with this issue, but this code also doesn't work on 2.4 but works on 2.3.

```cpp
#include "Adafruit_GFX.h"
#include "Adafruit_NeoMatrix.h"

Adafruit_NeoMatrix disp = Adafruit_NeoMatrix(8, 8);

void setup()
{
	pinMode(2, OUTPUT);
	disp.print("Test"); // throws exception on esp8266 2.4.0-rc1 / works on esp8266 2.3.0
}

void loop()
{
	// just visual feedback
	delay(50);
	digitalWrite(2, 1 - digitalRead(2));
}
```

izanette, 2017-09-09:
I'm having the exact same problem, using the SSD1306:

`#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define OLED_RESET 13
Adafruit_SSD1306 disp(OLED_RESET);

#if (SSD1306_LCDHEIGHT != 32)
#error("Height incorrect, please fix Adafruit_SSD1306.h!");
#endif

void setup()   {                
  Serial.begin(115200);
  disp.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  disp.clearDisplay();
  Serial.println("Boot");
  disp.drawPixel(1, 1, 1);
  Serial.println("drawPixel() OK");
  disp.setTextSize(1);
  disp.setTextColor(WHITE);
  disp.setCursor(0, 0);
  disp.print("Test"); // throws exception on esp8266 2.4.0-rc1 / works on esp8266 2.3.0
  Serial.println("print() OK");
  disp.display();
}

void loop() {
}`

izanette, 2017-09-09:
The drawBitmap() method of the library also makes it crash:

```
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

Adafruit_SSD1306 disp(13);

static const unsigned char PROGMEM iot_bmp[] = { 0x87, 0x82, 0xBA, 0xAA, 0xBA };

void setup()   {                
  Serial.begin(115200);
  disp.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  disp.clearDisplay();
  Serial.println("Boot");
  disp.drawPixel(1, 1, 1);
  disp.display();
  Serial.println("drawPixel() OK");
  disp.drawBitmap(120, 0, iot_bmp, 8, 5, WHITE);  
  disp.display();
  Serial.println("drawBitmap() OK");
}

void loop() {
}
```

izanette, 2017-09-09:
Found the source of the problem:
It is the **PROGMEM** statement that is used in the declaration of the bitmaps.
If you add this code in the Adafruit_GFX.h, just after the #includes, it won't crash anymore:

```
#ifdef PROGMEM
#undef PROGMEM
#define PROGMEM
#endif
```

I'm not sure what this PROGMEM does. I'll check it out...

xsrf, 2017-09-09:
@izanette You should update your library. [See latest commit](https://github.com/adafruit/Adafruit-GFX-Library/commit/07f8f65e4072c422f0c1e0e0c2a59dbda3d87bcb)
Also my example doesn't use bitmaps.

izanette, 2017-09-09:
@xsrf , yes, I'm using GFX 1.2.2. 

> Also my example doesn't use bitmaps.

By using the print method, you are using bitmaps indirectly. This is because the write method of the GFX library uses [this font file](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/glcdfont.c) that has the PROGMEM statement.

igrr, 2017-09-12:
Closing as the issue seems to be fixed in Adafruit-GFX-Library.

closed 2017-09-12 by igrr

izanette, 2017-09-12:
@xsrf, have you tried my changes? I was having the exact same behavior as you when using the print() method, but after doing these changes, they didn't occur anymore.

xsrf, 2017-09-12:
@igrr why do you assume the issue was fixed by adafruit? The issue still exists. As @izanette wrote, the issue appears when bitmaps / fonts / data are stored with PROGMEM statement. Removing PROGMEM resolves the issue, but that defeats the purpose of PROGMEM.
I'm not an experienced developer, but I think there is a problem with PROGMEM statement since 2.4.0 - not with the adafruit gfx library.

me-no-dev, 2017-09-12:
TBH this is not an issue at all. You need separate functions that are written specifically to deal with progmem data. We usually prepend the function name with ```_P```, like ```print_P(...)```. So if GFX does not support that, it's a design choice. You can submit an issue there and see if they will implement it. You could also implement it yourself and PR it to their repository, so others can benefit as well.

me-no-dev, 2017-09-12:
As for the exceptions: you can decode them to get more information with [this](https://github.com/me-no-dev/EspExceptionDecoder) tool.

xsrf, 2017-09-12:
@me-no-dev I'm sorry... If it's obvious a problem with the gfx lib, that's fine... But why does the same code work fine with 2.3.0 and fails with 2.4.0? Shouldn't it also fail with 2.3.0 then?

izanette, 2017-09-13:
I did more tests tonight and saw that if I revert the pgmspace.h to [this version](https://github.com/esp8266/Arduino/blob/3bfd5d51b470efc4979f46c7b3565d91ab4fc084/cores/esp8266/pgmspace.h), that has the pgm_read_byte() written in C instead of assembly, the GFX library works fine.

The crash happens at the method drawChar(), [in this file](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.cpp) at line 725, where it uses the pgm_read_byte.

Strangely, if I copy the code of the drawChar() method and adapt it to run without the other methods of the library, the crash does **not** happen:

```
#include "glcdfont.c"

void setup() {                
  Serial.begin(115200);
  drawStr("Test");
  Serial.println("print() OK");
}

void drawStr(const char* s) {
  int i = 0;
  char c = s[i];
  while(c != 0) {
    drawChar(c);
    c = s[++i];
  }
}

void drawChar(unsigned char c) {
  for(int8_t i=0; i<5; i++ ) { // Char bitmap = 5 columns
    uint8_t line = pgm_read_byte(&font[c * 5 + i]);
    for(int8_t j=0; j<8; j++, line <<= 1) {
      if(line & 0x80)
        Serial.print("#");
      else
        Serial.print(" ");
    }
    Serial.println();
  }
}

void loop() {
}
```

igrr, 2017-09-13:
@xsrf Sorry for not being entirely clear. There were two related issues, on in Adafruit's library and another one here. The issue here was that in 2.3.0, `pgm_read_byte` was defined as an inline function, but Adafruit lib did a `#ifdef pgm_read_byte` feature check, which didn't work when `pgm_read_byte` was not a macro. This issue has been fixed, and now `pgm_read_byte` is a macro again. If you pull the latest git version, your issue should be fixed.

xsrf, 2017-09-26:
Sorry for the late reply. Can also confirm that it works with the latest git version.