esp8266/Arduino #3591: Memory usage, weird behaviour, Wifi stopping working

mamama1, 2017-09-10:
#### Hardware
Hardware:			RobotDyn NodeMCU Style board (12E?)
Core Version:      	latest from board manager

### Description

Hi

I've been working on quite a big project lately. Basically a ESP8266 based 8 channel WiFi dimmer using a SX1509 driver which has an IR receiver, a TCP interface for raw byte command data and a Webserver using the ESP8266 Webserver Library for basic configuration like Wifi mode and so on.

Basically everything worked so far and I proceeded bugfixing, adding new features and polishing.

For the webserver, I put everything together in a char buffer variable which is created within the respective function which handles the http request and then send it out. for this, i have about 4 or 5 2000-3000 byte function-local char buffers. one global 3000 or 4000 byte buffer does not work at all. the esp keeps crashing or wifi is not connecting/pingable if i try that.

is there something with heap/stack?

At some point, when implementing features and stuff, the esp started acting weirdly it connects to wifi, one or two pings work and thats it. the program works but wifi is unreachable and constantly keeps reconnecting.
depending on the change i also get watchdog resets, stack, traces, everything. it didn't matter what i changed, every change produced random errors from some point on but the "previous version" of the code always worked flawlessly so i don't think that it is a hardware issue.
wifi even stopped working when i added one more word to a html output (buffer was large enough, no writing to unallocated memory).

i suspect that memory is an issue. i understand that there is a heap and a stack. how can i monitor those and check if they're running out of space?
when compiling, it shows that about 50 percent of the memory is used.

What could possibly go wrong here and what should I do?

Thanks!

### Settings in IDE

Module:  			ESP12E
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			whats dat
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		whats dat

mamama1, 2017-09-10:
i have similar behaviour as https://github.com/esp8266/Arduino/issues/3540 except that i don't have a big cap because i'm powering my board directly via usb.
using wifi.status and wifi.getmode repeadetly causes wifi completely to stop working.

mamama1, 2017-09-11:
Okay, I increased CONT_STACKSIZE in cont.h to 8192 and then everything worked like a charm again.
then I added some "strcat" commands in one routine, enlarged the char buffer which is used by those strcat commands and then it started acting weird again. further increasing CONT_STACKSIZE didn't change anything.

now, some float calculations sometimes fail but most importantly, when I tried to output the results of the float calculations via Serial.println(myFloatvariable), WiFi again stopped working (i.e. only sporadically responds to pings, loses connection and reconnects frequently).
as soon as I remove the Serial.println commands, everything works again, except that the float calculations are wrong under certain circumstances.

Any idea, anyone? I know, float calcs and strcat aren't the most beloved tasks of a cpu like the ESP8266 but if it has (or should have) the power and I'm in no rush, why not?

mamama1, 2017-09-11:
using core 2.4.0 rc1 wifi works even with the serial.println lines, i can even open the webpage but as soon as I send commands to my second tcp server instance (i have one ESP8266WebServerrunning + one WiFiServer), wifi disconnects, therefore pinging doesn't work anymore and it doesn't recover, as with core 2.3.0 where it frequently dis- and reconnects.

very strange...

mamama1, 2017-09-11:
ok ditch that, I had one buffer too small and that corrupted memory. darn C, that is...

devyte, 2017-09-12:
@mamama1 I think you already know how to deal with the problem: reduce stack usage and be careful with heap.
Be aware, using C-techiniques tend to be resource hogs, but they are far more fragile vs. C++ techiniques. Personally, I don't use C-style buffers anymore unless I have to. Instead, I use std containers, even on the ESP. Using C techniques requires carelessly usually leads to odd crashes and mem corruption. Using C++ techniques usually leads to inefficient code (granted, on the ESP being inefficient about mem usage can also lead to odd crashes). Overall, working extra to make your code more efficient is usually better than working to make it less buggy.
Having said that, this is not the right place to discuss these things. This is an issue tracker, meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow for assistance.
Closing as off-topic.

closed 2017-09-12 by devyte