esp8266/Arduino #3592: When Calling the  Airkiss Function,print "undefined reference to"

hangzxh, 2017-09-10:
Basic Infos

How to call airkiss function? for example "airkiss_version".When compiled, Arduino IDE print"undefined reference to `airkiss_version'
Hardware
Hardware:	ESP-12E
Core Version: 2.3.0

Description

How to call airkiss function? for example "airkiss_version".When compiled, Arduino IDE print"undefined reference to `airkiss_version'.
I have found the airkiss.h in "Arduino\hardware\espressif\esp8266\tools\sdk\include"
also the libairkiss.a in "Arduino\hardware\espressif\esp8266\tools\sdk\lib"

Settings in IDE

### Sketch

/*
 * NativeSdk by Simon Peter
 * Access functionality from the Espressif ESP8266 SDK
 * This example code is in the public domain
 * 
 * This is for advanced users.
 * Note that this makes your code dependent on the ESP8266, which is generally
 * a bad idea. So you should try to use esp8266/Arduino functionality
 * where possible instead, in order to abstract away the hardware dependency.
 */

// Expose Espressif SDK functionality - wrapped in ifdef so that it still
// compiles on other platforms
#ifdef ESP8266
extern "C" {
#include "user_interface.h"
#include "airkiss.h"
}

#endif

void setup() {
  Serial.begin(115200);
}

void loop() {
  // Call Espressif SDK functionality - wrapped in ifdef so that it still
  // compiles on other platforms
#ifdef ESP8266
  Serial.print("wifi_station_get_hostname: ");
  Serial.println(wifi_station_get_hostname());
  Serial.println(airkiss_version());
#endif
  delay(1000);
}

Debug Messages

Archiving built core (caching) in: C:\Users\Z00135~1.CHI\AppData\Local\Temp\arduino_cache_581525\core\core_espressif_esp8266_nodemcuv2_CpuFrequency_160,UploadSpeed_115200,FlashSize_4M1M_9e863c8058aaa52e06769f0454a69ae4.a
sketch\CallSDKFunctions.ino.cpp.o: In function `setup':

C:\Users\z00135220.CHINA\Documents\Arduino\CallSDKFunctions/CallSDKFunctions.ino:25: undefined reference to `airkiss_version'

sketch\CallSDKFunctions.ino.cpp.o: In function `loop':

C:\Users\z00135220.CHINA\Documents\Arduino\CallSDKFunctions/CallSDKFunctions.ino:31: undefined reference to `airkiss_version'

collect2.exe: error: ld returned 1 exit status

exit status 1

hangzxh, 2017-09-10:
I had found the problem .when linking ,Arduoino esp8266 just links the smartconfig lib ,not the airkiss lib.

Linking everything together...
"C:\Users\hangzxh\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -Wall -Wextra -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:\Users\hangzxh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lib" "-LC:\Users\hangzxh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/ld" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:\Users\hangzxh\AppData\Local\Temp\arduino_build_342081/CallSDKFunctions.ino.elf" -Wl,--start-group "C:\Users\hangzxh\AppData\Local\Temp\arduino_build_342081\sketch\CallSDKFunctions.ino.cpp.o" "C:\Users\hangzxh\AppData\Local\Temp\arduino_build_342081/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-LC:\Users\hangzxh\AppData\Local\Temp\arduino_build_342081"
"C:\Users\hangzxh\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe" -eo "C:\Users\hangzxh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/bootloaders/eboot/eboot.elf" -bo "C:\Users\hangzxh\AppData\Local\Temp\arduino_build_342081/CallSDKFunctions.ino.bin" -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "C:\Users\hangzxh\AppData\Local\Temp\arduino_build_342081/CallSDKFunctions.ino.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec

5chufti, 2017-09-12:
maybe it would help changing the topic to reflect the problem with linking process, so it gets picked up (attention) by relevant person.

closed 2018-01-02 by igrr