esp8266/Arduino #3594: OTA stopped working

AdamMiltonBarker, 2017-09-11:
### Basic Infos

#### Hardware
Hardware:			ESP-12 F
Core Version:      	2.1.0-rc2

### Description

Hi guys, I have not worked on ESP for a while, came back to do a new project and noticed that OTA updates stopped working in May, has anything changed that would make this happens, seems strange. The SSL has been updated since then so I have added the new fingerprint but it is still returning with:

scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

```
connected with TB_LNC_2_4, channel 1
dhcp client start...
ip:192.168.1.25,mask:255.255.255.0,gw:192.168.1.2
Successfully connected to WiFi
please start sntp first !
HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused
please start sntp first !

```

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

Line in question:

`t_httpUpdate_return ret = ESPhttpUpdate.update("https://iot.techbubbletechnologies.com/API/Rest/IntelliLan/UpdateControl/updater.php","","‎‎2e 11 c5 de 03 38 8c 7a 10 b7 7a b8 93 45 75 01 e1 6d 41 2c");`

5chufti, 2017-09-12:
you realise that you are working with an esp8266 core that's more than one year out of date?
2.1.0.-rc2 was released jan.'16

davisonja, 2017-09-12:
I periodically wonder if we should be pulling out references to such
varients so people can't pick them (I am assuming in this case it was a
recent user choice, but historically we've had people pick older versions)

igrr, 2017-09-12:
The problem with that (removing reference to older version from the package
index) is that Arduino does not handle such situation in a reasonable
manner. It decides that you don't have any version of the package installed
anymore, and when you install a new version, it fails to compile the
sketch, because there are now two versions of the core installed.

In this particular case i think that the issue reporter has simply not
updated one of the fields of the issue template.

davisonja, 2017-09-12:
Oh, so that will break existing installs? 😕 that's unfortunate.

AdamMiltonBarker, 2017-09-12:
@5chufti I am on a new dev terminal and installed the latest version to Arduino IDE. I added the following to preferences:

http://arduino.esp8266.com/stable/package_esp8266com_index.json

And then installed via IDE.

The only thing I can think of is the SSL cert which was renewed in May, as mentioned above I updated the fingerprint in this new device but it is not working. This has led to an issue I was unaware of, if it is the fingerprint, I have two devices that are part of circuits that are now inaccessible, does not solve the current issue though, with the new fingerprint I cannot connect to the server.

igrr, 2017-09-12:
FWIW, with the latest git version, TLS handshake with iot.techbubbletechnologies.com is successful. The server returns HTTP code 403, so OTA does not happen, but at least the crash during TLS handshake no longer happens.

AdamMiltonBarker, 2017-09-12:
OK cool thanks I will clone the latest, my Arduino folder seems to be stored in Documents but has no hardware folder, is this the right location and I should just create the hardware folder ?

AdamMiltonBarker, 2017-09-12:
I can now see your attempts in my updater logs which was not happening since May, the latest git version seems to be the answer, thanks.

AdamMiltonBarker, 2017-09-12:
OK so Arduino hardware seems to be stored in C://WindowsApps now

AdamMiltonBarker, 2017-09-12:
To solve the above issue follow this, works for me:

http://www.intowindows.com/how-to-access-windowsapps-folder-in-windows-10/

AdamMiltonBarker, 2017-09-12:
OK @igrr  so I get no output after the IP address using your code, I had to change the location of hardware folder in preferences, as even with the fix above you cannot create folders in the location arduino is installed to, you cannot even make a shortcut to Arduino IDE, nice one Windows ;)

AdamMiltonBarker, 2017-09-12:
Ok it is not working, you are unable to modify folders and files in the above location, even with admin access. I change the preferences to point to Documents folder but it got overwritten.

AdamMiltonBarker, 2017-09-12:
OK @igrr it is now connecting properly, but I still get the following:

```
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @300000, size=fb000, block=2000, page=100
SPIFFSImpl: mount rc=0
SPIFFS_close: fd=1
SPIFFS_close: fd=1
sta config unchangedscandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with TB_LNC_2_4, channel 1
dhcp client start...
ip:192.168.1.24,mask:255.255.255.0,gw:192.168.1.2
Successfully connected to WiFi
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr 117 117 0
:wrc 117 117 0
:sent 117
:rn 1440
:rd 5, 1440, 0
:rdi 1440, 5
:rd 80, 1440, 5
:rdi 1435, 80
State:	receiving Server Hello (2)
:rd 5, 1440, 85
:rdi 1355, 5
:rd 1350, 1440, 90
:rdi 1350, 1350
:c0 1350, 1440
:rn 1440
:rd 1440, 1440, 0:rch 1440, 1216
:rch 2656, 378

:rdi 1440, 1440
:c 1440, 1440, 3034
:rd 1585, 1594, 0
:rdi 1216, 1216
:c 1216, 1216, 1594
:rdi 378, 369
State:	receiving Certificate (11)
:rd 5, 378, 369
:rdi 9, 5
:rd 4, 378, 374
:rdi 4, 4
:c0 4, 378
State:	receiving Server Hello Done (14)
Fatal exception 29(StoreProhibitedCause):
epc1=0x4021b08e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000

Exception (29):
epc1=0x4021b08e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff1cd0 end: 3fff2330 offset: 01a0

>>>stack>>>
3fff1e70:  00820000 00000082 3fffa35c 40228d66  
3fff1e80:  00000214 3fff789c 3fffa35c 40228f60  
3fff1e90:  3fffbcac 00000214 3fffbcac 40100974  
3fff1ea0:  00000114 00007090 00000e12 00000e12  
3fff1eb0:  00000040 00000082 3fff51dc 40229279  
3fff1ec0:  00800000 00000080 00000000 00000041  
3fff1ed0:  00000040 0000003f 3fff897c 00000041  
3fff1ee0:  3fffb88c 3fff51dc 3fff897c 3fff7e04  
3fff1ef0:  3fffbdb0 ffffffff 3fff789c 40211fe0  
3fff1f00:  00000040 3fff789c 3fff51dc 3fff789c  
3fff1f10:  00000040 3fff789c 3fff51dc 40229db3  
3fff1f20:  3fff7964 0000003f 7351bb93 00000001  
3fff1f30:  3fff789c 3fff51bc 3fff51dc 00000001  
3fff1f40:  00000010 3fff789c 0000000f 4022a008  
3fff1f50:  3fff897c 3fff7bcc 0000000f 00000001  
3fff1f60:  00000001 3fff7bec 3fff5414 40228e0c  
3fff1f70:  00000100 3fff792c 3fff53df 00000000  
3fff1f80:  00000100 3fff792c 3fff53df 4022b189  
3fff1f90:  3fff1fc0 00000000 000000d0 00000030  
3fff1fa0:  82ac8351 3fff2010 3fff2010 00000004  
3fff1fb0:  3fff53d9 3fff52ec 3fff53d9 40225df4  
3fff1fc0:  a00d0303 22f75fd0 fa6fc978 d1d2ff91  
3fff1fd0:  fd881928 ed9d16b1 f87d1a4a d4dfef83  
3fff1fe0:  354b4b21 51631f77 004ecd28 8351f045  
3fff1ff0:  3fff53d9 00000004 3fff52ec 40226289  
3fff2000:  3fff2020 3fff2020 00000004 40223bc9  
3fff2010:  3ffeabfc 00000017 0000017a 3fff517c  
3fff2020:  3fff2040 3fff2058 40255314 00000000  
3fff2030:  00000017 00000090 00000064 40211fa0  
3fff2040:  76726553 48207265 6f6c6c65 00000004  
3fff2050:  3fff53d9 00000004 3fff52ec 00000004  
3fff2060:  3fff53d9 00000004 3fff52ec 40225bfc  
3fff2070:  5a764178 3fff50c0 00000000 0000003b  
3fff2080:  00000070 3fff53d9 3fff52ec 40225fd8  
3fff2090:  0000001e 3fff540c 00000200 401007ac  
3fff20a0:  3fff20f0 0000000e 00000010 00000000  
3fff20b0:  00000000 3fff5264 3fff52ec 01000000  
3fff20c0:  3fff20f0 3fff4ebc 3fff52ec 40225d64  
3fff20d0:  000001bb 3fff4ebc 3fff52a4 00001387  
3fff20e0:  000116aa 3fff4ebc 3fff4e8c 4020a07d  
3fff20f0:  3fff4e8c 00000000 3fff517c 402094a1  
3fff2100:  3fff517c 3fff2254 3fff1200 00000000  
3fff2110:  000001bb 3fff4e8c 3fff4f5c 00000000  
3fff2120:  000001bb 3fff4e8c 3fff4f5c 4020a3ed  
3fff2130:  3ffea218 c64b3025 3ffea218 c64b3025  
3fff2140:  3fff22dc 00000000 3fff2220 4020e194  
3fff2150:  00000000 00000000 3fff4e2c 40211367  
3fff2160:  00000028 00000000 3fff2220 4020e674  
3fff2170:  3fff4e38 00000000 3fff4e20 4021150d  
3fff2180:  3ffe9ee8 00000000 3fff2220 40211d4c  
3fff2190:  3ffe9e74 00000418 3fff2220 4020e043  
3fff21a0:  3fff21dc 00000000 3fff4c0c 3fff1220  
3fff21b0:  3fff22dc 3fff1004 3fff2220 4020e722  
3fff21c0:  3fff22dc 3fff1004 3fff2220 4020ed29  
3fff21d0:  00000000 00000000 00000000 3ffe9ee8  
3fff21e0:  00000000 00000000 00000459 4010020c  
3fff21f0:  00000000 3fff1004 3fff4e14 4010099c  
3fff2200:  40208bf9 40208bd5 3fff229c 3fff22d0  
3fff2210:  3fff22e8 3fff1004 3fff22dc 4020ef25  
3fff2220:  3fff4c84 3fff4e8c 3fff4f5c 0000001f  
3fff2230:  0000001e 350001bb 3f011f40 3fff374c  
3fff2240:  0000002f 0000002e 3fff4bf4 0000000f  
3fff2250:  00000005 3fff4f8c 0000011f 0000011e  
3fff2260:  3fff4c34 0000001f 00000013 3fff4c64  
3fff2270:  0000000f 00000000 3fff4e20 00000001  
3fff2280:  00000000 ffffffff 3fff2200 00000000  
3fff2290:  00000000 00000000 00000000 00000000  
3fff22a0:  00000000 00000000 3fff22d0 402113e6  
3fff22b0:  3ffe900c 0012699c 3fff1040 3fff1300  
3fff22c0:  3ffea5a0 3fff0ddc 3fff1004 402078dd  
3fff22d0:  3fff428c 0000004f 00000041 3fff372c  
3fff22e0:  0000000f 00000000 3fff4b84 0000005f  
3fff22f0:  00000054 3fff0ddc 3fff1240 402107d4  
3fff2300:  3fff0dd0 3fff0ddc 3fff1240 40207e2e  
3fff2310:  3fffdad0 00000000 3fff12f8 40211f6c  
3fff2320:  feefeffe feefeffe 3fff1310 40100a20  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,0)

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset

```
From logs:

FREE SPACE: 2809856
SKETCH SIZE: 335248
SDK: 2.0.0(656edbf)

AdamMiltonBarker, 2017-09-12:
Must be memory, your script works, that sucks

AdamMiltonBarker, 2017-09-13:
@igrr  please would it be possible for me to send you the script privately and you could maybe give some suggestions to optimization? The script is quite large, it includes MRC522, Paho MQTT, Spiffs, Webserver for storing the data to Spiffs, OTA update etc. Would be interested to see if there was any optimization that could be done to reduce it, I went through last night and stripped out as much as I could but could not get it to stop crashing.

igrr, 2017-09-13:
Sorry, i don't think i have bandwidth to offer such help now.

AdamMiltonBarker, 2018-01-07:
Sorry haven't worked on this for a while, I am still unable to get the original project working yet but have OTA working successfully on all other projects, will close this thanks for the help.

closed 2018-01-07 by AdamMiltonBarker