esp8266/Arduino #3595: Found errors in coding

OMADC, 2017-09-11:
Arduino: 1.6.6 (Mac OS X), Board: "LinkIt ONE"

Warning: platform.txt from core 'MediaTek ARM7 EJ-S (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
In file included from sketch/Arduino 2.app/Contents/Java/hardware/arduino/avr/bootloaders/caterina/Caterina.h:48:0,
                 from sketch/Arduino 2.app/Contents/Java/hardware/arduino/avr/bootloaders/caterina/Caterina.c:37:
sketch/Arduino 2.app/Contents/Java/hardware/arduino/avr/bootloaders/caterina/Descriptors.h:92:5: error: #error The selected AVR part is not currently supported by this bootloader.
    #error The selected AVR part is not currently supported by this bootloader.

     ^
exit status 1
Error compiling.
Warning: platform.txt from core 'MediaTek ARM7 EJ-S (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'MediaTek ARM7 EJ-S (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

OMADC, 2017-09-11:
The code was copied compatibly from an official website. 
However still don't know where the problem is.

devyte, 2017-09-11:
@OMADC you pretty much ignored the entire issue template. Among the few details you provided you mention the linkit one board. I took a quick look online, and that board is based on a mediatek soc.
The error messages mention a MediaTek board core. I think you are in the wrong repo, as this is for the ESP8266.
Closing.

closed 2017-09-11 by devyte