esp8266/Arduino #3596: No SoftwareSerial lib for v2.4?

JonSilver, 2017-09-11:
I note that SoftwareSerial has disappeared from core v2.4. If one needs to use SoftwareSerial for development, which library should now be used?

devyte, 2017-09-12:
@igrr I don't see SoftwareSerial under libraries in latest git. Is this intentional?

igrr, 2017-09-12:
It gets included when building a boards manager package here: https://github.com/esp8266/Arduino/blob/master/package/build_boards_manager_package.sh#L63-L68

I just ran this script locally and the SoftwareSerial library ended up being included in the package. Not sure why this didn't happen with 2.4.0-rc1 package though.

As a workaround, you can download SoftwareSerial library as a zip file and then install it into the IDE using "install library from zip file" feature.

Will troubleshoot this when building the package next time, thanks for the report.

JonSilver, 2017-09-12:
Thanks @igrr - good to know it's not an intentional omission 🙂
And while I'm here, thanks for all your great work on the ESP8266 Arduino Core.

devyte, 2017-11-25:
@igrr I propose to have git build with softwareserial in a similar manner as lwip or axtls. I think having libs missing in the git build causes too much confusion.
What do you think?

igrr, 2018-01-01:
Verified that this is fixed in latest git (and 2.4.0-rc2 has SoftwareSerial).

closed 2018-01-01 by igrr

dragondaud, 2018-12-11:
With a fresh install of the latest github release it is leaving an empty SoftwareSerial directory under libraries, which is generating errors:

``Invalid library found in /Users/daud/Documents/Arduino/hardware/esp8266com/esp8266/libraries/SoftwareSerial: no headers files (.h) found in /Users/daud/Documents/Arduino/hardware/esp8266com/esp8266/libraries/SoftwareSerial``

I am not using SoftwareSerial, but these errors were not present previously. Deleting the empty directory cures the problem.

devyte, 2018-12-11:
Were you playing with the ongoing release (tag 2.5.0-beta1), then tried rolling back?

dragondaud, 2018-12-11:
No, it was a fresh pull, this morning, from github, followed by a get.py for the tools.

d-a-v, 2018-12-11:
This new `SoftwareSerial` directory in `libraries` directory is a git-submodule.
To populate it, use
```
git submodule update --init
```

dragondaud, 2018-12-11:
Gotcha! FYI: I just discovered that its not from the fresh clone, but when my script does `git reset --hard origin/master` that the errant directory is created. I thought that command would make sure any local changes were reset, but apparently it does more...

devyte, 2019-01-04:
Note: need to update git install doc