esp8266/Arduino #3597: Stack and heap memory - what is used where

mamama1, 2017-09-11:
Hi!

currently I'm working on a quite big project using a ESP8266.
So far I've managed to sort out many issues and I'm very happy at the moment because everything works like a charm :-)

However I was wondering, what takes up stack and what takes up heap space.
Somewhere I've read that global variables (that is, for me, variables defined outside of any function, right?) are in the heap and local variables (that is, for me, variables defined INSIDE of functions, right?).

is that true?

I'm wondering because I'm hitting the stack limit a bit and I'm having multiple large buffers defined within functions, all about 1000 to 3500 bytes large, maybe about 5 or 6 of them all together.

to avoid hitting the stack, I wanted to define one big buffer on the top of my ino file and use that buffer over and over from all of my functions.
however, when I did that, everything started to act weirdly, as if I had memory corruption somewhere or as if I was running out of memory.

Is that expected, am I missing something? Or should it work and I have to have a mistake in my code somewhere else?

Directions, please :-)

regards
mamama1

Makuna, 2017-09-11:
local variables do use the stack.

One shared buffer will work as long as you make sure its not being used for two things at once; WiFi callbacks happen as interrupts; so you could be in the middle of using on the main Loop() at the same time?

mamama1, 2017-09-11:
thats weird... maybe it was because of the interrupts but that global buffer didn't work at all in my case.

Makuna, 2017-09-12:
The I suspect you have some bug in your code; as using a global is a common way to work around keeping stack small and even tons of alloc/dealloc fragmentation issues.

devyte, 2017-09-12:
@mamama1 declarations inside functions are on the stack, which on the ESP is at max 4KB. objects allocated with new, malloc, or self-managed objects, such as String or std::string, are on the heap.
On the ESP, you have a trade-off: using the stack is best to reduce mem fragmentation, but the stack is tiny so you must be careful about blowing it up. Allocating on the heap gives you the entire range of available mem, but you could run into fragmentation, which reduces the size of largest chunk possble at any moment. In addition, if you do use the heap, you should a std container, a smart pointer, or implement your own self-managed object, because _Reasons_.
Having said that, this is not the right place to ask for help for your particular project, let alone questions that are inherent to C++ as a language, or inherent about best programming practices. This is an issue tracker, meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow for assistance about such matters.
Closing as off-topic.

closed 2017-09-12 by devyte

gnbl, 2019-10-17:
Cross-post: https://www.esp8266.com/viewtopic.php?p=69937