esp8266/Arduino #3600: mills rollover

alexgrauer, 2017-09-12:
Hi,

Im working with the Node MCU ESP8266 and Arduino IDE, and am trying to find out how mills() work. I read everywhere that it will rollover every ~72 minutes, but I did some tests and for it worker around 12 hours before rollover happened. Millis got up to ~43 millions.

So I don't understand whats going on. Any idea?

Also, I need to keep track of how many rollover occurred since the program started, so I did something like this.

int32_t last_global_millis = 0;
uint32_t new_global_millis = 0;
uint16_t amount_of_millis_rollovers = 0;

//This happens in an interruption I have for some events
new_global_millis = millis();
if(new_global_millis < last_global_millis){
    amount_of_millis_rollovers++; 
}
last_global_millis = new_global_millis;

But variable amount_of_millis_rollovers is always 0. Any idea?

Also I would like to know if there's anyway of moving mills value close to the rollover value, so I can test the code without having to wait 12 hours.

Thanks in advance.

Hardware:	        ESP8266MOD
Vendor:                   HYSIRY

Settings in IDE

Module:  	        NODE MCU 1.0 (ESP-12E Module)
Flash Size: 		4M (3M SPIFFS)
CPU Frequency:    80Mhz
Upload Using:       SERIAL

igrr, 2017-09-12:
We take some effort to avoid `millis` rollover until it reaches UINT32_MAX, see the code here:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_wiring.c#L56-L68. So it should overflow after roughly 50 days, if i'm not mistaken (2^32 / 1000 / 3600 / 24 = 49.7)

alexgrauer, 2017-09-12:
And what do you think about what I said of the mills rolling over after 12 hours? How is that possible?

igrr, 2017-09-12:
If that is the case, there may be some bug... Running a test now, will see the result after 12 hours.

alexgrauer, 2017-09-12:
Great.. I'll come back to you in 12 hours then.
Thanks very much @igrr

alexgrauer, 2017-09-12:
Hi @igrr.

Have a look at this.

Im running this code that saves into a sd card when was the last time i received a message from different devices with a certain MAC.

//Code
new_millis = millis();

sprintf(data, "M>%02x:%02x:%02x:%02x:%02x:%02x T>%lu-%lu",new_mac_address[0], new_mac_address[1], new_mac_address[2], new_mac_address[3], new_mac_address[4], new_mac_address[5],amount_of_millis_rollovers,new_millis);

dataFile.println(data);
dataFile.close();

//End of the code

This is the output after running the code from ~14:30 to ~16:23.
(Just put the first and last lines of the txt)

M>58:e2:8f:79:4e:67 T>0-40658
M>d8:00:4d:f1:1e:5d T>0-41648
M>d8:00:4d:f1:1e:5d T>0-44629
M>d0:df:9a:92:b0:4e T>0-75198
...........................
M>8a:d6:f1:0a:92:e5 T>0-6703495
M>da:a1:19:66:82:0e T>0-6741951
M>d0:df:9a:92:b0:4e T>0-6741960
M>2a:01:94:dd:cb:4f T>0-6771379

As you can see, from the 14:30 to 16:23 there are 113 minutes = 6780 seconds = 6.780.000 milliseconds. (same number of the last lines before turning of the device)
So for sure the mills is not rolling over after 72 minutes...

Yesterday I didn't turn it off and it rolled over a bit before 12 hours. And whats curious is that 12 hours = 43.200 seconds = 43.200.000 milliseconds which is ~(2^32)/100. So of course I have no idea what this means, but I guess it has to be related. right?

Just a bit more of info on this. Let me know if you find something after your test.
Thanks

alexgrauer, 2017-09-13:
Any news on this?
Thanks

igrr, 2017-09-14:
I haven't been able to reproduce the wraparound/overflow. The millisecond counter has reached the value 79194816, which is consistent with the wall-clock time passed since I have started the test (approximately 22 hours).

Please provide the complete minimal sketch to reproduce the issue.

igrr, 2017-09-26:
```
 Wed Jan 14 07:13:13 1970    1149193355  -1857800717  -1871185801          267    0
```

14 days and 7 hours, 267 wraparounds of microsecond counter, 0 wraparounds of millisecond counter.

devyte, 2017-10-22:
Closing due to can't reproduce and no feedback in over a month.

closed 2017-10-22 by devyte