esp8266/Arduino #3604: Exception (29) happens while trying to receive long uart_rx_buffer

xuanduc611, 2017-09-14:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3.0

### Description
Exception(29). I'm making an application which needs to use UART interface to receive data of 30 to 40 bytes. I use Serial.available() to check if there were incoming buffers in the serial port. This task is executed every T ms. T is defined by millis().

- T = 50 - 100: After the module boots, I send rx_buffer to ESP8266 continuously, about every 1 second, everything seems to be fine after 2 minutes then the module reset and the console reports exception 29. 
- T = 500 - 700: Error still happens, the difference is that it happens after 3-4 minutes since the module boots.

If I don't send rx_buffer after the module boots, nothing happens and the module can run a long time before I send rx_buffer and exception(29) happens again.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:       80Mhz
Flash Mode:	        DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include "m_wifi.h"

#define _DB_OPCODE_UART_
#define _DB_WIFI_SET_

uint8_t g_ssid[SSID_PASS_LEN_MAX];
uint8_t g_pass[SSID_PASS_LEN_MAX];
uint8_t g_uart_rx_flg = 0;

uint64_t prev_millis = 0;
uint64_t cur_millis = 0;

IPAddress g_ip_addr;
IPAddress g_gateway;
IPAddress g_subnet;
IPAddress g_dns;

void setup()
{    
    Serial.begin(921600);
    Serial.println("M-WiFi");
    WiFi.mode(WIFI_STA);
}

void loop() 
{
    uint64_t db_millis;
    cur_millis = millis();
    db_millis = cur_millis;
    // 500ms event:
    if (cur_millis - prev_millis >= 500) {
        Serial.println((double)db_millis, DEC);
        prev_millis = cur_millis;
        uart_rx_manager();
    }
//    uart_rx_manager();
//    delay(200);
    // Other tasks like: reconnect,..
}

/* @brief uart frame indicator
 * @return buffer received from CPU
 */
uint8_t* uart_rx_data_ind(void)
{
    uint8_t *rx_data_buf = new uint8_t[100];
    uint8_t cnt = 0;
    uint8_t rx_data;

    while (Serial.available()) {
        g_uart_rx_flg = 1;
        rx_data = Serial.read();
        rx_data_buf[cnt++] = rx_data;
    }
    return rx_data_buf;
}

/* @brief uart rx frame manager
 */
void uart_rx_manager()
{
    uint8_t *p_buf;
    uint8_t opcode;
    
    p_buf = (uint8_t *)malloc(50);
    p_buf = uart_rx_data_ind();

    // UART_RX cmd opcode handler:
    if (g_uart_rx_flg) {
         Serial.println(__FUNCTION__);
         g_uart_rx_flg = 0;
         opcode = *p_buf++;
        
        // Process opcode
        uart_rx_opcode_handler(opcode, p_buf);
    }    
}

/* @brief uart opcode handler
 * @param opcode. 
 * @param ptr:  pointer to the Data portion of the Frame.  
 * @return none
 */
void uart_rx_opcode_handler(uint8_t opcode, uint8_t* ptr)
{   
    Serial.println(__FUNCTION__);
    switch (opcode) {
        case OPU_WIFI_SYNC:
        // Data = "MGW"
            #ifdef _DB_OPCODE_UART_
                Serial.println("OPU_WIFI_SYNC");
            #endif
        break;
        
        case OPU_WIFI_SET:
            #ifdef _DB_OPCODE_UART_
                Serial.println("OPU_WIFI_SET");
            #endif
            wifi_set_handler(ptr);
        break;

        case OPU_MQTT_SUB_TOP:
            #ifdef _DB_OPCODE_UART_
                Serial.println("OPU_MQTT_SUB_TOP");
            #endif
        break;

        case OPU_PHONE:
            #ifdef _DB_OPCODE_UART_
                Serial.println("OPU_PHONE");
            #endif
        break;

        case OPU_STATUS:
            #ifdef _DB_OPCODE_UART_
                Serial.println("OPU_STATUS");
            #endif
        break;
    }
}

/* @brief wifi sync handler
 * @param ptr: pointer to the Data portion of the Frame.
 * @return none.
 */
void wifi_sync_handler(const uint8_t* ptr)
{
    
}

/* @brief wifi set handler
 * @param ptr: pointer to the Data portion of the Frame.
 * @return none.
 * Data () = [Mode(1)][SsidLen(1)=N1][Ssid(N1)][PassLen(1)][Pass (N2)] 
 *          [StaticIp(1)][IpAddr(4)][Subnet(4)][Gateway(4)][Dns(4)]
 */
void wifi_set_handler(const uint8_t* ptr)
{
    //uint8_t i;
    uint8_t w_mode;
    uint8_t ssid_len;
    uint8_t pass_len;
    uint8_t static_ip;
    uint8_t ip_addr[4];
    uint8_t subnet[4];
    uint8_t gateway[4];
    uint8_t dns[4] = DEFAULT_DNS;
    const uint8_t* p_cfg_buf = ptr;
    String str_ssid, str_pass;

    // Get config param:
    w_mode = *p_cfg_buf++;
    ssid_len = *p_cfg_buf++;
    memcpy(g_ssid, p_cfg_buf, ssid_len);
    p_cfg_buf += ssid_len;

    pass_len = *p_cfg_buf++;
    memcpy(g_pass, p_cfg_buf, pass_len);
    p_cfg_buf += pass_len;

    static_ip = *p_cfg_buf++;
    memcpy(ip_addr, p_cfg_buf, 4);
    p_cfg_buf += 4;
    g_ip_addr = (ip_addr[0], ip_addr[1], ip_addr[2], ip_addr[3]);

    memcpy(subnet, p_cfg_buf, 4);
    p_cfg_buf += 4;
    g_subnet = (subnet[0], subnet[1], subnet[2], subnet[3]);

    memcpy(gateway, p_cfg_buf, 4);
    p_cfg_buf += 4;
    g_gateway = (gateway[0], gateway[1], gateway[2], gateway[3]);

    memcpy(dns, p_cfg_buf, 4);
    p_cfg_buf += 4;
    g_dns = (dns[0], dns[1], dns[2], dns[3]);

    #ifdef _DB_WIFI_SET_
        str_ssid = String((char*)g_ssid);
        str_pass = String((char*)g_pass);
        Serial.println(str_ssid);
        Serial.println(str_pass);
    #endif
}
```

m_wifi.h
```
#ifndef _M_WIFI_H_
#define _M_WIFI_H_

#include <Arduino.h>

// Received data opcode:
#define OPU_WIFI_SYNC       0x01
#define OPU_WIFI_SET	 	    0x02
#define	OPU_MQTT_SUB_TOP	  0x03
#define OPU_PHONE			      0x04
#define OPU_STATUS			    0x05	

// Wifi Mode:
#define W_DISABLE           0x00
#define W_STA               0x01
#define W_AP                0x02
#define W_AP_STA            0x03

// Wifi param
#define SSID_PASS_LEN_MAX   20
#define DEFAULT_DNS         {192, 168, 1, 1}

#endif

```

### Debug Messages

```
Exception (29):
epc1=0x40201c59 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef310 end: 3ffef500 offst: 01a0

>>>stack>>>
3ffef4b0:  3fffdad0 3ffee30 00000000 40201c55  
3ffef4c0:  3fffdad0 3ffee380 3ffee388 40201dfa  
3ffef4d0:  00000000 3ffe380 3ffee388 40201e94  
3ffef4e0:  3fffdd0 00000000 3ffee4c8 402029dc  
3ffef4f0:  feefeffe feefeffe 3ffee4e0 40100718  
<<<stack<<<
```
### Message reported by ESP Exception Decoder:
```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 6 results
0x40201c59: uart_rx_data_ind() at D:\Projects\m-gateway\firmware\MGW-Wifi\MGW-Wifi/MGW-Wifi.ino line 128
0x40201c55: uart_rx_data_ind() at D:\Projects\m-gateway\firmware\MGW-Wifi\MGW-Wifi/MGW-Wifi.ino line 128
0x40201dfa: uart_rx_manager() at D:\Projects\m-gateway\firmware\MGW-Wifi\MGW-Wifi/MGW-Wifi.ino line 70
0x40201e94: loop at D:\Projects\m-gateway\firmware\MGW-Wifi\MGW-Wifi/MGW-Wifi.ino line 40
0x402029dc: loop_wrapper at C:\Users\xuand\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\xuand\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```

devyte, 2017-09-14:
@xuanduc611 at a quick glance, your memory allocation, and the way you are filling up your buffers, is very wrong. You have at least two mem leaks and one potential buffer overrun.

Please review your code and adjust your project. If you still need assistance, refer to esp8266.com or stackoverflow.
Closing as user error.

closed 2017-09-14 by devyte

xuanduc611, 2017-09-14:
@devyte 
Could you show me the position of buffer overrun and two mem leaks ?
Thanks for your help.