esp8266/Arduino #3606: I tried to upload this sample blick program but I get error like this

dilanWijerathne, 2017-09-14:
### Basic Infos

#### Hardware
Hardware:			ESP8266 Mod
Core Version:      	Ver 0.1
Vendor:  HYSIRY
ISM  :   2.4 GHZ

### Description

I tried to upload this sample blick program but I get error like this 

_Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\DILANW~1\AppData\Local\Temp\arduino_cache_620386\core\core_esp8266_esp8266_nodemcu_CpuFrequency_80,UploadSpeed_9600,FlashSize_4M3M_ba73b1ba46806f1fbac0a23074e5c5c0.a
Sketch uses 222221 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31588 bytes (38%) of dynamic memory, leaving 50332 bytes for local variables. Maximum is 81920 bytes.
Uploading 226368 bytes from C:\Users\DILANW~1\AppData\Local\Temp\arduino_build_899801/Blink.ino.bin to flash at 0x00000000
error: failed sending 1072 bytes
.error: failed sending 1044 bytes
.error: failed sending 1040 bytes
.error: failed sending 1040 bytes
.error: failed sending 1070 bytes
.error: failed sending 1084 bytes
.error: failed sending 1069 bytes
.error: failed sending 1080 bytes
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed_

### Settings in IDE
Arduino IDE
Module:  			?Generic ESP8266 Module
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?

Flash Frequency:	?40Mhz?
Upload Using:		?/ SERIAL?
Reset Method:		?/ nodemcu?

### Sketch

```cpp

`
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}`

### Debug Messages

Uploading 226368 bytes from C:\Users\DILANW~1\AppData\Local\Temp\arduino_build_899801/Blink.ino.bin to flash at 0x00000000
error: failed sending 1072 bytes
.error: failed sending 1044 bytes
.error: failed sending 1040 bytes
.error: failed sending 1040 bytes
.error: failed sending 1070 bytes
.error: failed sending 1084 bytes
.error: failed sending 1069 bytes
.error: failed sending 1080 bytes
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

devyte, 2017-09-14:
@dilanWijerathne the comm error is documented in the docs. Either your have wrong wiring, wrong drivers, permission problems, wrong port configuration, or your board is bad.
I just flashed an ESP myself with latest git.
If you can't figure it out, I suggest asking at esp8266.com, stackoverflow, or gitter for assistance. 
Closing.

closed 2017-09-14 by devyte

nardev, 2018-01-28:
I had same issue on couple of electrodragon relay boards with ESP12F, when trying to program with Serial dongle

but the strange thing is that it worked first few times.. and than.. failing to upload with same error 

warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed

I tried original AT firmware and the module works but it was not possible to burn arduino firmware..

any clue??

MazenATaha, 2018-06-14:
if u are using  Lolin NodeMCU v3 (CH340G) please consider using those tool setting , It worked for me .
[image]

MazenATaha, 2018-06-15:
At first I Thought the same but after searching for a bit I figured out that the written 9600 Baud Rate is for Other USB Boot-loaders & WiFi devices to configure with the WiFi ESP module not for Ardunio IDE.

RedRush1, 2019-04-22:
> if u are using Lolin NodeMCU v3 (CH340G) please consider using those tool setting , It worked for me .
> [image]

registered to say thank you

das800, 2020-08-13:
> 
> 
> if u are using Lolin NodeMCU v3 (CH340G) please consider using those tool setting , It worked for me .
> [image]

worked for me. thanks alot

ilan-perez, 2021-09-05:
I have this board.
https://www.ebay.com.au/itm/324201034080?ul_noapp=true
Wemos D1 Mini Pinout - https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

I was able to upload tot he baord for a long time and now i can't. I get this error
error: espcomm_upload_mem failed

I have tried all the settings above to get the simple blink program to load onto the board but i get the same error, Can anyone help out here