esp8266/Arduino #3607: Exception (3) appears randomly in various moments

jd4096, 2017-09-14:
#### Hardware
Hardware:			ESP-8266
Core Version:      	2.4.0 - rc1 (happens on 2.3.0 too)
Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		DOUT
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

#Problem description
Hi! I have problem with my ESP8266 and ESP8266WiFi library. I'm using only <ESP8266WiFi.h>and i'm making something like bridge from Arduino to remote devices. Arduino is sending temperature via serial port directly to ESP and ESP is sending it to all connected client via TCP.
The problem is that Exception 3 appears randomly. I have literally added debug messages after every code line and as a result i know that this exception appears randomly.

This is my CLIENTS_NUM variable, which will be used in iterating all connected clients:
`const int CLIENTS_NUM = 5;
WiFiClient clients[CLIENTS_NUM];`

1. Exception (3) when trying to send data to all clients:
https://pastebin.com/di9zvzxF

2. Exception (3) when trying to disconnect client
https://pastebin.com/pzJsJZui
as you can see, this must appear on client[i].stop(), because last debug message is: DISCNCL1

3. Exception (3) when trying to read and print free heap
https://pastebin.com/A9pM9RYr

devyte, 2017-09-14:
@jd4096 issue template was largely ignored, no minimal complete sketch, missing basic info.
Random crashes like this are usually indicative of mem corruption. Keep in mind that the stack size is only 4KB max, and that you can't have code paths that take too long to execute, especially in presence of high wifi traffic. Also, I'm not sure about the whole sending to all connected clients thing. Try one client first, see if that works at all.
I suggest asking at esp8266.com or stackoverflow for help with your particular project. If you can provide minimal code that shows an issue in the core, please open a new issue and provide the details.
Given the current cleanup effort, closing due to lack of info.

closed 2017-09-14 by devyte

jd4096, 2017-09-14:
Thanks for explaining things about memory. And yes -> that was causing a problem. I was initializing char* buffers with "malloc" function in a runtime and that was causing an Exception. The funny thing is that I have checked heap memory every loop and it was at 35-40KB, which seems OK. I'm still newbie at C/C++ but... problem is solved! ;-)