esp8266/Arduino #3609: loadCACert problem

rudivandrunen, 2017-09-15:
Hi all,

I installed the *github* version of the esp8266/Arduino package (so NOT using package/boards manager, but installing using the git clone method). 

Now I have an issue using loadCACert.

I can use loadCertificate and loadPrivateKey without problems, but I get a compile time issue on loadCACert. see this snippet:

```
...
  File ca = SPIFFS.open("/ca.der", "r"); 
  if (!ca) {
    Serial.println("Failed to open CA file");
  }
  else {
  Serial.println("Success to open CA file");
  if(espClient.loadCACert(ca))
    Serial.println("CA loaded");
    else
    Serial.println("CA not loaded");
  }
...
```

The error I get shows as: 

```
/Users/Arduino/Documents/Arduino/_20170913b/_20170913b.ino: In function 'void setup_certs()':
_20170913b:156: error: no matching function for call to 'WiFiClientSecure::loadCACert(fs::File&)'
   if(espClient.loadCACert(ca))
```

Can it be that the file Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.h is missing a

```
  template<typename TFile>
  bool loadCACert(TFile& file) {
    return loadCACert(file, file.size());
  }
```

or am I to naive here !?
BTW, I am using the Arduino IDE 1.8.1 on Mac

Can someone help ?

Thanks
Rudi

igrr, 2017-09-15:
Indeed, there is no such function (`loadCACert(file)`) defined, only `loadCACert(Stream& size, size_t size)` is. Feel free to open a PR adding a missing function.

rudivandrunen, 2017-09-15:
PR Raised.

crywolf87, 2017-11-17:
hello.
I have the same problem.

I upload the ca.crt with SPIFFS but the ca not load.

I search in this forum a solution but i don't undestand like build the correctly .ino.

this is my  .ino file  and my debug serial .

`#include "FS.h"
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

// Update these with values suitable for your network.

const char* ssid = "TIM-94277189";
const char* password = "l2VjbBJjBVbkvJT4cjJ4VYih";
const char* mqtt_server = "192.168.1.20"; //MQTT broker ipAddress server(192, 168, 1, 20);
const char* mqttusername = "davide";
const char* mqttpass = "53Amm@n021";

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

}
WiFiClientSecure espClient;
PubSubClient client(mqtt_server,8883,callback,espClient); //set  MQTT port number to 8883 as per //standard
long lastMsg = 0;
char msg[50];
int value = 0;

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client",mqttusername ,mqttpass)) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  Serial.begin(115200);
  setup_wifi();
  delay(1000);
  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
    return;
  }

File ca = SPIFFS.open("/ca.crt", "r"); //replace ca.crt eith your uploaded file name
  if (!ca) {
    Serial.println("Failed to open ca file");
  }
  else
  Serial.println("Success to open ca file");

if(espClient.loadCertificate(ca))
  Serial.println("loaded");
  else
  Serial.println("not loaded");

  //client.setServer(mqtt_server, 8883);
  //client.setCallback(callback);
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, 75, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
}`

WiFi connected
IP address: 
192.168.1.39
Success to open ca file
not loaded
Attempting MQTT connection...connected
Publish message: hello world #1

the mqtt is connected but the ca isn't verify.

p.s. sorry for my English! 
 :-(

crywolf87, 2017-11-17:
hello.
I have the same problem.

I upload the ca.crt with SPIFFS but the ca not load.

I search in this forum a solution but i don't undestand like build the correctly .ino.

this is my .ino file and my debug serial .

`#include "FS.h"
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

// Update these with values suitable for your network.

const char* ssid = "TIM-xxxxxxxx";
const char* password = "xxxxxxxxxxxxxxxxxxh";
const char* mqtt_server = "192.168.1.20"; //MQTT broker ipAddress server(192, 168, 1, 20);
const char* mqttusername = "xxxxxxxxx";
const char* mqttpass = "xxxxxxxxxxxx";

void callback(char* topic, byte* payload, unsigned int length) {
Serial.print("Message arrived [");
Serial.print(topic);
Serial.print("] ");
for (int i = 0; i < length; i++) {
Serial.print((char)payload[i]);
}
Serial.println();

}
WiFiClientSecure espClient;
PubSubClient client(mqtt_server,8883,callback,espClient); //set MQTT port number to 8883 as per //standard
long lastMsg = 0;
char msg[50];
int value = 0;

void setup_wifi() {

delay(10);
// We start by connecting to a WiFi network
Serial.println();
Serial.print("Connecting to ");
Serial.println(ssid);

WiFi.begin(ssid, password);

while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}

Serial.println("");
Serial.println("WiFi connected");
Serial.println("IP address: ");
Serial.println(WiFi.localIP());
}

void reconnect() {
// Loop until we're reconnected
while (!client.connected()) {
Serial.print("Attempting MQTT connection...");
// Attempt to connect
if (client.connect("ESP8266Client",mqttusername ,mqttpass)) {
Serial.println("connected");
// Once connected, publish an announcement...
client.publish("outTopic", "hello world");
// ... and resubscribe
client.subscribe("inTopic");
} else {
Serial.print("failed, rc=");
Serial.print(client.state());
Serial.println(" try again in 5 seconds");
// Wait 5 seconds before retrying
delay(5000);
}
}
}

void setup() {
Serial.begin(115200);
setup_wifi();
delay(1000);
if (!SPIFFS.begin()) {
Serial.println("Failed to mount file system");
return;
}

File ca = SPIFFS.open("/ca.crt", "r"); //replace ca.crt eith your uploaded file name
if (!ca) {
Serial.println("Failed to open ca file");
}
else
Serial.println("Success to open ca file");

if(espClient.loadCertificate(ca))
Serial.println("loaded");
else
Serial.println("not loaded");

//client.setServer(mqtt_server, 8883);
//client.setCallback(callback);
}

void loop() {

if (!client.connected()) {
reconnect();
}
client.loop();

long now = millis();
if (now - lastMsg > 2000) {
lastMsg = now;
++value;
snprintf (msg, 75, "hello world #%ld", value);
Serial.print("Publish message: ");
Serial.println(msg);
client.publish("outTopic", msg);
}
}`

WiFi connected
IP address:
192.168.1.39
Success to open ca file
not loaded
Attempting MQTT connection...connected
Publish message: hello world #1

the mqtt is connected but the ca isn't verify.

p.s. sorry for my English!
:-(

igrr, 2017-12-27:
Closing since the PR has been merged.

closed 2017-12-27 by igrr