esp8266/Arduino #3617: error: pll_cal exceeds 2ms!!! after flashing

mamama1, 2017-09-17:
### Basic Infos

#### Hardware
Hardware:			NodeMCU by RobotDyn
Core Version:      	2.3.0

### Description
After flashing the NodeMCU with a new sketch version, WiFi doesn't connect and shows the following behaviour:
WiFi.status() returns "WL_NO_SSID_AVAIL"
Debug outputs "error: pll_cal exceeds 2ms!!!"

When I disconnect the module from USB, wait a second and reconnect to USB, the WiFi connection establishes within seconds.

I have found two issues here, regarding this, both not really satisfying nor 100% like my issue (because my issue disappears, when power cycling the module).

Any idea?

Thanks

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL/USB
Reset Method:		?
`
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <Wire.h> // Include the I2C library (required)
#include <Adafruit_PWMServoDriver.h>

char staSSID[51] = { "something" };
char staPSK[51] = { "something" };
Adafruit_PWMServoDriver io = Adafruit_PWMServoDriver();
unsigned long WiFiCheckMillis = 0;
bool WiFiEnabled = true;
bool WiFiConnected = false;
unsigned long WiFiLEDMillis = 0;
uint16_t WiFiLEDBrightness = 0;
bool WiFiLEDfadeUp = true;
bool WiFiLEDOff = true;
uint8_t led_status_blue_channel = 10;

void setup()
{
	// Init UART
	if (DEBUG)
	{
		Serial.begin(115200);
		Serial.setDebugOutput(true);
	}
	WiFi.mode(WIFI_STA);
	WiFi.persistent(false);
	WiFi.disconnect(true);
	WiFi.begin(staSSID, staPSK);
	io.begin();
	io.setPWMFreq(1600);
}

void loop()
{
	if (millis() > WiFiCheckMillis + 1000)
	{
		if (WiFi.getMode() == WIFI_STA && WiFiEnabled)
		{
			if (DEBUG)
			{
				Serial.print("WiFi status: ");
				Serial.println(WiFi.status());
			}
			if (WiFi.status() != WL_CONNECTED)
			{
				WiFiConnected = false;
			}
			else
			{
				WiFiConnected = true;
			}
		}
		WiFiCheckMillis = millis();
	}

	if (WiFiConnected == false)
	{
		if (millis() > WiFiLEDMillis + 50)
		{
			if (WiFiLEDBrightness <= 1000 && WiFiLEDfadeUp)
			{
				WiFiLEDBrightness += 100;
			}
			else if (WiFiLEDBrightness >= 100 && WiFiLEDfadeUp == false)
			{
				WiFiLEDBrightness -= 100;
			}
			else if (WiFiLEDBrightness == 0)
			{
				WiFiLEDfadeUp = true;
				WiFiLEDBrightness = 100;
			}
			else if (WiFiLEDBrightness == 1100)
			{
				WiFiLEDfadeUp = false;
				WiFiLEDBrightness = 1000;
			}

			WiFiLEDOff = false;
			writePWM(led_status_blue_channel, WiFiLEDBrightness);

			WiFiLEDMillis = millis();
		}
	}
	else
	{
		if (WiFiLEDOff == false)
		{
			writePWM(led_status_blue_channel, 0);
			WiFiLEDOff = true;

			if (DEBUG)
			{
				Serial.println("");
				Serial.print("Connected to ");
				Serial.println(staSSID);
				Serial.print("IP address: ");
				Serial.println(WiFi.localIP());
			}
		}
	}
	
void writePWM(uint8_t channel, uint16_t value)
{
	io.setPin(led_pins[channel], value);
}`

devyte, 2017-09-17:
@mamama1 please try with latest git. There are changes related to wifi radio calibration.

devyte, 2017-10-22:
@mamama1 will you be testing latest git? Also, please do a full flash erase before uploading a new sketch.

mamama1, 2017-10-27:
Hi

sorry for the late response. The issue just disappeared a while ago, maybe because of some buffer overflows I corrected.
I started getting used to it and at some point in development the issue just vanished.

Thanks for your help, anyway. If it reappears, I'll try the latest git (btw is that 2.4.0-rc2 or will latest git be even more recent?

Thanks

devyte, 2017-10-27:
Latest git is always the most recent.
Closing as resolved.

closed 2017-10-27 by devyte