esp8266/Arduino #3619: V2.4.0RC1 Memory Leak using WiFiClientSecure on closing connection

mars000, 2017-09-17:
Hi

After many many hours of problem solvingI have verified the same code running on V2.3.0 and also V2.4.0RC1. Is this a known issue ?
I have found it runs for days and days with no issues on V2.3.0 irrespective, if I keep connection open or if I close it.
But on V2.4.0RC1 it crashes after an hour if I choose to close the connection. Specifically, I have found that in situations where I close WiFiClientSecure connection and later re-open it fails to recover all memory. However, on V2.4.0RC1 I always leave the connection open as per the first piece of code below then it works on V2.4.0RC1.

**Specifically:**
The below code leaves connection open and this seems to work for both V2.3.0 and V2.4.0RC1.
```cpp
  client->print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + blynk_server + "\r\n" +
                 "User-Agent: BuildFailureDetectorESP8266\r\n" +
                 "Connection: open\r\n\r\n"); // leave connection open
```

However when closing the connect with below code causes memory leak on V2.4.0RC1.
```cpp
  client->print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + blynk_server + "\r\n" +
                 "User-Agent: BuildFailureDetectorESP8266\r\n" +
                 "Connection: close\r\n\r\n"); // close connection
```

### Basic Infos

#### Hardware
Hardware:			ESP-12E (Wemos Mini Pro)
Core Version:      	2.4.0-rc1

### Description

Problem description

### Settings in IDE

Module:  			Wemos Mini Pro
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

igrr, 2017-09-20:
Hi @mars000, thank you for the report.

I have tried reproducing this issue, but wasn't able to do so based on your description. Could you share the sketch which exhibits the issue?

Here's what i have tried:

```c++
/*
 *  HTTP over TLS (HTTPS) example sketch
 *
 *  This example demonstrates how to use
 *  WiFiClientSecure class to access HTTPS API.
 *  We fetch and display the status of
 *  esp8266/Arduino project continuous integration
 *  build.
 *
 *  Created by Ivan Grokhotkov, 2015.
 *  This example is in public domain.
 */

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = ".................";
const char* password = "...................";

const char* host = "api.github.com";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C";

// Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  
void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }

  String url = "/repos/esp8266/Arduino/commits/master/status";
  Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
  }
  String line = client.readStringUntil('\n');
  if (line.startsWith("{\"state\":\"success\"")) {
    Serial.println("esp8266/Arduino CI successfull!");
  } else {
    Serial.println("esp8266/Arduino CI has failed");
  }
  

  Serial.println("closing connection");
  Serial.println(ESP.getFreeHeap());
}
```

This sketch has been running for a couple of hours, and amount of free heap memory is constantly reported as 23296, unless there is a connection failure. When failure happens, free heap value is lower, but it recovers on the next attempt.

mars000, 2017-09-20:
Hi @igrr 

I've run your code and still seeing memory leaking. Admittedly I am getting connection failed for some reason even though I am running your sketch without change and can confirm my ESP does connect to internet via my wifi. I have also confirmed from Arduino IDE that I am compiling with V2.4.0Rc1. Not sure why you seeing something different.

Pls see attached serial output I've captures.

Regards
mars000
[CoolTerm Capture 2017-09-20 15-29-50.txt](https://github.com/esp8266/Arduino/files/1316701/CoolTerm.Capture.2017-09-20.15-29-50.txt)

igrr, 2017-09-20:
Okay, then the difference may be in that you are using 2.4.0-rc1 and i am using latest git version.

There was a memory leak in axTLS (https://github.com/esp8266/Arduino/issues/3428, https://github.com/igrr/axtls-8266/pull/50) which was fixed after 2.4.0-rc1. Could it be that you are running into that issue?

mars000, 2017-09-20:
thank you @igrr that was the issue. Appreciate your assistance.

BTW: I'm trying to port all my code to ESP32 platform...is there a site that provides an easy to use side by side comparison on core functions that are supported on ESP32 and if there are any differences what they might be ?  -- would help make porting a lot simpler. ESPhttpUpdate.update is one such function as an example I'm trying to see if ESP32 support.

igrr, 2017-09-20:
I don't think there is. You may ask on arduino-esp32 gitter channel, maybe the folks who hang out there can give you some pointers.

tarzan115, 2017-09-22:
#3627

bobybobybob, 2017-09-28:
Dear Mr Igrr,
thank you very much for Hard ESP Work for Happy Users :-)

closed 2017-10-01 by igrr