esp8266/Arduino #3620: How to participate in gihub when install Board manger version?

cheng3100, 2017-09-18:
### Basic Infos
How to modify the code and test so that to push to github when use the Board Manger install the esp8266 library in arduino?
#### Hardware
Hardware:			ESP-8266

### Description

I am new to github. Maybe the issue is inappropriate. I have installed the esp8266 lib by board manger in arduino. It download the file in the disk and the file structure is diffrent from the github version. I want to konw that:
- In this situtation how to modify code and push to github? Must I install the github version? 
- If I install the github version , will it have some distraction with the board manger version ?I have did this and find some .ino example in it can't compile anymore(report that some symbol can't find)

tttapa, 2017-09-19:
What do you want to push to GitHub? 
Unless you are registered as a collaborator, you can't push to this repository.
If you want to make your own changes/improvements, you can fork this repo and clone it locally, using the git version installation instructions, but using your own GitHub URL (probably /cheng3100/Arduino) instead of /esp8266/Arduino.

Pieter

cheng3100, 2017-09-19:
Thank you for your reply.So I must use the git version then to make improments.BUT how to uninstall a board manager version in arduino which I have already install? If I install the git version and board manager version in the same time ,it seems has some distraction...

devyte, 2017-09-20:
@cheng3100 you need to
- fork the repo
- make a branch in your fork
- commit to that branch
- make a PR from github
- delete the branch once the pr is merged

There are tutorials on stackoverflow, just google How to make pr on github.

closed 2017-09-20 by devyte