esp8266/Arduino #3621: SPIFFS size is not an integer multiple of block size.

ABH10, 2017-09-18:
For some of the boards defined in boards.txt, the allocation for SPIFFS is not an integer multiple of the SPIFFS block size. 
For 4MB boards, for example, there is 1MB for PROGMEM, 3052 kB for SPIFFS and 20kB for EEPROM. The block size is 8 kB, but 3052 kB is not dividable by 8 kB blocks.
I propose to modify eagle.flash.4m.ld linker file and boards.txt to use an SPIFFS block size of 4 kB, which works fine.

me-no-dev, 2017-09-19:
We need to check if changing block size would make the partition unmountable. I agree otherwise that block size should be 4K.

ABH10, 2017-09-20:
I have been using 4 kB block size for some time now. Before switching from 8 kB to 4 kB block size I had all kind of issues. I am working with large files in SPIFFS that are updated often and are opened in multiple file descriptors. Using an SPIFFS size that is a non-integer multiple of block size is not compliant with the mount instructions from Peter Andersson.
When an SPIFFS with 8 kB block size is unmounted and you mount an SPIFFS with 4 kB block size, then the "SPIFFS_USE_MAGIC" feature will not recognize the SPIFFS and the system will be reformatted using the new block size.

chaitanyabd, 2018-11-25:
I've confirmed this behaviour causing trouble with OTAs. We have OTAs over 2G (yes!) which work well 95% of the time, the remaining 5% after a couple of retries. The problem is Arduino IDE-based deployments still go out with 8K blocks (though for now using mkspiffs utility for a server-side build reasonably resolves the issue for me).

Minor tweak suggestion for this issue: SPIFFS binary size should be the _minimum_ viable size based upon blocksize 4K + size of actual files required. Currently with 1 file size 2K, it (default) generates 64K files which is unnecessarily large, when 16K should be Ok as per the spec.

earlephilhower, 2018-12-06:
This is the kind of problem I saw when using the head MSPIFFS tool.  Version 0.2.0 of mkspiffs doesn't check this when passing in the size to the SPIFFS backend, so it silently (probably) rounds down to an integer number of pages.

I don't agree with @ABH10 's suggestion of going to 4K sectors, though, due to the data loss it would cause.  I'd recommend that boards.txt.py simply supplies the largest space possible while observing the block size defined.

earlephilhower, 2018-12-06:
SPIFFS will round down the number of blocks it thinks the FS has even if you pass in a non-integer multiple of blocksize due to its integer division in the macro:

https://github.com/pellepl/spiffs/blob/f5e26c4e933189593a71c6b82cda381a7b21e41c/src/spiffs_nucleus.h#L185

PR #5989 (2019-07-09): Make SPIFFS be an integer number of blocks
Fixes #3621

boards.txt.py simply calculated SPIFFS end and start using flash sizes, but
in cases where an 8K page was used (>512KB SPIFFS), this could leave a
4K half-block left at the end of SPIFFS.

mkspiffs and the SPIFFS code uses integer division to calculate the
maximum block number, so it worked fine in practice and the code simply
ignored the extra, fractional block.

Now actually take block size into account when calculating the end of
SPIFFS, ensuring no fractional blocks are passed in.  Does not result in
data loss on pre-existing SPIFFS f [... cut]
closed 2019-07-09 by d-a-v