esp8266/Arduino #3625: STA+STATION mode unconnectable to other esp8266 if SSID is hidden

Alasknnj, 2017-09-19:
### Basic Infos

#### Hardware
Hardware:		esp8266
Core Version:      	2.3.0

### Description

I have two nodeMCUs, one of them connects to a router network in WIFI_STA mode, then goes to WIFI_AP_STA mode and opens a hidden AP. The other nodeMCU then should connect to the hidden AP the first node created. 

The problem is, the second node cannot connect to the first node's hidden AP. 

I observed that if the first node doesn't connect to the router network, and open the hidden AP in WIFI_STA_MODE, the second nodes manages to connect. I also observed that even if the first node connects to the router network and opens the AP, if the AP is not hidden, the second node can connect. 

However the combination: first node connects to router network + opens hidden AP doesn't result in a successful connection for the second node, it just keeps trying to connect.

I have also been able to connect to the first node's hidden AP through my PC without any issue.

### Settings in IDE

Module:  			NodeMCU 1.0(ESP-12E Module)
Flash Size: 		4MB/3MB
CPU Frequency:	80MHz
Upload Speed:        115200
Upload Using:		Serial

### Sketch

I used the HTTPRequest example as a base to test this issue.

For the first node:
```cpp

#include <ESP8266WiFi.h>

const char* ssid = "ROUTER_SSID";
const char* password = "ROUTER_PASSWORD";

void setup() {
  Serial.begin(115200);
  //Serial.println();
  Serial.print("connecting to ");
  
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.println(WiFi.status());
 }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("Setting AP");
  WiFi.mode(WIFI_AP_STA);
  Serial.println(WiFi.softAP("AP_SSID", "AP_PASSWORD", WiFi.channel(), 1));

}

void loop() {

}
```
For the second node:

```cpp

/*
 *  HTTP over TLS (HTTPS) example sketch
 *
 *  This example demonstrates how to use
 *  WiFiClientSecure class to access HTTPS API.
 *  We fetch and display the status of
 *  esp8266/Arduino project continuous integration
 *  build.
 *
 *  Created by Ivan Grokhotkov, 2015.
 *  This example is in public domain.
 */

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "AP_SSID";
const char* password = "AP_PASSWORD";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.println(WiFi.status());
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

}

void loop() {

}
```

### output

First node:
```
connecting to ROUTER_SSID
.6
.3

WiFi connected
IP address: 
192.168.1.101
Setting AP
1

```

Second node:
```
connecting to BANANA
.6
.6
.6
.6
.6
.6
.6
.6
.6
.6
.6
.6
.6
.1
.1
.1
.1
.1
.1

...

```

devyte, 2017-09-26:
There is a known issue when connecting to an AP that has a hidden SSID. The changelist of SDK 2.1.0 shows a fix that seems related.
Please try the 2.1.0 branch, PR #3215 , to see if if it fixes your issue.

Alasknnj, 2017-09-28:
Are you familiar on how to try this branch through Platformio? I guess I would have to define it on the platformio.ini, but I don't think they have a way for me to use any branch freely, out of their framework system.

devyte, 2017-10-10:
@Alasknnj I don't use PlatformIO, I use the Arduino IDE for building and flashing, and gvim for code editing :p
I do believe that you can build against the git version. If so, then you should be able to build against a branch of the git version.

Alasknnj, 2017-10-10:
I have tried and unsuccessfully managed to do this by their structure. I really don't want to dig deep on how it works to do this, but I might just have to u.u

When I manage to do so, I will send a feedback about version 2.1.0 on regard of this issue

closed 2017-10-10 by Alasknnj

ufm, 2018-03-15:
The same trouble in SDK version 2.2.1(cfd48f3).
Unable to connect to esp8266 when esp work in AP_STA mode and AP have hidden SSID.