esp8266/Arduino #3626: Sending multipart data from HTTPCLIENT on ESP32

srinadhpotnuru, 2017-09-20:
I am looking for sending a multipart data(in my case to send a binary file) from ESP32 httpclient to a server. But in httpclient library I found only String data type in payload. Can anyone help or redirect to a correct issue thread.

Thanks

me-no-dev, 2017-09-20:
wait... ESP32? This is the ESP8266 repository. ESP32 does not include the HTTPClient yet.

srinadhpotnuru, 2017-09-20:
There is HTTPClient part of ESP32 arduino core.
https://github.com/espressif/arduino-esp32/tree/master/libraries/HTTPClient.

I thought this would be similar to ESP8266's httpclient's functionality.
Is it possible to send multipart data from ESP8266 or ESP32 HTTPClient

me-no-dev, 2017-09-20:
pfff it is and I totally forgot we added it.... anyway if your question is about ESP32 it should be submitted there and not here ;)

srinadhpotnuru, 2017-09-20:
Okay, I am adding it there

closed 2017-09-20 by srinadhpotnuru