esp8266/Arduino #3627: Got Disconnected when sent 3 messages in once through MQTT

tarzan115, 2017-09-20:
### Basic Infos

#### Hardware
Hardware:			`ESP-12F`
Core Version:      	`2.4.0-rc1`

###  Description

I use library [`ArduinoMQTT`](https://github.com/monstrenyatko/ArduinoMqtt) with SSL
I tested on MQTT use SSL when I sent 3 messages in once. The device just receives the first message and then disconnected.
I also tried in MQTT does not have SSL, it works fine.
My message has the length about `100 bytes` for each.

### Settings in IDE

Module:  			`Generic ESP8266 Module`
Flash Size: 		`4MB/1MB`
CPU Frequency:		`80Mhz`
Flash Mode:			`dio`
Flash Frequency:	`40Mhz`
Upload Using:		`SERIAL`
Reset Method:		`ck`

### Sketch

```cpp
#include "MQTT-Functions.h"
#include <MqttClient.h>
// #include <WiFiClient.h>

static MqttClient *mqtt = NULL;
static char *payload = NULL; 
static bool isHaveMessage = false;

#if WITH_SECURE
static WiFiClientSecure network;
#else
static WiFiClient network;
#endif // WITH_SECURE

class System: public MqttClient::System
{
  public:
      unsigned long millis() const
      {
        return ::millis();
      }

      void yield()
      {
        ::yield();
      }
};

static void
callback(MqttClient::MessageData& md)
{
  delay(0);

  const MqttClient::Message& msg = md.message;

  payload = (char*)calloc(msg.payloadLen + 1, 1);

  memcpy(payload, msg.payload, msg.payloadLen);
  payload[msg.payloadLen] = '\0';

  isHaveMessage = true;
  return;
}

static void
CallbackMqttLoop()
{
  if(isHaveMessage)
  {
    isHaveMessage = false;
    processJsonFromServer(payload);
  }
  return;
}

bool
reconnectMQTT()
{
  mqtt->disconnect();
  network.stop();
  if(!network.connect(MQTT_SERVER, MQTT_PORT))
  {
    mqtt->disconnect();
    network.stop();
    return false;
  }
  else
  {
    MqttClient::ConnectResult connectResult;
    {
      MQTTPacket_connectData options = MQTTPacket_connectData_initializer;
      options.MQTTVersion            = 4;
      {
        char clientID[25] = {0};
        sprintf(clientID, "%lu_", epoch);
        strcat(clientID, deviceSerial);
        options.clientID.cstring      = clientID;
      }
      options.cleansession            = true;
      options.keepAliveInterval       = 10;

      MqttClient::Error::type rc      = mqtt->connect(options, connectResult);
      if(rc != MqttClient::Error::SUCCESS)
      {
        return false;
      }
    }
    {
      if(!subscribeMQTT(MQTT_TOPIC_RECEIVE))
      {
        mqtt->disconnect();
        return false;
      }
    }
  }
  return true;
}

void
setupMQTT()
{
  MqttClient::System *mqttSystem      = new System;
  MqttClient::Logger *mqttLogger      = new MqttClient::LoggerImpl<HardwareSerial>(Serial);
  MqttClient::Network *mqttNetwork    = new MqttClient::NetworkClientImpl<WiFiClient>(network, *mqttSystem);
  MqttClient::Buffer *mqttSendBuffer  = new MqttClient::ArrayBuffer<300>();
  MqttClient::Buffer *mqttRecvBuffer  = new MqttClient::ArrayBuffer<300>();
  MqttClient::MessageHandlers *mqttMessageHandlers = new MqttClient::MessageHandlersImpl<15>();
  // MqttClient::MessageHandlers *mqttMessageHandlers = new MqttClient::MessageHandlersDynamicImpl<15>();
  MqttClient::Options mqttOptions;
  mqttOptions.commandTimeoutMs = 10000;
  mqtt = new MqttClient(mqttOptions, *mqttLogger, *mqttSystem, *mqttNetwork, *mqttSendBuffer, *mqttRecvBuffer, *mqttMessageHandlers);
  if(!mqttIsConnected())
  {
    reconnectMQTT();
  }
}

bool
loopMQTT()
{
  if(!mqttIsConnected())
  {
    Serial.println(F("disconnect"));
    if(reconnectMQTT())
    {
      sendJson_BACK_INFO();
    }
    else 
    {
      return false;
    }
  }
  mqtt->yield(1);
  CallbackMqttLoop();
  return true;
}

void
publishMQTT(const char* _Message, const bool _retained)
{
  delay(0);
  MqttClient::Message message;

  message.qos        = MqttClient::QOS0;
  message.retained   = _retained;
  message.dup        = false;
  message.payload    = (void *)_Message;
  message.payloadLen = strlen(_Message) + 1;

  mqtt->publish(MQTT_TOPIC_SEND, message);
}

bool
subscribeMQTT(const char* _Topic)
{
  MqttClient::Error::type rc = mqtt->subscribe(_Topic, MqttClient::QOS1, callback);
  return (rc == MqttClient::Error::SUCCESS);
}

bool
mqttIsConnected()
{
  return mqtt->isConnected();
}

```

### Debug Messages

```
Message 1
Feedback 1
disconnect
```

tarzan115, 2017-09-20:
https://github.com/monstrenyatko/ArduinoMqtt/issues/15

tarzan115, 2017-09-20:
#3335

igrr, 2017-12-27:
Closing as duplicate of https://github.com/esp8266/Arduino/issues/2256.

closed 2017-12-27 by igrr