esp8266/Arduino #3628: 用微信客户端给设备配网很难成功

wh17762392194, 2017-09-21:
硬件：nodemcu 0.9
核心版本 ：2.3.0
描述：
使用智能配网功能，我用微信客户端给设备配网很难成功。如果使用IOT_Espressif_ESPTouch软件可以轻易配网成功。可以优化这个问题吗。
我的代码如下：
void smartConfig()
{
  WiFi.mode(WIFI_STA);
     WiFi.beginSmartConfig();
  while (1)
  { 
    Serial.print(".");
    digitalWrite(LED, 0);
    delay(100);
    digitalWrite(LED, 1);
    delay(100);
    if (WiFi.smartConfigDone())
    {
      Serial.println("SmartConfig Success");
      Serial.printf("SSID:%s\r\n", WiFi.SSID().c_str());
      Serial.printf("PSW:%s\r\n", WiFi.psk().c_str());
      break;
    }
  }
}

devyte, 2017-09-21:
@wh17762392194 there is a lot of diversity among contributors and collaborators here, but the common language here for all is English.
I think this is the first time I have seen a question about Wechat, but in any case, this is not the right place to request help for your particular project. Please ask at esp8266.com or stackoverflow, or any other forum, for discussions and assistance.
Closing as off-topic.

closed 2017-09-21 by devyte

igrr, 2017-09-21:
There were some fixes/optimizations to SmartConfig in the new SDK version, you may try https://github.com/esp8266/Arduino/pull/3278 and see if it is any better. If not, please raise this issue on bss.espressif.com, as the smartconfig functionality is implemented inside the SDK. Arduino only provides a thin wrapper around it.