esp8266/Arduino #3630: ESP8266 Connecting but not getting IP address from Cisco router

BobGriswoldEmbedded, 2017-09-21:
We have been using the ESP8266 for an IoT product for some time but recently we have one customer who has a Cisco AIR-LAP1142N-A-K9 Aironet 1142 Controller-based AP. When my device tries to connect, it succeeds in connecting but does not get an IP address. The same program can connect to other APs in the same location but different models - non Cisco. We tried to connect to all 3 of the Cisco routers in the building on different channels with the same results. I only have access to the router settings if I have a specific question (I can't get a dump of the settings). I do have one of the same model of routers on order to try to duplicate the problem. I have searched for hours and tried everything I found to try.

I have two questions. 
1. Any ideas what might be causing it that I can try or questions I can ask to help narrow it down?
2. Where can I find what the little details in the debug output mean like state: 0 -> 2 (b0). See below. I am digging into the source code now but some direction would be helpful.

I am compiling this with the Arduino Genuino IDE version 1.6.12 and ESP8266 Community version 2.3.0.
Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

The security on the routers is WPA2

Here is the debug and log output:

Connecting to CiscoAP
..scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 14
cnt 

connected with CiscoAP, channel 1
dhcp client start...
wifi evt: 0
WIFI_EVENT_STAMODE_CONNECTED
........state: 5 -> 0 (0)
rm 0
wifi evt: 1
STA disconnect: 8
WiFi lost connection
f r0, ..scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 14
cnt 

connected with CiscoAP, channel 1
dhcp client start...
wifi evt: 0
WIFI_EVENT_STAMODE_CONNECTED
..........pm open,type:2 0
.........................

Which is generated from a very simplified program I created to make sure I was not doing something in the larger program to cause it:

#include <ESP8266WiFi.h>

const char* ssid     = "CiscoAP";
const char* password = "xxxxxx";

#define DEBUG_PRINT(x) Serial.println(x)

void setup() {
  Serial.begin(115200);
  delay(10);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.onEvent(WiFiEvent);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
}

int value = 0;

void loop() {
  delay(500);
  ++value;

if (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
else{
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

  if(value==10)
   {
    WiFi.disconnect();
    delay(1000);
     WiFi.begin(ssid, password);
   }
}

void WiFiEvent(WiFiEvent_t event) {
  IPAddress ip_addr;

  switch(event) {
    case WIFI_EVENT_STAMODE_CONNECTED:
        DEBUG_PRINT("WIFI_EVENT_STAMODE_CONNECTED");
      break;
      
    case WIFI_EVENT_STAMODE_DISCONNECTED:
        DEBUG_PRINT("WiFi lost connection");       
      break;
      
    case WIFI_EVENT_STAMODE_AUTHMODE_CHANGE:
        DEBUG_PRINT("WIFI_EVENT_STAMODE_AUTHMODE_CHANGE");
        break;

    case WIFI_EVENT_STAMODE_GOT_IP:
        DEBUG_PRINT("WiFi connected GOTIP");     
      break;
      
    default:
        DEBUG_PRINT("[WiFi-event] can not recognize event: " + String(event));
      break;
  }
}

Thanks,
Bob

BobGriswoldEmbedded, 2017-09-28:
More information about the problem.
1.	The Cisco 1140 is configured in Autonomous mode meaning it is not using a controller as the configuration entity.
2.	If you set the security settings on the Cisco 1140 AP for the SSID to be WPA TKIP only (Not WPA AES+TKIP) the problem we are seeing in the field is reproduced. See screenshot.
3.	I found an online reference to something that sounded similar with a fix but I cannot figure out how to carry out the fix because I cannot get the commands to work in the console. It is like it is a different OS
[http://itsnippet.com/2015/09/18/packet-to-client-xxxx-reached-max-retries/](url)

[image]

I am in the process of trying to capture all log messages and packets between the ESP8266 and the AP to help diagnose. I noticed a few posts through searching of people having TKIP problems with no solutions or a solution of going back to 2.0.0.0.

Any ideas would be appreciated.

maherahmed, 2017-10-22:
Any progress?

igrr, 2017-12-27:
Hi, this is probably a valid issue, but it is unlikely that one of the contributors to this repository will be able to reproduce it, as it requires some specific hardware. Feel free to use this issue for discussion or troubleshooting though.

artua, 2018-11-05:
Hi @BobGriswoldEmbedded 
Today I found similar issue: ESP can't get dhcp address from one of my Mikrotik networks.
Doing a lot of tests I found the cause: ESP doesn't get it's IP address in case of Hw. Fragmentation Threshold (maximum fragment size in bytes when transmitted over wireless medium) set below 340. 
With fragmentation threshold disabled or set higher then 340 it is connects and gets it's IP address again. 
Hope this helps.

GittiMcHub, 2019-06-04:
@artua,could you kindly share how to set 

for this code:
`WiFi.mode(WIFI_STA);
  WiFi.begin(run_wlan_ssid, run_wlan_pass);
  
  while (WiFi.status() != WL_CONNECTED) { // Wait for the Wi-Fi to connect
    delay(1000);
    Serial.print(".");
  }
  
  Serial.println("");
  Serial.println("initWifi | WiFi connected");
  Serial.println("initWifi | IP address: " + WiFi.localIP());`

if get this debug output

`del if1
usl
mode : sta(60:01:94:xx:xx:xx)
add if0
..scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with SensorWifi, channel 6
dhcp client start...
..........pm open,type:2 0
...........................................................................................................................................................................................................................................ip:169.254.198.24,mask:255.255.0.0,gw:0.0.0.0
.
initWifi | WiFi connected
nitWifi | IP address: 
`

artua, 2019-06-09:
You should set it on your wifi router

GittiMcHub, 2019-06-10:
I almost thought to myself, but makes sense - since I moved and got a new router from my provider, the connection doesn't work...
For anyone who stumbles across this problem, my router is provided by Vodafone 
Firmware-Version:           01.01.117.01.EURO
Firmware-Version:           AR01.01.117.01_091718_70.PC20.10
(I guess one version for Router FW and one for Webinterface, but it's not clearly seperated...)
AboutProductName:       Vodafone Docsis 3.1

timuroe, 2019-10-22:
same router same problem...

GittiMcHub, 2019-10-22:
It works since I got my <FRITZ!Box 6591 Cable>. The replacing workflow to use my own cable device is also straight foward with the doucmentation provided by Vodafone and Fritz. 
Everything works now since the first minute

devyte, 2019-11-10:
The original issue reported is specific to the Cisco hw. As stated above, current maintainers can't investigate it.
To anyone reading this, there are some router settings that aren't compatible with the ESP. Examples are channel width of 40MHz, WMM, and apparently fragmentation size below 340. These incompatibilities aren't related to the core, but rather to the ESP itself and possibly Espressif's SDK.
Closing.

closed 2019-11-10 by devyte

torsj, 2020-02-11:
This issue is not specific to the Cisco router. 
It is however related to the version of the EPS8266 module.
My latest batch of NodeMCU ESP modules all fail the same, previous batches work perfectly.

This old module works with all the tested WiFi routers:
MAC0 a97d0000
MAC1 0200d774
CHIPID 5a00b000
getChipId 00d774a9
WiFi.macAddress 18:FE:34:D7:74:A9

This new module fails with two of the tested WiFi routers:
MAC0 98b50000
MAC1 02006972
CHIPID c900b000
getChipId 00697298
WiFi.macAddress 50:02:91:69:72:98

I've tested with 5 different WiFi routers:
These fail:
deco tp-link M1300, linksys EA6350
These work:
dlink DIR-657, huawei CPE B593, jensen AL59300 v6

The attached files show my test program, logs from an old and a new esp module, and a summary of observations.

I've tried to modify router settings (such as bandwidth) without any success.

I've tried to use a fixed IP address on the ESP8266. Then it seems that transmit works but no data is received. This explains that dhcp never succeeds.

Please look into this problem. It is a real show stopper for my project.

[esp-wifitest-source.txt](https://github.com/esp8266/Arduino/files/4185946/esp-wifitest-source.txt)
[esp-wifitest-summary.txt](https://github.com/esp8266/Arduino/files/4185950/esp-wifitest-summary.txt)
[esp-wifitest-new.txt](https://github.com/esp8266/Arduino/files/4185952/esp-wifitest-new.txt)
[esp-wifitest-old.txt](https://github.com/esp8266/Arduino/files/4185954/esp-wifitest-old.txt)

baamiis, 2020-11-06:
we have thousands of devices out there and we are experiencing the exact problem with all the new routers. I have Just received a version Media HUB4 and it is exhibiting the same thing.
there is a setting in these new routers that stops the routers issuing DHCP address only to the ESP8266. I did try ESP32 with IDF and it works fine. However when I used same code in the ESP32 even the ESP32 failed. hence I think the issue or the bug is in the Ardruino Lib.
please help us resolve this issue we are all in panic mode 👍

d-a-v, 2020-11-06:
> However when I used same code in the ESP32 even the ESP32 failed

This is hardly understandable.

If you need fast support for your commercial application, we won't be able to help. Someone able to debug and fix needs to have such hardware to reproduce your issue and accept to dig/solve it during his/her time. Maybe you should ask on specialized platforms for such a person.

baamiis, 2020-11-06:
However when I used same code in the ESP32 even the ESP32 failed
sorry i should have been a bit more clear on that. 
what I mean is when I use the Arduino library and not IDF library I can connect with no problems.

BobGriswoldEmbedded, 2020-11-06:
Make sure the library is not trying to connect to an AP while it is trying to do a scan. There is a problem that when the system is trying to connect, the scan will fail. We forced it to stop trying to connect and then we could scan and get the AP and thy to connect. What we thought was a problem getting an IP was really a problem scanning for the AP. This may not be what you are doing but you need to be aware.

baamiis, 2020-11-07:
Thanks for the help.
From what I can see and using wireshark to sniff the dhcp packets. The device connected to the WiFi successfully but it is the dhcp server in the router not responding or not offering an IP address to the device.