esp8266/Arduino #3636: UDP sendPacket not returning errors

Hokahonay, 2017-09-22:
Problem description
not getting expected error codes from UDP.sendPacket()

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/3MB?
CPU Frequency:	?80Mhz?
Flash Mode:		?dio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL? both
Reset Method:		?ck

hi I am having problems getting error codes from UDP.sendPacket with UDP clien
using : 
`t#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <Streaming.h>

WiFiUDP port;

char ssid[] = "********";
char passWd[] = "********";

uint16_t UDPPort = 9999;
char rBuff[255];
float tmr = millis() + 20000;
char freda[25] = "itsssss meeeeeeeeee";
uint8_t err=0;

void setup() {
  Serial.begin(115200);
  Serial << endl;
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, passWd);
  while(WiFi.status() != WL_CONNECTED) {
    Serial << "."; delay(500);
  }
  Serial << endl << "Connected to: " << String(ssid) << endl;
  port.begin(UDPPort);
  Serial << "listening on IP: " << (WiFi.localIP().toString().c_str()) <<
  " Port: " << UDPPort << endl;
}

void loop() {
  int len = port.parsePacket();
  if(len) {
    int bytes = port.read(rBuff, 255);
    if(bytes > 0 ){
      rBuff[bytes] = 0;
      Serial << "Received: " << String(rBuff) << endl;
    }
  }
  if(millis() > tmr) {
    if(!port.beginPacket("192.168.4.1", UDPPort)) err |= 0X01;
    if(!port.write(freda, sizeof(freda))) err |= 0X02;
    if(!port.endPacket()) err |= 0X04;
    tmr = millis() + 20000;
    if(err) Serial << "Error: " << err << endl;
  }
}
`

all works fine sending to and receiving from the server
The problem comes if i switch the server off with the client still on
according to the documentation at http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/udp-class.html
after trying to send to the server i should have 3 errors registered i.e. err = 7
instead i am getting only one error, err = 4 indicating that endPacket failed but port.beginPacket() and port.write() had no errors
is this the correct behaviour ? if so can someone please explain what it is i am missing
what i need is a reliable way of ensuring that the client is still connected so that i can take some action if the server fails

thanks for any help
Dave

xsrf, 2017-09-26:
UDP does not provide a way to tell if a client is connected, because UDP is a CONNECTIONLESS protocol. Sending UDP packets only relies on the MAC Address of the destination being known by the IP protocol stack, but this is usually a cached value, thus exists even if the server is turned off (for some time).
You should choose TCP for your application or make your server respond with a value you can check.

Hokahonay, 2018-03-05:
thanks for the reply

closed 2018-03-05 by Hokahonay