esp8266/Arduino #3640: Wemos Board D1 mini Pro

Ciriaco08, 2017-09-23:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			Wemos Board D1 mini Pro
Core Version:      	2.3.0-rc

### Description

Problem description

This board have  16M bytes(128M bit) Flash and this Arduino environment. only recognize 4 MB

### Settings in IDE

Module:  			Wemos Board D1 mini Pro
Flash Size: 		16 MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Uttarwarsanket33, 2017-09-26:
Use ESP8266WIFI.h header file Instead of Arduino.h in your program.
you need to add library for Wemos Board D1 mini Pro!!

larsjohn, 2017-09-26:
How do you add that library?

Ciriaco08, 2017-09-27:
The Arduino software does no option , 16M Memory, with the ESP8266WIFI.h 
header file installed and Wemos board cheked.

Do not read the code sent is not representative

PR #3940 (2017-12-12): update and fix current generated boards.txt
fix 16M spiffs
less confusing lwip versions in menus
fix #3850
fix #3707
fix #3640
fix #3098
fix #2740
fix #2683
fix #2550
commit bd261c05 (2017-12-12): update and fix current generated boards.txt (#3940)

less confusing lwip versions in menus
fix #3850
fix #3707
fix #3640
fix #3098
fix #2740
fix #2683
fix #2550
closed 2017-12-12 by devyte