esp8266/Arduino #3641: Documentation for WiFi.persistent() is misleading

herm, 2017-09-23:
Documentation at http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html says
```Setting persistent to false will get SSID / password written to flash only if currently used values do not match what is already stored in flash.```

But the code does something different:
https://github.com/esp8266/Arduino/blob/35d5fabe33c24931bc629704b645119c0218a0ed/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L136-L151

It writes settings to flash if they don't match the values in flash if persistent is set to **true**. If it is false it will not write to flash independent of the current value in flash.

tablatronix, 2017-11-04:
All functions that use persistent and config equal or saves needs to have default/current conditional based on persistent. There are a few, I will make note of them.

tablatronix, 2017-11-05:
`ESP8266WiFiAPClass::softAP`
`ESP8266WiFiSTAClass::begin`
`ESP8266WiFiGenericClass::mode`

tablatronix, 2017-11-05:
This is what I am thinking

https://github.com/esp8266/Arduino/pull/3798

herm, 2017-11-05:
I'm not sure if the changes you propose are required or not. However they are unrelated to this issue. The problem here was that the code does something else than the documentation says. But the code's behavior is good so the documentation has to be changed.

tablatronix, 2017-11-05:
Then I will create another issue, because I believe it is wrong and badly implemented.

Raynman77, 2018-04-25:
@herm Just curious, whats the reason for the underscore in your line containing WiFi._persistent?

tablatronix, 2018-04-25:
That is the name of the variable used in the library.

Raynman77, 2018-04-25:
I haven't looked into it yet. Was looking at storing wireless credentials,
read the esp8266 core general class docs, and "WiFi.persistent(false)"
didn't contain the underscore. Was just curious if this was going to be a
"gotcha" when I finally come around to coding for the esp8266

Raynman77, 2018-04-25:
> No thats the private variable you have a function call your good

Cool, thanx Shawn

closed 2018-08-29 by devyte

tablatronix, 2018-08-29:
"Frequently calling these functions could cause wear on the flash memory"

Only if ! notconfigequals, perhaps we should add debugging strings when doing set config defualts to help debug that?

devyte, 2018-08-29:
Paranoia is your friend in these cases, so I agree.

tablatronix, 2018-08-29:
there is currently 
`   DEBUGV("sta config unchanged");`
but that is the opposite of what you would want
Ill open an issue and try to pr it in the 2-3 places that it happens, would help find really bad stuff like loops

PR #5081 (2018-08-29): doc: improve WiFi.persistent documentation
fixes #3641
commit 1ed7bee1 (2018-08-29): doc: improve WiFi.persistent documentation (#5081)

fixes #3641