esp8266/Arduino #3642: http requests over a public network

ejri, 2017-09-24:
Hi,

I am trying to connect an ESP8266 e-12 over a public network for http requests.

Usually to connect to the public network, I'd either have to connect manually to the network then authenticate using a log in/ password, or I can bypass the authentication by registering the MAC address of the device, then the device would automatically connect to the network. 

I tried registering the ESP's MAC address, and it seems to connect to the network but not authenticate. 

I am using the ESP8266HTTPClient library, and I get an error.  

/// HTTP client errors
#define HTTPC_ERROR_CONNECTION_REFUSED  (-1)    

The same http requests work fine over a private network. 

Any thoughts/ suggestions? 

Thank you.

suhajdab, 2017-10-12:
Any chance the public network has a captive portal?

ejri, 2017-10-13:
There was a problem with the mac address I was obtaining using the example in 
https://www.arduino.cc/en/Reference/WiFiMACAddress

For some reason, the addresses I was obtaining were reversed or something. 

I was able to obtain the mac addresses and register them using another example

#include <ESP8266WiFi.h>
void setup(){
   Serial.begin(115200);
   delay(500);
   Serial.println();
   Serial.print("MAC: ");
   Serial.println(WiFi.macAddress()); 
}
void loop(){}

closed 2017-10-13 by ejri