esp8266/Arduino #3644: ESP.restart() not working when GPIO-15 interrupt detected

HugoML, 2017-09-25:
### Basic Infos & Problem description
Using interrupt handler to count button press, which triggers ESP.restart(). 
But ESP.restart() does Not boot sketch. 

#### Hardware
Hardware:		ESP-8285
Core Version:      	2.3.0

### Settings in IDE

Module:  			ESP-8285
Flash Size: 		1MB
CPU Frequency:      80Mhz
Flash Mode:	        not applicable
Flash Frequency:	not applicable
Upload Using:		SERIAL
Reset Method:		not applicable

### Sketch

```cpp
#define PowerOnGPIO       15          // Power control of ESP. 
#define ButtonPressGPIO   13         // Button is attached to GPIO-13 
#define ButtonHoldTime     3000    // hold button for 3s or longer to go into RESET mode

volatile byte interruptCounter = 0; 
volatile unsigned long ButtonPressTimer;           
volatile bool ResetFlag = false;  

void setup() {
    Serial.begin(116200);
    Serial.println("\nButton Test Program Starts");
    pinMode(PowerOnGPIO, OUTPUT);
    digitalWrite(PowerOnGPIO, HIGH);                        // Keep ESP power supply
    pinMode(ButtonPressGPIO, INPUT);                        // Use interrupt to read button status 
    attachInterrupt(digitalPinToInterrupt(ButtonPressGPIO), handleButtonPressInterrupt, CHANGE);   
}

void loop() {
      Serial.print("interruptCounter=");
      Serial.println(interruptCounter);
      if (ResetFlag){
          Serial.print("Reset triggered. ESP.restart()");
          ESP.restart();
      }
}

void handleButtonPressInterrupt(){  
  interruptCounter++;  
  if(interruptCounter == 2){
    if(digitalRead(ButtonPressGPIO)== LOW)
        ButtonPressTimer = millis();                                    // start counting for button press down time  
   }  
   if(interruptCounter == 3) {
        if(digitalRead(ButtonPressGPIO) == HIGH){
            if (millis() - ButtonPressTimer < ButtonHoldTime )          
                  digitalWrite(PowerOnGPIO,LOW);               // turn off power supply for ESP
            else                                                                   // reset trigger
                  ResetFlag = true;                 
        }
        interruptCounter = 0;                                           // restart interrupt counter
   }   
}

```

### Debug Messages
Button Test Program Starts
interruptCounter=0
interruptCounter=0
...
interruptCounter=1
interruptCounter=1
interruptCounter=1
interruptCounter=1
...
interruptCounter=2
interruptCounter=2
interruptCounter=0
Reset triggered. ESP.restart()
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

```
Not sure whether GPIO-15 usage here as interrupt detection is proper here. It is used for boot loader mode selection. 
Any advice is appreciated. 
```

HugoML, 2017-09-25:
Tested above program on NodeMCU. 
ESP8266 (not ESP8285) restarts and boot into sketch. 

changed 
   #define ButtonPressGPIO   13 
to 
   #define ButtonPressGPIO   0      // the Flash button

Serial output shows: 

interruptCounter=2
interruptCounter=2
interruptCounter=0
Reset triggered. ESP.restart()
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
�
Button Test Program Starts
interruptCounter=0
interruptCounter=0
interruptCounter=0

I guess the ESP8285 has some issues in boot loading process. 
In addition, when press my button attached on GPIO-13 of ESP8285, a gentle press for half second will Not boot the sketch at all.

5chufti, 2017-09-25:
I would discourage using gpio15 as output and pulling high.
gpio15 is needed to be low for booting user code from spi-flash.

5chufti, 2017-09-25:
you're sure that using gpio15 as output and setting it high is ok?
For esp8266 gpio15 needs to be low for booting usercode from spi-flash.
Unfortunately I can't find any reference for esp8285 and boot sequence, as espressif allways references to esp8266 in single esp8285 document....

devyte, 2017-09-25:
Some gpio pins have special uses on bootup, as explained above. The pins may be different for the 8285. If there is no documentation about it available, you'll have to figure it out yourself, e.g. by trial and error.
Given that the code works on a Nodemcu, I don't see an issue in the core, so closing.

closed 2017-09-25 by devyte

HugoML, 2017-09-25:
OK. Thanks. 
The 8285 doc shows the PIN is identical to ESP8266. 
GPIO-15 set to HIGH is after sketch booted to setup(); 
It works so far.  
I guess I need to work on ESP8266.