esp8266/Arduino #3646: How to use server sent events?

andrewbourhis, 2017-09-25:
### Basic Infos

#### Hardware
Hardware:			Adafruit HUZZAH board
Core Version:      	?2.1.0-rc2?

### Description

Trying to have HUZZAH board perform some action, then once it is complete, have it post this data to the server without the client requesting the page be updated.

So the idea is this: Huzzah board runs a server, and when a client connects to this server, we send a webpage that is essentially a loading page. After sending this webpage to the client, the huzzah board will perform some action (i.e. run a function) and when it is complete, I want to be able to hide the loading spinner and unhide a textbox and button. I also want to send data to fill the textbox. I'm not entirely sure how to do this in html or javascript - does anyone have any ideas?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

//output LED used for debugging/signaling to user (also used for bootloading)
const int LEDPIN = 0;
//I2C pins
const int SDAPIN = 4;
const int SCLPIN = 5;
//address pins for MAX4734 (U3)
const int A0PIN = 16;
const int A1PIN = 14;

//DO NOT FORGET TO ENTER IN WIFI INFO HERE
const char* ssid = "********";
const char* password = "********";

MDNSResponder mdns;

ESP8266WebServer server(80);

const char CALIB_HTML[] =
"<!DOCTYPE html>"
"<html>"
"<FORM action='/sweep/' method='post'>"

"<h1>Performing calibration sweeps - please wait</h1>"

"<style>"
  ".loader {border: 16px solid #f3f3f3;border-radius: 50%;border-top: 16px solid #3498db;width: 120px;"
  "height: 120px;-webkit-animation: spin 2s linear infinite;animation: spin 2s linear infinite;}"
  
  "@-webkit-keyframes spin {0% { -webkit-transform: rotate(0deg); }"
    "100% { -webkit-transform: rotate(360deg); }}"
  
  "@keyframes spin {0% { transform: rotate(0deg); }"
    "100% { transform: rotate(360deg); }}"
"</style>"

"<div class='loader'></div>"

"<div style='display:none;' id='myDiv'>"
  "<h2> Done Calibrating! Please download calibration data</h1>"
  "<textarea cols='50' rows='10' id='source' name = 'gainFactors' readonly>empty calibration data</textarea>"
  "<INPUT type='submit' id='save' value='Download'>"
  "<script type='text/javascript'>"
    "document.getElementById('save').onclick = function(){"
      "var content = document.getElementById('source').value;"
      "window.open('data:application/txt,' + encodeURIComponent(content),'_self');}"
  "</script>"
"</div>"

"</FORM>"
"</html>";

void handleRoot()
{
  if (server.hasArg("gainFactors")) {
    handleSubmit();
  }
  else { //should run once on startup to load index page
    server.send(200, "text/html", CALIB_HTML);
    Serial.println("no gainFactors arg");
    delay(1000);
    sendCalibData();
  }
}

void returnFail(String msg)
{
  server.sendHeader("Connection", "close");
  server.sendHeader("Access-Control-Allow-Origin", "*");
  server.send(500, "text/plain", msg + "\r\n");
}

//this runs when we already loaded the index page once. Maybe should attempt to reload data that was previously entered (if any)
void handleSubmit()
{
  if (!server.hasArg("gainFactors")) return returnFail("BAD ARGS");
  sendCalibData();
}

//once calibration is complete for all sensors, we must upload this data to the server for them to download it
void sendCalibData(){
  if (server.hasArg("gainFactors")){
    //update textarea with gainfactors
    Serial.println("have gainFactors arg");
    String tmp = "data: test data that we could send";
    //char calibBuff[] = tmp
    server.send(200, "text/event-stream", tmp);
  }
  else{
    Serial.println(server.args());
    Serial.println(server.argName(0));
    server.send(200, "text/html",CALIB_HTML);
  }
}

void returnOK()
{
  server.sendHeader("Connection", "close");
  server.sendHeader("Access-Control-Allow-Origin", "*");
  server.send(200, "text/plain", "OK\r\n");
}

void handleNotFound()
{
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void writeLED(bool LEDon)
{
  // Note inverted logic for Adafruit HUZZAH board
  if (LEDon)
    digitalWrite(LEDPIN, 0);
  else
    digitalWrite(LEDPIN, 1);
}

void setup(void)
{
  
  pinMode(LEDPIN, OUTPUT);
  digitalWrite(LEDPIN,HIGH);//note, inverted logic for ESP8266 (this turns LED off)
  
  //Wire.pins(SDAPIN,SCLPIN);
  //Wire.begin();
  
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (mdns.begin("TuftsNanoLab_Server", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);
  server.onNotFound(handleNotFound);

  server.begin();
  Serial.print("Connect to http://TuftsNanoLab_Server.local or http://");
  Serial.println(WiFi.localIP());

}

void loop(void)
{
  server.handleClient();
}

andrewbourhis, 2017-09-25:
Ended up using auto-refresh of webpage to check for page updates

closed 2017-09-25 by andrewbourhis

vicnevicne, 2017-09-25:
Glad you found an alternative. If you're still looking for a permanent connection, don't know if server-sent events are implemented but Websockets is available and reliable. Example: http://adityatannu.com/blog/post/2016/01/31/ESP8266-Witty-Cloud-Board-Demo.html