esp8266/Arduino #3650: An error occurred while uploading the sketch with ARDUINO IDE 1.8.4 on NODEMCU ESP8266 module

Gromblisk, 2017-09-26:
##### Description
Hello! 
I'm trying to use <ESP8266WiFiMulti.h> and <FS.h> libraries in same project. However when I use SPIFFS.begin(); and server.handleClient(); I get error while uploding. It takes 5 minutes to upload to 99% and than error! If i upload program without SPIFFS.begin(); it uploads fine. Code and debug messages are below. I hope that someone can help me. 

##### Hardware
Hardware:		NodeMCU(ESP-12E Module), 80 MHz, 9600, 4M (3M SPIFFS)

##### Sketch
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266WebServer.h>
#include <FS.h> 

ESP8266WebServer server(12666);
ESP8266WiFiMulti wifiMulti;

boolean connectioWasAlive = true;
int pWifi = 13;  //D7
char *ESPssid = "ESP8266WIFI";
char *ESPpass = "ESP8266PASS";

//////////////////// **** SETUP **** ////////////////////
void setup() 
{
  pinMode(pWifi, OUTPUT);
  digitalWrite(pWifi, LOW);
  Serial.begin(9600);
  Serial.println("... and giberish STOP!");

  SPIFFS.begin();
  wifiMulti.addAP(ESPssid, ESPpass);
}

void loop() {
  monitorWiFi();
  server.handleClient();
}

void monitorWiFi()
{
  if (wifiMulti.run() != WL_CONNECTED)
  {
    if (connectioWasAlive == true)
    {
      connectioWasAlive = false;
      Serial.println("Searching for WiFi signal:");
    }
    Serial.print(".");
    digitalWrite(pWifi, !digitalRead(pWifi));
    delay(500);   
  }
  else if (connectioWasAlive == false)
  {
    connectioWasAlive = true;
    Serial.println("!");
    Serial.println("Server created!");
    digitalWrite(pWifi, HIGH);
  }
}

##### Debug Messages
Archiving built core (caching) in: C:\Users\Tom\AppData\Local\Temp\arduino_cache_394679\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_9600,FlashSize_4M3M_53e17bf51bb4f250e1f45a5c11d63252.a
Sketch uses 265713 bytes (25%) of program storage space. Maximum is 1044464 bytes.
Global variables use 36272 bytes (44%) of dynamic memory, leaving 45648 bytes for local variables. Maximum is 81920 bytes.
Uploading 269856 bytes from C:\Users\Tom\AppData\Local\Temp\arduino_build_24175/flash.ino.bin to flash at 0x00000000
................................................................................ [ 30% ]
................................................................................ [ 60% ]
................................................................................ [ 90% ]
.....................An error occurred while uploading the sketch

Also I have another issue. Code is uploading very slow. It takes like 5 minutes to upload simple code like this. I am wondering if this is normal.

Kind regards, 

Gromblisk

Uttarwarsanket33, 2017-09-26:
You can Increase the upload speed from menu>>tools>>upload speed>> 256000
It will significantly reduce upload time.
You can also choose greater speed but it may produce an error.

Gromblisk, 2017-09-26:
Thank you, this realy increases upload speed :D

Gromblisk, 2017-09-26:
woohoooo! It seems higher upload speed also solved my problem! It uploads code fine! 

Thank you very much :D

Uttarwarsanket33, 2017-09-26:
Welcome!

devyte, 2017-09-26:
9600 was likely causing a timeout somewhere.
Closing as resolved.

closed 2017-09-26 by devyte

furbanow, 2018-01-05:
Hi thank you, i was trying at 9600 or 115200 a,d i always got the same error. 
I switched to 57600 and the upload works fine

dhoneva, 2019-01-29:
Same problem was with me , Just change uploading speed to 115200 or higher.

anupriyaAmir, 2019-02-26:
Hi,
   My problem is uploaded 100% but not print anything in serial monitor .