esp8266/Arduino #3651: Speed Issues with page loading in Firefox

timofpv, 2017-09-26:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0-rc2

### Description

Hi I'm have speed issues with my SD Webserver. The loading of an html page with css takes about 10 seconds with Firefox  - with my Iphone Safari it goes way faster.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <SPI.h>
#include <SD.h>
#include <mcp_can.h>

#define DBG_OUTPUT_PORT Serial

const char *ssid = "Heisenberg";
const char *password = "Heisenberg";
IPAddress ip(192, 168, 0, 1);

static bool hasSD = false;

ESP8266WebServer server(80);

MCP_CAN CAN0(D2); 

void setup() {
  //Serielle Schnittstelle einstellen
	Serial.begin(115200);
  Serial.println("Serielle Schnittstelle initialisiert");
  // WiFi einstellen
	WiFi.mode(WIFI_AP);
 // WiFi.mode(WIFI_AP_STA);
  WiFi.softAP(ssid,password);
  WiFi.softAPConfig(ip, ip, IPAddress(255, 0, 0, 0));
  WiFi.setPhyMode(WIFI_PHY_MODE_11N);;
	server.onNotFound(handleNotFound);
	server.begin();
	Serial.println("HTTP server initialisiert");
  
  //SD-Card einstellen
  if (SD.begin(SS)){
     DBG_OUTPUT_PORT.println("SD Card initialisiert");
     hasSD = true;
  }
  //CAN-Schnittstelle einstellen
   if(CAN0.begin(CAN_500KBPS,MCP_8MHz) == CAN_OK) Serial.println("CAN initialisiert");
  else Serial.println("CAN Fehler");
}

void loop() {
	server.handleClient();
}

bool loadFromSdCard(String path){
  String dataType = "text/plain";
  if(path.endsWith("/")) path += "index.htm";

  if(path.endsWith(".src")) path = path.substring(0, path.lastIndexOf("."));
  else if(path.endsWith(".htm")) dataType = "text/html";
  else if(path.endsWith(".css")) dataType = "text/css";
  else if(path.endsWith(".js")) dataType = "application/javascript";
  else if(path.endsWith(".png")) dataType = "image/png";
  else if(path.endsWith(".gif")) dataType = "image/gif";
  else if(path.endsWith(".jpg")) dataType = "image/jpeg";
  else if(path.endsWith(".ico")) dataType = "image/x-icon";
  else if(path.endsWith(".xml")) dataType = "text/xml";
  else if(path.endsWith(".pdf")) dataType = "application/pdf";
  else if(path.endsWith(".zip")) dataType = "application/zip";

  File dataFile = SD.open(path.c_str());
  if(dataFile.isDirectory()){
    path += "/index.htm";
    dataType = "text/html";
    dataFile = SD.open(path.c_str());
  }

  if (!dataFile)
    return false;

  if (server.hasArg("download")) dataType = "application/octet-stream";

  if (server.streamFile(dataFile, dataType) != dataFile.size()) {
    DBG_OUTPUT_PORT.println("Sent less data than expected!");
  }

  dataFile.close();
  return true;
}

void handleNotFound(){
  
  // Send CAN message from POST
  int16_t ID = strtol(server.arg("ID").c_str(), NULL, 16);
  unsigned char DATA[8] = {0, 1, 2, 3, 4, 5, 6, 7};
  int8_t ML = strtol(server.arg("ML").c_str(), NULL, 16);
  DATA[0] = strtol(server.arg("D1").c_str(), NULL, 16);
  DATA[1] = strtol(server.arg("D2").c_str(), NULL, 16);
  DATA[2] = strtol(server.arg("D3").c_str(), NULL, 16);
  DATA[3] = strtol(server.arg("D4").c_str(), NULL, 16);
  DATA[4] = strtol(server.arg("D5").c_str(), NULL, 16);
  DATA[5] = strtol(server.arg("D6").c_str(), NULL, 16);
  DATA[6] = strtol(server.arg("D7").c_str(), NULL, 16);
  DATA[7] = strtol(server.arg("D8").c_str(), NULL, 16);
  if(ML > 0)
  {
  CAN0.sendMsgBuf(ID, 0, ML, DATA);
  }
  if(hasSD && loadFromSdCard(server.uri())) return;
  String message = "SDCARD Not Detected\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " NAME:"+server.argName(i) + "\n VALUE:" + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  DBG_OUTPUT_PORT.print(message);
}

RudyFiero, 2017-09-29:
What do you see with Firefox's Developer tools? 

What you provided here is insufficient. There are numerous factors and it often has to do with how you structured the web page you are serving. Different browsers will respond differently depending on what you have done. 

One common reason for slow web page loads is too many simultaneous requests to the ESP8266. It can only manage five connections. If the browsers are making more requests that the device can handle there will be delays. So the web page needs to be examined closely. 

Use the developer tools for the browser. They are very useful for determining what problems exist.

d-a-v, 2017-09-29:
@timofpv could you try lwip2 pull request #3362 with your sketch ?
I don't have your hardware, but replacing the sd datafile with a long string, it does not work (google-chrome and wget sitting and waiting), whereas they both run well with this pull request.

horendus, 2017-10-03:
I have done alot of work with ESP8266 as a webserver

The most likely cause of your long loading time is your Content-Length in the request header being larger than what your actually sending. 

Firefox tends to sit there waiting for the entire content-length ammount  where moble safari does not.

One option is to work out what content-length header your esp is sending and then work out what the actual size of your content is...and fix this problem. Perhaps do a size count on the html your sending from the sd card and make sure its passed into the content-length header.

Another option is to switch to a chunked transfer method where the content length header is 0 and the browser waits for a termiantion character to end the transmission (a bit more to it than that, reaserch it)

Chunked is the only way you will be able to send really Large websites without running into memory limitations.

devyte, 2017-10-22:
@timofpv will you be testing the requested PR?

mbm60, 2019-10-10:
Hi
@devyte @d-a-v
I have the same problem with both Firefox and Google Chrome.
But in "WIFI_AP_STA" mode, the speed is higher when the WiFi is connected.

**When wifi not connected:**
```
13:30:19.355 -> SoftAP Mode started. IP address: 192.168.1.1
13:30:19.355 -> Connecting WIFI to ssid : G5_MBM
13:30:22.219 -> Disconnected from SSID: G5_MBM
13:30:22.219 -> Reason: 201
13:30:29.959 -> handleFileRead: /index.htm
13:30:36.575 -> handleFileRead: /navigation-wifi.html
13:30:37.685 -> handleFileRead: /Acces_Point.html
13:30:37.991 -> handleFileRead: /header.html
13:30:38.159 -> handleFileRead: /style.css
13:30:46.035 -> handleFileRead: /tc2func.js
13:30:51.659 -> handleFileRead: /logo.gif
13:31:18.286 -> handleFileRead: /acc.json

```

**When wifi connected:**
```
13:33:52.397 -> SoftAP Mode started. IP address: 192.168.1.1
13:33:52.397 -> Connecting WIFI to ssid : G5_MBM
13:33:57.179 -> Disconnected from SSID: G5_MBM
13:33:57.179 -> Reason: 4
13:34:00.136 -> Connected to SSID : G5_MBM
13:34:03.193 -> handleFileRead: /index.htm
13:34:03.332 -> handleFileRead: /header.html
13:34:03.373 -> handleFileRead: /navigation-wifi.html
13:34:03.465 -> handleFileRead: /Acces_Point.html
13:34:03.754 -> handleFileRead: /logo.gif
13:34:04.003 -> handleFileRead: /style.css
13:34:04.244 -> handleFileRead: /tc2func.js
13:34:04.525 -> handleFileRead: /acc.json

```

devyte, 2019-11-06:
@mbm60 I'd say in AP_STA mode you're running into the STA side scanning for the router to connect to, which drags the AP side along for the channel hops. It can even result in the AP dropping the connected clients. That's an entirely different problem for which there is no fix, because it's a limitation of the single-radio nature of the ESP.

devyte, 2019-11-06:
This issue is old. A whole lot has been fixed since it was reported.
In AP mode, STA mode, or AP_STA mode with the STA side connected or disconnected (i.e.: not searching for the ssid to connect to), I don't see any slowness loading resources with latest git.
Closing.
If you think the problem still exists, please open a new issue and follow the template instructions.

closed 2019-11-06 by devyte