esp8266/Arduino #3657: 3.4.0 getcoreversion reports 00000000?

tablatronix, 2017-09-26:
Is this normal ? Is there something wrong with my platform environment ?

davisonja, 2017-09-26:
Are you using the latest version from git, or the latest release package?

tablatronix, 2017-09-26:
ahhhhh, I am using platformio staging, which uses repo not release.

So this is normal I assume

davisonja, 2017-09-26:
I believe the coreversion gets computed and stored as part of the release
build process, so this would be expected for repo usage.
I haven't looked, so I'm not sure how simple it would be to change that,
but if people are using direct from repo it might be useful if it could be
something more meaningful (so you can actually tell the version from code,
which I presume is the aim :)

igrr, 2017-09-26:
It should work for git version, if you are using Arduino IDE. There is a
hook in platform.txt which launches git to figure out the version.
Might be something that hasn't been ported over to platformio build script.

davisonja, 2017-09-26:
Ah, okay. The version I currently have in operation has `## windows-compatible version may be added later
` which is why it's not done that for me using Windows :)

tablatronix, 2017-09-26:
I see the hook, let me see if i can figure something out with platformio

tablatronix, 2017-09-26:
Ahhh bash..

tablatronix, 2018-01-28:
Looks like I am having issues with this on osx platformio with release
`ARDUINO_ESP8266_GIT_VER`

d-a-v, 2018-03-06:
I ran into this too.
I just discovered that, from inside core/esp8266/*.cpp
```#include <core_version.h>``` is well behaving
```#include "core_version.h"``` is not
The default file with generic default value lives in the same directory, the other one is generated by platform.txt's recipes in arduino's os-dependant temporary directory. I don't know details about platformio.

d-a-v, 2019-11-10:
Can you try [my alpha release channel](https://d-a-v.github.io/) to check if this issue is fixed ?

d-a-v, 2019-11-13:
This issue has received multiple corrections over time, latest related ones are: #6735, #6745 already fixed in git master, verified on the alpha release channel.
Closing. Please wait for core-2.6.1.

closed 2019-11-13 by d-a-v