esp8266/Arduino #3658: Exception (28): epc1=0x40201fc3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

slonkar, 2017-09-29:
### Basic Infos

I am trying to hookup hard wired security system in my house with HomeAssistant with the help of NodeMCU and MQTT. 

Below I have included sketch for my Security System nodemcu. Every time I run with multiple reed sensors I am getting following stack dump. I tried to decode stack trace but I was not able to get stack dump working.

IP address: 
192.168.1.38
Attempting MQTT connection...connected

Exception (28):
epc1=0x40201fc3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef580 end: 3ffef770 offset: 01a0

>>>stack>>>
3ffef720:  3fffdad0 3ffee540 3ffee71c 40203cf4  
3ffef730:  3fffdad0 3ffee540 3ffee540 3ffee748  
3ffef740:  3fffdad0 00000000 3ffee540 40202067  
3ffef750:  feefeffe feefeffe 3ffee740 40203a90  
3ffef760:  feefeffe feefeffe 3ffee750 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

#### Hardware
Hardware:		ESP8266 NodeMCU

### Description

If I run below sketch with only one ReedSensor setup, By calling checkStatus, publishStatus only with following defines
#define STATUS_TOPIC "SecuritySystemNodeMCU_1/GPIO5/Status"
#define FRONT_DOOR_STATUS_PIN 5
 
then it works as it supposed to. But as soon as I start dealing with those class objects I get above stack dump.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz
Upload Using:	SERIAL
Reset Method:        nodemcu

### Sketch
#include <PubSubClient.h>
#include <ESP8266WiFi.h>

class ReedSensor {
  public:
    int pin;
    int last_status;
    unsigned long last_switch_time;
    String mqtt_topic;
  };

#define wifi_ssid "--------------"
#define wifi_password "-------------"

#define mqtt_server "-------------"
#define mqtt_user "-------------"
#define mqtt_password "-------------"

// Window Topics
#define DINNING_ROOM_WINDOW_TOPIC "SecuritySystemNodeMCU_1/GPIO14/Status"
#define FAMILY_ROOM_WINDOW_TOPIC "SecuritySystemNodeMCU_1/GPIO12/Status"
#define KITCHEN_WINDOW_TOPIC "SecuritySystemNodeMCU_1/GPIO13/Status"

// Door Topics
#define FRONT_DOOR_TOPIC "SecuritySystemNodeMCU_1/GPIO5/Status"
#define BACK_DOOR_TOPIC "SecuritySystemNodeMCU_1/GPIO4/Status"
#define DECK_DOOR_TOPIC "SecuritySystemNodeMCU_1/GPIO0/Status"

#define DINNING_ROOM_WINDOW 14  // Gree Wire
#define FAMILY_ROOM_WINDOW 12   // Blur Wire
#define KITCHEN_WINDOW 13       // Purple Wire
#define FRONT_DOOR 5           //Black Wire
#define BACK_DOOR 4             //White Wire
#define DECK_DOOR 0             // Gray Wire

ReedSensor *dinning;
ReedSensor *family;
ReedSensor *kitchen;
ReedSensor *front;
ReedSensor *back;
ReedSensor *deck;

WiFiClient espClient;
PubSubClient client(espClient);

int debounceTime = 2000;
int lastStatusValue = 2;
unsigned long lastSwitchTime = 0;

#define STATUS_TOPIC "SecuritySystemNodeMCU_1/GPIO5/Status"
#define FRONT_DOOR_STATUS_PIN 5

void setup_wifi(){
  delay(10);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(wifi_ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(wifi_ssid, wifi_password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void init_reed_sensor(ReedSensor *senor, int pin, String mqtt_topic) {
  senor = new ReedSensor();
  senor->pin = pin;
  senor->last_status = 2;
  senor->last_switch_time = 0;
  senor->mqtt_topic = mqtt_topic;
}

void setup_reed_switches() {

  init_reed_sensor(dinning, DINNING_ROOM_WINDOW, DINNING_ROOM_WINDOW_TOPIC);
  init_reed_sensor(family, FAMILY_ROOM_WINDOW, FAMILY_ROOM_WINDOW_TOPIC);
  init_reed_sensor(kitchen, KITCHEN_WINDOW, KITCHEN_WINDOW_TOPIC);
  init_reed_sensor(front, FRONT_DOOR, FRONT_DOOR_TOPIC);
  init_reed_sensor(back, BACK_DOOR, BACK_DOOR_TOPIC);
  init_reed_sensor(deck, DECK_DOOR, DECK_DOOR_TOPIC);

  //Window
  pinMode(DINNING_ROOM_WINDOW, INPUT_PULLUP);  
  pinMode(FAMILY_ROOM_WINDOW, INPUT_PULLUP);
  pinMode(KITCHEN_WINDOW, INPUT_PULLUP);

  //Doors
  pinMode(FRONT_DOOR, INPUT_PULLUP);  
  pinMode(BACK_DOOR, INPUT_PULLUP);
  pinMode(DECK_DOOR, INPUT_PULLUP);
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("SecuritySystemNodeMCU_1", mqtt_user, mqtt_password)) {
      Serial.println("connected");
      client.subscribe("SecuritySystemNodeMCU_1/#");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(500);
    }
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println("Booting Security System NodeMCU 1");  
  
  //Setup WiFi
  setup_wifi();

  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  //Setup Reed Switches
  setup_reed_switches();
}

// the loop function runs over and over again forever
void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
  checkStatusForReedSensors();
}

void checkStatusForReedSensors() {
    // Windows
    checkStatusOfReedSensor(dinning);
    checkStatusOfReedSensor(family);
    checkStatusOfReedSensor(kitchen);
    
    //Doors
    checkStatusOfReedSensor(front);
    checkStatusOfReedSensor(back);
    checkStatusOfReedSensor(deck);
    //checkStatus(FRONT_DOOR_STATUS_PIN);
  }

void checkStatusOfReedSensor(ReedSensor *sensor) {
    int currentStatusValue = digitalRead(sensor->pin);
    if (currentStatusValue != sensor->last_status) {
      unsigned int currentTime = millis();
      if (currentTime - sensor->last_switch_time >= debounceTime) {
        publishStatusForReedSensor(sensor);
        sensor->last_switch_time = currentTime;
        sensor->last_status = currentStatusValue;
      }
    }
  }
  
void publishStatusForReedSensor(ReedSensor *sensor) {
    const char *stateTopic = (sensor->mqtt_topic).c_str();
    if(digitalRead(sensor->pin) == LOW) {
      client.publish(stateTopic, "closed", true);
      Serial.print(stateTopic);
      Serial.println("Closed");
    } else {
      client.publish(stateTopic, "open", true);
      Serial.print(stateTopic);
      Serial.println("Open");
    }
}

void checkStatus(int SENSOR_PIN) {
  int currentStatusValue = digitalRead(SENSOR_PIN);
  if (currentStatusValue != lastStatusValue) {
    unsigned int currentTime = millis();
    if (currentTime - lastSwitchTime >= debounceTime) {
      publishStatus(SENSOR_PIN);
      lastStatusValue = currentStatusValue;
      lastSwitchTime = currentTime;
    }
  }
}

void publishStatus(int SENSOR_PIN) {
    const char *stateTopic = STATUS_TOPIC;
    if(digitalRead(SENSOR_PIN) == LOW) {
      Serial.print("Door Closed -");
      Serial.print(SENSOR_PIN);
      Serial.println(" .....");
      client.publish(stateTopic, "closed", true);
      Serial.print(stateTopic);
    } else {
      Serial.print("Door Opened");
      Serial.print(SENSOR_PIN);
      Serial.println(" .....");
      client.publish(stateTopic, "open", true);
      Serial.print(stateTopic);
    }
}

void callback(char* topic, byte* payload, unsigned int length) {
  payload[length] = '\0';
  String strTopic = String((char*)topic);
  String payloadValue = String((char*)payload);
}

### Debug Messages

IP address: 
192.168.1.38
Attempting MQTT connection...connected

Exception (28):
epc1=0x40201fc3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef580 end: 3ffef770 offset: 01a0

>>>stack>>>
3ffef720:  3fffdad0 3ffee540 3ffee71c 40203cf4  
3ffef730:  3fffdad0 3ffee540 3ffee540 3ffee748  
3ffef740:  3fffdad0 00000000 3ffee540 40202067  
3ffef750:  feefeffe feefeffe 3ffee740 40203a90  
3ffef760:  feefeffe feefeffe 3ffee750 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

devyte, 2017-10-03:
@slonkar this:
void init_reed_sensor(ReedSensor *senor, int pin, String mqtt_topic) {
senor = new ReedSensor();
senor->pin = pin;
senor->last_status = 2;
senor->last_switch_time = 0;
senor->mqtt_topic = mqtt_topic;
}

is wrong. You are assigning the result of new to the local argument sensor, not the global ptr. Why are you creating the objects with new? There is no need to do that. You can create static objects with a begin() method that does the configuration, or assign a temp constructed object.

Closing as language usage error. If you need further assistance, please refer to a community forum, see #3655 .

closed 2017-10-03 by devyte

sandevid, 2019-03-11:
### How to fix this , i use nodeMCU
**Error :**
Exception (28):
epc1=0x4020a5d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000001bb depc=0x00000000

stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 01a0
3fffff30: 3fffdad0 00000000 3ffeed38 4020261b
3fffff40: 000001bb 3ffeed44 3ffe8884 3ffeeedc
3fffff50: 000001bb 3ffeed44 3ffe8884 40204a42
3fffff60: 40207ec0 dca79a95 3ffeed44 40203cbf
3fffff70: 3fffdad0 3ffeee10 3ffeed38 402057f0
3fffff80: 3fffdad0 3ffeee10 3ffeee64 402025d5
3fffff90: 40207ec0 552ba8c0 feefeffe feefeffe
3fffffa0: feefeffe 00000000 3ffeeeac 40207550
3fffffb0: feefeffe feefeffe 3ffe8558 40100a0d
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(1,6)

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

**Source Code :**

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <TelegramBot.h>

#define LED 14 //D3

const char* ssid = "Telegram";
const char* password = "getupdate";
const char BotToken[] = "660345429:ACGlSKhCRMmgKHNJH4pG3ut_1R_66flenqQ";

WiFiClientSecure net_ssl;
TelegramBot bot (BotToken, net_ssl);

void setup() {
Serial.begin(115200);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("WiFi connected");
Serial.println(WiFi.localIP());

bot.begin();

pinMode(LED, OUTPUT);
}

void loop() {

message m = bot.getUpdates();

if (m.text.equals("on")){
digitalWrite(LED, HIGH);
bot.sendMessage(m.chat_id, "The Led 1 is now ON");

}else if (m.text.equals("off")){
digitalWrite(LED, LOW);
bot.sendMessage(m.chat_id, "The Led 1 is now OFF");
}
}`

Board : NodeMCU 1.0 (ESP-12E Module)
Upload Speed : 115200
Flash Size : 4M (no-SPIFFS)
Debug Port : Disabled
Debug level : None
IwIP Variant : v2 lower memoty
VTables : Flash
Exceptions : Disable
Erase Flash : Only Sketch
Port : COM4

Please help,
I have searched everywhere but there is no right solution

naimulh247, 2022-09-06:
@slonkar were you able to fix this? I am using a similar version of the code you have provided but running into the same error.