esp8266/Arduino #3659: v2.4.0rc1: Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads

freeck, 2017-09-29:
The application work fine with 2.3.0. 
When using 2.4.0 it crashes repetitively at certain point , see dump.

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		esp8266mod
Core Version:      	2.4.0-rc1

### Description

The application work fine with 2.3.0. In case of using 2.4.0. It crashes (restarts) repetitively at certain point.

> Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

too big....

### Debug Messages

ets Jan  8 2013,rst cause:1, boot mode:(1,7)
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 20 results
0x402122aa: _strtol_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtol.c line 152 (discriminator 1)
0x4020f2d8: yield at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57
:  (inlined by) __yield at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 96
0x4020f320: optimistic_yield at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57
0x40209beb: WiFiClient::available() at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 324
0x402123e4: strtol at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtol.c line 224
0x4020feac: atoi at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/atoi.c line 71
0x40204844: ftpClient::setPasvMode() at M:\esp8266\Libraries\ftpClientSYNC/ftpClient.cpp line 119
0x40201700: micros_overflow_tick at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_wiring.c line 58
0x40202968: Controller::FtpPutFilesToServer() at M:\workspace\Smartmeterv2_3\esp8266/..\Controller.cpp line 368
0x40106d80: delayMicroseconds at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_wiring.c line 76
0x40204c6c: TwoWire::endTransmission(unsigned char) at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\libraries\Wire/Wire.cpp line 252
:  (inlined by) TwoWire::endTransmission() at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\libraries\Wire/Wire.cpp line 143
0x4020dc4d: LiquidCrystal_I2C::write(unsigned char) at C:\Program Files (x86)\Arduino\libraries\Arduino-LiquidCrystal-I2C-library-master/LiquidCrystal_I2C.cpp line 189
0x40106d80: delayMicroseconds at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_wiring.c line 76
0x40202cc3: Controller::Scheduler() at M:\workspace\Smartmeterv2_3\esp8266/..\Controller.cpp line 183
0x4020176a: delay at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_wiring.c line 52
0x401004b1: loop at M:\workspace\Smartmeterv2_3\esp8266/..\main.cpp line 45
0x4020f23b: loop_wrapper() at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57
0x40101084: cont_norm at C:\sloeber\arduinoPlugin\packages\esp8266-2.4.0-rc1\cores\esp8266/cont.S line 109

igrr, 2017-10-06:
Thank you for reporting this. Could you please try to isolate the issue and provide a small sketch which demonstrates it?

hemangjoshi37a, 2017-10-07:
Where to get version 2.3.0? Can you please provide any download link?

freeck, 2017-10-15:
Same issue with 2.4.0-rc2. I will try to isolate the relevant parts of the application, and will report the results.

freeck, 2017-10-18:
Hyu

freeck, 2017-10-18:
The exception occurred "around"  a rename-statement (of the SDFAT-library). I renamed/moved a file to another directory where a file with the same name already resided. Normally the rename fails with return-code false.....it looks as if the system now crashes at this point. 
I was not able to reproduce this error when the code is executed in another context.
My workaround is:  first to test whether a file with the same name already exists in the destination directory. If so: delete this file, then move the source-file to the destination.
Why release 2.4.0 rc1 and rc2 seems to introduce this crash is not clear to me yet, I will search further on...

freeck, 2017-10-19:
Well, it was an application error in my ftpClient. It seems that due to a different timing of the response of the socket-connections with release v2.4.0-rc2, the handshake of the ftpClient behaved differently than with v2.3.0. The response  of the ftpserver was not completed which resulted in a conversion error in function strtol (I have to admit: bad programming of my ftp-client).
 See crash dump: newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtol.c line 152 (discriminator 1).
What a great tool is the Exception decoder!!!

devyte, 2017-10-19:
Can you please explain the error with strtol better?

freeck, 2017-10-19:
Yes of course. My explanation was not adequate enough.
The ftp-client was waiting for an IP-number and port number of the ftp-server's data connection. As the returned string was timed-out too quickly (programming error) the conversion of the expected IP-number (octets) failed: strtol() tried to convert the string to a number which led to the exception.

NOTE: it seems the socketconnection of v2.4.0 is slower than v2.3.0.....?

closed 2017-10-24 by freeck