esp8266/Arduino #3660: Accesspoint / Security issue / AC Point don't disapear after upload of an other sketch

f41ardu, 2017-09-29:
### Basic Infos

#### Hardware
Hardware:			NodeMCU 1.1
Core Version:      	-----

### Description
Problem description

Upload Access Point Example. 
-- works fine 
But after Upload of the blink example 
blink example work fine, but the Accesspoint still active and visible. 

Seems to be a security issue. I expect after upload of different code 
that the Wifi code is overwritten.

f41ardu, 2017-09-30:
Solution: 
.....
/* 
 * Add WiFi.mode(WIFI_STA) 
 * before WiFi.begin(....) 
 * / 
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
.....

devyte, 2017-09-30:
@f41ardu I believe this is on purpose. The SDK stores the wifi credentials, among other info, in a flash area that is separate from the sketch itself. The stored credentials are used by the SDK for fast wifi start when booting, before even the setup function is called. Also, I remember a discussion about making sure the credentials are kept after update, i.e.: you update your sketch with a newer version of it, and the new version uses the credentials already stored.
If you wish to make sure this doesn't happen, you should explicitly erase the entire flash with esptool before flashing a different sketch, like the blink example in your case.
Closing as non-issue.

closed 2017-09-30 by devyte