esp8266/Arduino #3662: WDT reset when using modem sleep

trentbrown13, 2017-09-30:
### Basic Infos

#### Hardware
Hardware:		Wemos d1 mini
Core Version:      	2.4.0-rc1

### Description

I am using modem sleep with the wemos D1 mini which works fine with 2.3.0. After upgrading to 2.4.0-rc1 I get a wdt reset immediately after issuing "WiFi.forceSleepBegin();" I later downgraded back to 2.3.0 and it works fine again. One note, my free heap reported by "ESP.getFreeHeap(); seemed to be much lower when using 2.4.0-rc1 but I was primarily trying to isolate the wdt reset cause. I only noticed this as I am chasing a memory leak problem and printing heap size all over the place.

### Settings in IDE

Module:  			Wemos D1 R2 mini
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>  // for mqtt
#include <OneWire.h>
#include <DallasTemperature.h>
#include <Wire.h>  
#include "SSD1306.h"
#include <ESP8266mDNS.h> 
#include <ArduinoOTA.h>
#include <WiFiUdp.h> 
#include <cactus_io_BME280_I2C.h>
// ntp includes
#include <TimeLib.h>
//#include "WifiConfig.h" 
#include <NtpClientLib.h>

void setup() {

}

void loop() {
************ snip *************
 getTemperatures();
     
     client.publish("/TbOffice/Tub/temperature", tubTempFString);
     client.publish("/TbOffice/Out/temperature", outTmpFString);

    Serial.print(" End loop number: ");
    Serial.println(x);
   
  lcdDisplayTemps();
  yield();
  } // end for loop

 Serial.println("Going down for a 60 second nap");
 yield();
 //delay(5000);
 //Serial.println(F("just before Wifi.forceSleepBegin"));
  WiFi.forceSleepBegin();

  ///////////// NEVER GETS HERE when using 2.4.0-rc1
  
 //Serial.println(F("just after Wifi.forceSleepBegin"));
 delay(1 * sleepM); // 10 seconds for test
}
************** end snip ***************
```

### Stack Trace
Soft WDT reset

ctx: cont 
sp: 3fff0990 end: 3fff0c00 offset: 01b0

>>>stack>>>
3fff0b40:  0001260d 00000600 3ffef794 4021ad48  
3fff0b50:  0fffffff 40203d5e 00000000 40203d49  
3fff0b60:  3ffefb18 3ffef914 3ffef794 40203452  
3fff0b70:  00000000 00000000 00000000 4020a77c  
3fff0b80:  00000000 00000000 00000000 00000000  
3fff0b90:  00000000 00000000 00000000 00000000  
3fff0ba0:  00000000 00000000 00000000 00000000  
3fff0bb0:  3ffe9730 b564a8c0 3ffe9730 b564a8c0  
3fff0bc0:  00000006 42955b23 4479c281 feefeffe  
3fff0bd0:  feefeffe feefeffe feefeffe 3ffefbd8  
3fff0be0:  3fffdad0 00000000 3ffefbd0 4020b4b0  
3fff0bf0:  feefeffe feefeffe 3ffefbe0 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vf6d232f1
~ld

Decoding 8 results
0x4021ad48: ieee80211_setup_ratetable at ?? line ?
0x40203d5e: ESP8266WiFiGenericClass::enableSTA(bool) at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp line 96
0x40203d49: ESP8266WiFiGenericClass::getMode() at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp line 96
:  (inlined by) ESP8266WiFiGenericClass::enableSTA(bool) at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.cpp line 329
0x40203452: loop at D:\Programming\ESP\Weather nodes\Trents_Office_Msleep1G/Trents_Office_Msleep1G.ino line 738 (discriminator 3)
0x4020a77c: String::changeBuffer(unsigned int) at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 720
0x4020b4b0: String::move(String&) at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 205
:  (inlined by) String::String(String&&) at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 51
0x4010070c: cont_resume at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 99

igrr, 2017-10-06:
Sorry, i tried running the sketch you provided, but it does not compile. Could you please check and make sure that the sketch can be compiled and reproduces the issue? Thanks.

trentbrown13, 2017-10-07:
Sorry Ivan, that was just a code snippet. Here is another example

#include <TimeLib.h>
#include "WifiConfig.h"
#include <NtpClientLib.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
//#include <DHT.h>
#define DEBUG 1

//#define DHTTYPE DHT11   // DHT 11
//#define DHTTYPE DHT21   // DHT 21 (AM2301)
//#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321

extern "C" {
  #include "gpio.h"
}

extern "C" {
#include "user_interface.h" 
}
/*#ifndef WIFI_CONFIG_H
#define YOUR_WIFI_SSID BlueZebra
#define YOUR_WIFI_PASSWD jY7bzy2XJv
#endif // !WIFI_CONFIG_H
*/

#ifndef WIFI_CONFIG_H
#define YOUR_WIFI_SSID "YOUR_WIFI_SSID"
#define YOUR_WIFI_PASSWD "YOUR_WIFI_PASSWD"
#endif // !WIFI_CONFIG_H

//const char* YOUR_WIFI_SSID = "BlueZebra";
//const char* YOUR_WIFI_PASSWD = "jY7bzy2XJv";
const char* ssid = "BlueZebra";
const char* password = "jY7bzy2XJv";
//const char* ssid = "SRGTEST_2G";
//const char* password = "5martRG!";

static int connectAttempts = 0;
static int connectPass = 0;
static int connectFail = 0;
long previousMillis = 0;
long interval = 5100;

const char* ahostname = "MQTT_Test";
//#define ONBOARDLED 2 // Built in LED on ESP-12/ESP-07
//#define WIFI_PASS_LED 5
//#define WIFI_FAIL_LED 4

#define SLEEPM  60000  // one minute

//int8_t timeZone = 1;

int8_t timeZone = -8; // PST

// Start NTP only after IP network is connected
void onSTAGotIP(WiFiEventStationModeGotIP ipInfo) {
  //NTP.stop();
//pinMode(WIFI_PASS_LED, OUTPUT);
//pinMode(WIFI_FAIL_LED, OUTPUT);
	Serial.printf("Got IP: %s\r\n", ipInfo.ip.toString().c_str());
  connectPass ++;
  //digitalWrite(WIFI_PASS_LED, HIGH);
  //delay(1000);
  //digitalWrite(WIFI_PASS_LED, LOW);
  //Serial.println("Just before NTP.begin - CRASHES HERE");
	NTP.begin("pool.ntp.org", timeZone, true);
  //Serial.println("Just after NTP.begin and before ntp.setInterval");
	NTP.setInterval(63);
[... 208 lines cut ...]
   // delay(1000);
    //digitalWrite(WIFI_PASS_LED, LOW);
    //pinMode(ONBOARDLED, OUTPUT); // Onboard LED
    //digitalWrite(ONBOARDLED, HIGH); // Switch off LED
    delay(500);
    #ifdef DEBUG
    Serial.println(F("Just before on NTP sync event"));
    Serial.println("Free Heap      = " + String(ESP.getFreeHeap()));
    #endif
    NTP.onNTPSyncEvent([](NTPSyncEvent_t event) {
    ntpEvent = event;
    syncEventTriggered = true;
    #ifdef DEBUG
    Serial.println(F("Just after sync event triggered"));
    Serial.println("Free Heap      = " + String(ESP.getFreeHeap()));
    #endif
  });

    //e1 = WiFi.onStationModeGotIP(onSTAGotIP);// As soon WiFi is connected, start NTP Client
 // e2 = WiFi.onStationModeDisconnected(onSTADisconnected);

		
	}
	//delay(0);
}

trentbrown13, 2017-10-18:
Not sure if you wouold like like a new ticket opened or just add to this one. This time running 2.3.0 with a much simplified sketch.
stack Decode:

{code}
decoding 21 results
0x40104927: lmacRxDone at ?? line ?
0x401004d8: malloc at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40107084: pvPortMalloc at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 13
0x40222670: pbuf_alloc at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 366
0x4022550e: udp_sendto at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/udp.c line 450
0x40222b9f: sntp_initialize_request at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/sntp.c line 698
:  (inlined by) sntp_send_request at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/sntp.c line 885
0x40222bb3: sntp_send_request at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/sntp.c line 889
0x40105cdc: sntp_dns_found at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/sntp.c line 917
0x40221cf5: dns_recv at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dns.c line 825
0x40221d04: dns_recv at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dns.c line 826
0x40210400: scan_parse_beacon at ?? line ?
0x402252a8: udp_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/udp.c line 343
0x40107084: pvPortMalloc at C:\Users\Windows7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 13
0x402269e4: ip_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip.c line 553
0x40225ded: ethernet_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c line 1379
0x40205faf: pp_tx_idle_timeout at ?? line ?
0x40205f22: pp_tx_idle_timeout at ?? line ?
0x4021a01b: ets_snprintf at ?? line ?
{code}

Sketch:
{code}

#define HEAPCHECKER 1        // set to 1 to test HEAP loss fix
#ifdef HEAPCHECKER
  #include "lwip/tcp_impl.h" // losing bytes work around
#endif
#include <TimeLib.h>
//#include "WifiConfig.h"
#include <NtpClientLib.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#define DEBUG 1
#define NTP_ON 1

extern "C" {
#include "gpio.h"
}

extern "C" {
#include "user_interface.h"
}

int count_sleep = 0;
int count_wake = 0;
uint32_t originalram;

//*************** SSID Stuff *********************************
const char* ssid = "BlueZebra";
const char* password = "jY7bzy2XJv";
//const char* ssid = "SRGTEST_2G";
//const char* password = "5martRG!";
//**********************************************************************

//******************* Connect Attempt ***********************************
static int connectAttempts = 0;
static int connectPass = 0;
static int connectFail = 0;
//*********************************************************************

//************** Millis stuff *****************************************
unsigned long previousMillis = 0;
const unsigned long sleeptime = 30000; // milliseconds
[... 250 lines cut ...]
       delay(1000);
       Serial.println(F("*************** Still sleeping but checking TIME **********************"));
       Serial.print(F("sleeping for  = "));
       Serial.print(sleeptime /1000);
       Serial.print(F(" seconds, Time Elapsed = "));
       Serial.print((currentMillis - previousMillis)/1000);
       Serial.println(F(" seconds"));
       Serial.println(F("***********************************************************************"));
     }
         if ((unsigned long)(currentMillis - previousMillis) >= sleeptime) {
         Serial.println(F("**************Time to Wake up and get to work!******************"));
         Serial.println(F("************** awake = TRUE ******************"));
             awake = true;
             count_wake += 1;
             Serial.print(F("Wake count = "));
                 Serial.println(count_wake);
                 Serial.println(F("**************** Waking up WiFi ***************"));
                 WiFi.forceSleepWake();
                 previousMillis = millis();
               }
   }
{code}

This crash appears at random times, perhaps I need to reformat the flash?
Please let me know if I should add updates etc in a better manor

trentbrown13, 2017-10-18:
Sorry, did not mean to close this ticket. I reformatted the flash with esptool and uploaded again. monitoring for now.  
thanks
-trent

devyte, 2017-10-24:
@trentbrown13 what was the result of your monitoring?

trentbrown13, 2017-10-25:
It was still crashing after re-flashing and I am now downgraded again to 2.3 which is running flawlessly on several esp's. I'll try upgrading to 2.4.0-rc2 soon and try again on a few devices.  Thanks for the help!

tgmaxx, 2017-11-04:
I have the same problem with 2.4.0-rc2

gsimard, 2018-01-01:
I too get this issue with 8edeac0.

The issue seems to be timing related and is hard to pin point.

Essentially I do:

    setup() {
      // Credentials are known from a previous WPS transaction and retrieved from EEPROM.
      WiFi.begin(WiFi.SSID().c_str(),WiFi.psk().c_str());
      // Some more code to wait the above to complete. Connects fine.
      ...
    }

    loop() {
      // This runs fine although it takes a bit of time to complete the transaction
      // after remote end has hanged up
      doHttpsGetTransaction(); 

      // Reset WDT in case we blocked without yield'ing for a while, but shouldn't be necessary
      ESP.wdtFeed();

      // Turn the WiFi off before we sleep
      WiFi.mode(WIFI_OFF);
      WiFi.forceSleepBegin();

      // The WDT (soft or hard if I disable the software wdt) trips HERE, before deepSleep.

      ESP.deepSleep(20e6, WAKE_RF_DEFAULT);
    }

If I do NOT call doHttpsGetTransaction, the ESP8266 goes to sleep and the WDT does NOT trip.

gsimard, 2018-01-02:
Turns out reducing the timeout value of the WiFiSecureClient from 15s to 100ms helps a lot with this issue.

Indeed after the peer hangs up on the connection it took a while for my sketch to take notice and this is due to `Stream::timedRead()` waiting for `_timeout = 15000` set in `WiFiClientSecure::WiFiClientSecure()`.

It looks like shaving these previous seconds allows the return from loop() and the hardware WDT being reset on time.

I don't like the feel of this though as I expect to be bitten again later down the road.

Is there a maximum absolute amount of time one should stay in loop before returning ? Does calling `ESP.wdtFeed()` reset the hardware WDT or just the software WFT ?

Best regards,

SunboX, 2018-08-08:
Can confirm, `WiFi.forceSleepBegin();` triggers a `Soft WDT reset` also for me. Commenting out that line works.

trentbrown13, 2019-04-18:
As with issue https://github.com/esp8266/Arduino/issues/4082, WiFiForceSleepBegin() still caused a WDT reset with core 2.5 until I added a delay(1000) just before it.  With this delay, my sketch has been running for 2+ days with a WiFiForceSleepBegin() every 60 seconds or so.

Not sure if the "waiting for feed back" label is still pertinent, let me know if I can provide more info.

earlephilhower, 2019-07-25:
Wow, looks like WiFi power down has been an issue a looong time.  Closing as we've got an active discussion of this in #6172 where folks are making some progress, I believe.  This will also make a link there to this discussion so they can avoid trying the same dead-ends you ran into.

closed 2019-07-25 by earlephilhower