esp8266/Arduino #3664: Bizarre SPIFFS readBytes crash - only when exactly 52 byes read

philbowles, 2017-09-30:
### Basic Infos

#### Hardware
Hardware:		Wemos D1 Mini
Core Version:      	2.3.0

### Description

readBytes on a SPIFFS file causes reboot (with no error messages) - but only when exactly 52 bytes are read. The file is the same in all case (a "config" file named "/conf").

Dozens of other (shorter AND longer) strings work perfectly and have done for weeks.

The only thing I can think of is that my SPIFFS memory area has become corrupted somehow and that some historic "footprint" of a bad 52-byte file is lying around and keeps getting re-used when the same length string is written again.

### Settings in IDE

Module:  			WeMos D1 R2 & Mini
Flash Size: 		4MB (1M SPIFFS) 
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Sketch

```cpp
String ICACHE_FLASH_ATTR ESPArto::readSPIFFS(const char* fn){
	String rv;
	File f=SPIFFS.open(fn, "r");
	if(f) {
		int n=f.size();
		Serial.printf("readSPIFFS size=%d\n",n);
		char* buff=(char *) malloc(n);
		f.readBytes(buff,n);
		rv=buff;
		free(buff);
	}
	f.close();
	return rv;	
}
```

### Debug Messages

Here's a working Example:

```
readSPIFFS size=53
CFG data=Middle EarthX,Righthander,0,30000,30000,300000,300000

09826c6d
uptime=0d 0h 0m 2s H=27496 W=0 WiFi Connected SSID=ToiioT-Etage
```
Here's another:
```
readSPIFFS size=51
CFG data=MiddleEarth,Righthander,0,30000,30000,300000,300000

09826c6d
uptime=0d 0h 0m 1s H=27680 W=0 asyncp: event/5/4 [1]
uptime=0d 0h 0m 1s H=27856 W=0 asyncp: sound [1]
uptime=0d 0h 0m 2s H=27320 W=0 WiFi Connected SSID=ToiioT-Etage
```

And a 3rd:
```
readSPIFFS size=59
CFG data=Middle Earth,Much longer string,0,30000,30000,300000,300000

09826c6d
uptime=0d 0h 0m 2s H=27488 W=0 WiFi Connected SSID=ToiioT-Etage
```
But as soon as the same file containing exactly 52 bytes is read:
```
readSPIFFS size=52
"B΃K⸮⸮	⸮%⸮⸮u⸮⸮)S
!⸮oXÈ
```
There are no other messages: no boot reason code, nothing but the gibberish and then my setup() routine  running again.

I'm convinved it's a SPIFFS corruption as the same code running on a NodeMCU v0.9 shows:
```
uptime=0d 0h 0m 0s H=36832 W=0 Alive as esparto6
readSPIFFS size=52
CFG data=Middle Earth,Righthander,0,30000,30000,300000,300000

09826c6d
uptime=0d 0h 0m 2s H=27504 W=0 WiFi Connected SSID=ToiioT-Etage
````

How to further diagnose or fix? I have reuploaded my SPIFFS image (which does NOT include the /conf file) many times using the IDE (1.8.3) tool, as well as automatically on boot via ESPhttpUpdate. After each such re-upload, I then use my config web page to write the file, which appears to be written fine. Upon next reboot, the problem of the 52-byte read appears.

I'm going to try one more thing and that is to change the SPIFFS size and try the whole process again, unless someone has any other ideas?

devyte, 2017-09-30:
@philbowles You are mallocing 53 bytes for your buff, and I count 53 chars in your printout. Aren't you forgetting the null term? You should allocate n+1, and probably also do buff[n] = 0; after reading (index right after your read data).
Also, why are you reading into an alloced char * only to assign to a String, when you can call f.readString() directly?
Closing due to usage error (see #3655 ). If after you fix the errors you still see an issue, please revisit or open a new issue, and also please provide a MCVE.

closed 2017-09-30 by devyte

philbowles, 2017-09-30:
I changed the SPIFFS size to 4M (3M SPIFFS) uploaded SPIFFS image using IDE ...same problem. Will now look into reformatting the SPIFFS area and starting again...

philbowles, 2017-09-30:
DOH! Schoolboy error - just spotted it myself and working to fix...I feel such an idiot! Sorry to have wasted your time (and tons of mine!) :(