esp8266/Arduino #3667: Example: how to send client certificate?

bscheshir, 2017-10-01:
I can see many powerfull examples, great!
[with CA](https://github.com/esp8266/Arduino/blob/f6d232f1ac68fd52a69bb9a03995eaa6d7fde922/libraries/ESP8266WiFi/examples/HTTPSRequestCACert/HTTPSRequestCACert.ino)

But how I can send  `client certificate+key` in `TLS Handshake` phase for case server AND client verification (`Certificate`, `CertificateVerify`)?

I think this case is typical for remote sensor based on esp8266 and mounted out of smartHome local network, right?

bscheshir, 2017-10-01:
for something like this
```sh
curl -v -k --key ssl/client01.key --cert ssl/client01.crt --url https://smartHomeRESTAPIDomain:8082/v1/sensor/1/measure/create
```

bscheshir, 2017-10-01:
So... CA self assigned, server and all of clients has been signed by this self assigned CA.

devyte, 2017-10-03:
Closing as off topic, see #3655 .

closed 2017-10-03 by devyte

copercini, 2017-10-04:
@bscheshir I have some examples here: https://github.com/copercini/esp8266-aws_iot

But due memory limitations it runs on limit